openbmc/openbmc #419: Barreleye builds failing

shenki, 2016-06-28:
```
07:25:22 Currently 2 running tasks (2270 of 2283):
07:25:55 ERROR: obmc-phosphor-image-1.0-r0 do_rootfs: Unable to install packages. Command '/var/lib/jenkins/workspace/openbmc-build/distro/ubuntu/target/barreleye/openbmc/build/tmp/sysroots/x86_64-linux/usr/bin/smart --log-level=warning --data-dir=/var/lib/jenkins/workspace/openbmc-build/distro/ubuntu/target/barreleye/openbmc/build/tmp/work/barreleye-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/rootfs/var/lib/smart install -y packagegroup-core-ssh-dropbear@all packagegroup-obmc-phosphor-apps-policy-mgmt@all packagegroup-obmc-phosphor-apps-chassis-mgmt@all inarp@armv5e packagegroup-obmc-phosphor-apps-fan-mgmt@all packagegroup-obmc-phosphor-apps-sensor-mgmt@all packagegroup-obmc-phosphor-apps-event-mgmt@all packagegroup-core-boot@barreleye screen@armv5e u-boot-fw-utils@barreleye packagegroup-obmc-phosphor-apps-flash-mgmt@all shadow@armv5e packagegroup-obmc-phosphor-apps-system-mgmt@all bash@armv5e packagegroup-obmc-phosphor-apps-user-mgmt@all run-postinsts@all obmc-console@armv5e inventory-upload@all packagegroup-obmc-phosphor-apps-extras@all i2c-tools@armv5e packagegroup-base-extended@barreleye base-passwd@armv5e locale-base-en-us@armv5e' returned 1:
07:25:55 Loading cache...
07:25:55 Updating cache...               ######################################## [100%]
07:25:55 
07:25:55 Computing transaction...error: Can't install skeleton-1.0-r1@armv5e: no package provides libopenbmc_intf.so
07:25:55 
07:25:55 
07:25:55 Currently 2 running tasks (2270 of 2283):
07:25:55 ERROR: obmc-phosphor-image-1.0-r0 do_rootfs: Function failed: do_rootfs
07:25:55 Currently 2 running tasks (2270 of 2283):
07:25:56 ERROR: Logfile of failure stored in: /var/lib/jenkins/workspace/openbmc-build/distro/ubuntu/target/barreleye/openbmc/build/tmp/work/barreleye-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/temp/log.do_rootfs.12655
07:25:56 Currently 1 running tasks (2270 of 2283):
07:25:56 ERROR: Task 10 (/var/lib/jenkins/workspace/openbmc-build/distro/ubuntu/target/barreleye/openbmc/meta-phosphor/common/recipes-phosphor/images/obmc-phosphor-image.bb, do_rootfs) failed with exit code '1'
```

williamspatrick, 2016-06-28:
Did you run this manually or on the server with the sstate cache?  I was observing this myself as part of the upgrade to Yocto 2.1 but I never saw it before and the Jenkins server never failed this way.  I don't believe I see it when building a clean stand-alone compile, so I suspect the Jenkins server's sstate cache is messing this up now.

Brad does have a fix in the v1.0.1 tag.

shenki, 2016-06-29:
This is on the Jenkins server. We're seeing it with master (https://github.com/openbmc/openbmc/commit/2d1d6b72143ce6d8bf4a5bbbb95eb28994876a54):

https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=barreleye/131/console

Has this somehow stuffed up the sscache? Do I need to remove it before builds will work again?

williamspatrick, 2016-06-29:
I suspect so?  Can you do a build with sstate disabled temporarily and see if that solves this?

shenki, 2016-06-30:
Barreleye is okay now that I removed the cache.

closed 2016-06-30 by shenki