openbmc/openbmc #421: Store device tree in separate partition

shenki, 2016-06-29:
We want to move to storing the device tree separately from the kernel in the flash. This enables u-boot to perform modifications to the device tree before handing it off to Linux, such as setting the amount of RAM or even NC-SI vs PHY.

Changes required:
 - stop building cuImage
 - modify flash layout to set aside space for board.dtb
 - adjust u-boot environment so it loads the dtb

williamspatrick, 2016-06-29:
Do we actually need a new partition for this or can we put it into the u-boot image directly?  Can we make a single EXT? fs for the dtb and kernel image?

shenki, 2017-06-06:
This was before we moved to FIT. Given we're using FIT in the system, we will no longer plan to store the device tree separately.

closed 2017-06-06 by shenki