openbmc/openbmc #423: BIOS version doesn't show via openBMC REST

williamspatrick, 2016-06-29:
@Kenthliu commented on [Wed Jun 29 2016](https://github.com/openbmc/skeleton/issues/116)

With newer BMC releases (v1.0) and BIOS releases (V 0.2.2) I am noticing that we can't get the BIOS version info with curl command via the BMC (even when host is up).  But BMC v0.8 was work.
BMC 0.8 and BIOS 0.2.2
[image]
BMC 1.0 and BIOS 0.2.2
[image]

williamspatrick, 2016-06-29:
@anoo1 - Can you screen this?

anoo1, 2016-07-05:
Hi @Kenthliu, I'm unable to recreate on v1.0.1 or the new v1.0.2. If you still see this issue after power off/on, could you check if there are inventory files created under /tmp/ipmifru* and if there's a bios file "ipmifru00" if you could dump its contents with "hexdump -C /tmp/ipmifru00"

johnhcwang, 2016-07-06:
Hi @anoo1 , there is no any fru file under /tmp/.
[image]

May I know what BIOS version you used? We used 0.2.2 on our board.

By the way, there are some error messages in journal log, but I'm not sure if it is helpful to this issue.
Please get them from https://github.com/openbmc/skeleton/pull/98#issuecomment-229573501

anoo1, 2016-07-06:
hi @johnhcwang , this is the bios version currently on our system:

Asset Tag = 
Custom Field 1 = 	buildroot-fda20bd
Custom Field 2 = 	skiboot-5.2.0-opdirty
Custom Field 3 = 	hostboot-c223637-1d3dbf3
Custom Field 4 = 	linux-4.4.6-openpower1-opdirty-99d283b
Custom Field 5 = 	petitboot-72928ed-opdirty-47f54ff
Custom Field 6 = 	barreleye-xml-1250f06-opdirty
Custom Field 7 = 	occ-1093bf9-opdir
Custom Field 8 = 
FRU File ID = 
Manufacturer = 
Model Number = 
Name = OpenPOWER Firmware
Serial Number = 
Version = open-power-barreleye-v1.8

I see the same error messages so they're not the cause, I'd look to see why they're being printed. The bios info seems to be sent down later around the middle of the power on. Without any /tmp/ fru files seems that no inventory was collected or sent. Are other ipmi commands working? Want to check that ipmi process is running. Perhaps also power off and restart ipmi with traces enabled: "/usr/sbin/ipmid -d 0xff &" and power on to see if there are any errors while receiving the fru data.

johnhcwang, 2016-07-07:
Yes, I can get bmc info from petitboot with ipmi command.
[image]

And I restart the ipmi service with the debug mode and power on, here are some journal log for your information.
```
systemd[1]: Stopping Phosphor OpenBMC IPMI daemon...
barreleye systemd[1]: Stopped Phosphor OpenBMC IPMI daemon.
barreleye systemd[1]: Started Phosphor OpenBMC IPMI daemon.
barreleye system_manager.py[974]: Turn on power and boot
barreleye system_manager.py[974]: Running System State: HOST_POWERING_ON
barreleye system_manager.py[974]: Setting watchdog poll interval to: 30000
barreleye system_manager.py[974]: Starting watchdog with poll interval: 30000
barreleye kernel: ftgmac100: NCSI interface down
barreleye kernel[982]: ftgmac100: NCSI interface down
barreleye kernel: ftgmac100: NCSI interface up
barreleye kernel[982]: ftgmac100: NCSI interface up
barreleye system_manager.py[974]: Running System State: HOST_POWERED_ON
barreleye system_manager.py[974]: Starting process: /usr/sbin/goto_system_state.py INVENTORY_UPLOADED inventory_upload.py: inventory_upload
Abarreleye kernel: ftgmac100: NCSI interface down
barreleye kernel[982]: ftgmac100: NCSI interface down
barreleye kernel: ftgmac100: NCSI interface up
barreleye kernel[982]: ftgmac100: NCSI interface up
barreleye system_manager.py[974]: ERROR (name_properties): "name" property in /bmc/inventory/hdd_backplane is incorrect ("BARRELEYE Exander" instead of base node name)
barreleye system_manager.py[974]: ERROR (name_properties): "name" property in /bmc/inventory/sas_expander is incorrect ("BARRELEYE BPx15" instead of base node name)
barreleye system_manager.py[974]: ERROR (name_properties): "name" property in /bmc/inventory/misc is incorrect ("BARRELEYE MB" instead of base node name)
barreleye system_manager.py[974]: ERROR: Input tree has errors, aborting (use -f to force output)
barreleye system_manager.py[974]: About to erase 0x01e75000..0x01e7e000 !
barreleye system_manager.py[974]: Erasing...
barreleye system_manager.py[974]: [569B blob data]
barreleye system_manager.py[974]: About to program "/tmp/tmp289yaO" at 0x01e75000..0x01e7e000 !
barreleye system_manager.py[974]: Programming & Verifying...
barreleye system_manager.py[974]: [                                                  ] 0%
barreleye system_manager.py[974]: Updating actual size in partition header...
barreleye system_manager.py[974]: Uploading inventory to PNOR in dtb format...
barreleye system_manager.py[974]: Running System State: INVENTORY_UPLOADED
barreleye system_manager.py[974]: Starting process: /usr/sbin/pcie_slot_present.exe: pcie_present
barreleye system_manager.py[974]: Setting fans to max
barreleye system_manager.py[974]: Booting host
barreleye system_manager.py[974]: Checking Presence: SLOT0_RISER_PRESENT
barreleye system_manager.py[974]: GPIO Lookup:  SLOT0_RISER_PRESENT = 424,in
barreleye system_manager.py[974]: Checking Presence: SLOT1_RISER_PRESENT
barreleye system_manager.py[974]: GPIO Lookup:  SLOT1_RISER_PRESENT = 425,in
barreleye system_manager.py[974]: Checking Presence: SLOT2_RISER_PRESENT
barreleye system_manager.py[974]: GPIO Lookup:  SLOT2_RISER_PRESENT = 426,in
barreleye system_manager.py[974]: Checking Presence: SLOT0_PRESENT
barreleye system_manager.py[974]: GPIO Lookup:  SLOT0_PRESENT = 427,in
barreleye system_manager.py[974]: Checking Presence: SLOT1_PRESENT
barreleye system_manager.py[974]: GPIO Lookup:  SLOT1_PRESENT = 428,in
barreleye system_manager.py[974]: Checking Presence: SLOT2_PRESENT
barreleye system_manager.py[974]: GPIO Lookup:  SLOT2_PRESENT = 429,in
barreleye system_manager.py[974]: Using primary side of the bios flash
barreleye system_manager.py[974]: Checking Presence: MEZZ0_PRESENT
barreleye system_manager.py[974]: GPIO Lookup:  MEZZ0_PRESENT = 432,in
barreleye system_manager.py[974]: Checking Presence: MEZZ1_PRESENT
barreleye system_manager.py[974]: Running System State: HOST_BOOTING
barreleye system_manager.py[974]: GPIO Lookup:  MEZZ1_PRESENT = 433,in
barreleye kernel: ftgmac100: NCSI interface down
barreleye kernel[982]: ftgmac100: NCSI interface down
barreleye kernel: ftgmac100: NCSI interface up
barreleye kernel[982]: ftgmac100: NCSI interface up
barreleye system_manager.py[974]: Setting watchdog poll interval to: 120000
barreleye system_manager.py[974]: Stopping watchdog
barreleye system_manager.py[974]: Starting watchdog with poll interval: 120000
barreleye ipmid[1233]: No registered handlers for NetFn:[0xA], Cmd:[0x12] trying Wilcard implementation
barreleye ipmid[1233]: Registering handler:[/usr/lib/host-ipmid/liboemhandler.so]
barreleye ipmid[1233]: Registering NetFn:[0x32], Cmd:[0xF0]
barreleye ipmid[1233]: Registering NetFn:[0x32], Cmd:[0x10]
barreleye ipmid[1233]: Registering handler:[/usr/lib/host-ipmid/libhostservice.so]
barreleye ipmid[1233]: Registering handler:[/usr/lib/host-ipmid/libapphandler.so]
barreleye ipmid[1233]: Registering NetFn:[0x6], Cmd:[0x36]
[... 49 lines cut ...]
barreleye ipmid[1233]: 0x000000: 05 a8 00 04 00 00 00 00 14 00                   ..........
barreleye ipmid[1233]: IPMI SET_SENSOR [0x05]
barreleye ipmid[1233]: Attempting to set a dbus Variant Sensor 0x05 via setValue with a value of FW Progress, Baseboard Init
barreleye ipmid[1233]: IPMI Response:
barreleye ipmid[1233]: 0x000000: 00                                              .
barreleye ipmid[1233]: IPMI Incoming: Seq 0x05, NetFn 0x06, CMD: 0x22
barreleye ipmid[1233]: WATCHDOG RESET
barreleye ipmid[1233]: IPMI Response:
[... cut ...]
barreleye ipmid[1233]: 0x000030: 72 72 65 6c 65 79 65 2d 76 30 2e 32             rreleye-v0.2
barreleye ipmid[1233]: Handling STORAGE WILDCARD Netfn:[0xA], Cmd:[0x12]
barreleye ipmid[1233]: IPMI Response:
barreleye ipmid[1233]: 0x000000: 00                                              .
````
Seems BIOS try to write the data by Write FRU command (NetFn 0x0a, CMD 0x12), but there is no registered handler to handle this command.
So I experimented this command from petitboot and get it from journal log.
[image]

anoo1, 2016-07-07:
Thanks John for all the info. It appears that ipmid is not picking up the storage handler library, this trace is is not being printed when the ipmid process was started:
'''
Registering handler:[/usr/lib/host-ipmid/libstrgfnhandler.so]
Registering NetFn:[0xA], Cmd:[0x12]
Registering handler:[/usr/lib/host-ipmid/liboemhandler.so]
'''
Could you check the system has this library: " ls -l /usr/lib/host-ipmid/libstrgfnhandler.so* "
If it's missing maybe the system would can be updated and verify it's back on the system.

johnhcwang, 2016-07-07:
Do you mean llibstrgfnhandler.so.1?? I can see it under /host-ipmid/, but still no register handler.
````
root@barreleye:~# ls /usr/lib/host-ipmid/
libapphandler.so       liboemhandler.so
libhostservice.so      libstrgfnhandler.so.1
root@barreleye:~#
````

anoo1, 2016-07-07:
Hi John, thanks, you're correct, checking with the team there was a recent change to add a version to some of the bmc libraries. This shouldn't cause an issue except for applications that issue dlopen, but ipmid uses dlopen like you pointed out. I need to make a code change in ipmid so that it loads the versioned library files.

closed 2016-07-11 by williamspatrick