openbmc/openbmc #426: witherspoon fails under jenkins due to bash-completion test directory

shenki, 2016-06-30:
Jenkins throws an exception due to a filename with question marks in it:

```
16:36:50 Caused by: java.nio.file.InvalidPathException: Malformed input or input contains unmappable chacraters: /var/lib/jenkins/workspace/openbmc-build/distro/ubuntu/target/witherspoon/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/bash-completion/2.1-r0/bash-completion-2.1/test/fixtures/_filedir/a??
```

The only image to pull in the bash-completion package is Witherspoon.

Looking at the dependencies:

```
$ TEMPLATECONF=meta-openbmc-machines/meta-openpower/meta-ibm/meta-witherspoon/conf source oe-init-build-env
$ bitbake -g obmc-phosphor-image
$ grep '>.*bash-comp' pn-depends.dot 
"systemd" -> "bash-completion" [style=dashed]
"bash-completion" -> "bash-completion-dev" [style=dashed]
"bash-completion" -> "bash-completion" [style=dashed]
"util-linux" -> "bash-completion" [style=dashed]
"kmod" -> "bash-completion" [style=dashed]
"glib-2.0" -> "bash-completion" [style=dashed]
"dbus-glib" -> "bash-completion" [style=dashed]
```

williamspatrick, 2016-06-30:
Can't we fix Jenkins?

williamspatrick, 2016-07-01:
All of the Barreleye images appear to be failing on Jenkins as well.  Are you sure that only witherspoon is pulling this in?

williamspatrick, 2016-07-27:
Fixed.

closed 2016-07-27 by williamspatrick