openbmc/openbmc #427: Duplicated binary installed for phosphor-event

adamliyi, 2016-06-30:
In "meta-phosphor/common/recipes-phosphor/obmc-phosphor-event/obmc-phosphor-event.bb", there is:
```
S = "${WORKDIR}/git"
INSTALL_NAME = "event_messaged"

do_install() {
        install -d ${D}/var/lib/obmc/events/
        install -m 0755 -d ${D}${sbindir}
        install -m 0755 ${S}/${INSTALL_NAME} ${D}/${sbindir}/obmc-phosphor-eventd
}
```
However, both "event_messaged" and "obmc-phosphor-eventd" are installed into /usr/sbin/.
These two binaries are actually the same, with different names.

```
root@palmetto:~# ls /usr/sbin/event_messaged 
/usr/sbin/event_messaged
root@palmetto:~# ls /usr/sbin/obmc-phosphor-eventd 
/usr/sbin/obmc-phosphor-eventd
```
This is not a big issue, but it is confusing, and it wastes additional 100KB space in BMC.

williamspatrick, 2016-06-30:
Assign to @adamliyi .

gtmills, 2018-05-23:
obmc-phosphor-event/obmc-phosphor-event.bb no longer exists.

gtmills, 2018-05-23:
Closing this issue since it no longer applies.

closed 2018-05-23 by gtmills