openbmc/openbmc #428: CORE and DIMM Present value as False

sivassrr, 2016-06-30:
Based on the discussion in issue #93 

https://github.com/openbmc/openbmc/issues/93

With level 125, Still sees failure in Palmetto as well for DIMMS. 
rg.openbmc.InventoryItem
setPresent(string present) → setFault(string fault) → update(dict {string → variant} data) → fault = False fru_type = DIMM is_fru = 1

williamspatrick, 2016-06-30:
What is "level 125"?

williamspatrick, 2016-06-30:
Assign to adamliyi .

sivassrr, 2016-07-01:
with the latest level (Jun 30 build ) - Here is the output.
Sivass-MBP:~ sivas$ curl -c cjar -b cjar -k https://9.3.23.32/org/openbmc/inventory/enumerate
{
  "data": {
    "/org/openbmc/inventory/system": {
      "fault": "False",
      "fru_type": "SYSTEM",
      "is_fru": 1,
      "present": "True",
      "version": ""
    },
    "/org/openbmc/inventory/system/bios": {
      "fault": "False",
      "fru_type": "SYSTEM",
      "is_fru": 1,
      "present": "True",
      "version": ""
    },
    "/org/openbmc/inventory/system/chassis": {
      "fault": "False",
      "fru_type": "SYSTEM",
      "is_fru": 1,
      "present": "True",
      "version": ""
    },
    "/org/openbmc/inventory/system/chassis/fan0": {
      "fault": "False",
      "fru_type": "FAN",
      "is_fru": 1,
      "present": "False",
      "version": ""
    },
    "/org/openbmc/inventory/system/chassis/fan1": {
      "fault": "False",
      "fru_type": "FAN",
      "is_fru": 1,
      "present": "False",
      "version": ""
    },
    "/org/openbmc/inventory/system/chassis/fan2": {
      "fault": "False",
      "fru_type": "FAN",
      "is_fru": 1,
      "present": "False",
      "version": ""
    },
    "/org/openbmc/inventory/system/chassis/fan3": {
      "fault": "False",
      "fru_type": "FAN",
      "is_fru": 1,
      "present": "False",
      "version": ""
    },
    "/org/openbmc/inventory/system/chassis/fan4": {
      "fault": "False",
      "fru_type": "FAN",
      "is_fru": 1,
      "present": "False",
      "version": ""
    },
    "/org/openbmc/inventory/system/chassis/fan5": {
      "fault": "False",
      "fru_type": "FAN",
      "is_fru": 1,
      "present": "False",
      "version": ""
    },
    "/org/openbmc/inventory/system/chassis/hdd_backplane": {
      "fault": "True",
      "fru_type": "DAUGHTER_CARD",
[... 624 lines cut ...]
      "Name": "BARRELEYE MB",
      "Part Number": "020305R00-600-G",
      "Serial Number": "020305R00X1571001",
      "fault": "False",
      "fru_type": "SYSTEM",
      "is_fru": 0,
      "present": "True",
      "version": ""
    },
    "/org/openbmc/inventory/system/systemevent": {
      "fault": "False",
      "fru_type": "SYSTEM_EVENT",
      "is_fru": 0,
      "present": "False",
      "version": ""
    },
    "/org/openbmc/inventory/system/systemevent/event": {
      "endpoints": [
        "/org/openbmc/records/events/10"
      ]
    }
  },
  "message": "200 OK",
  "status": "ok"
}

williamspatrick, 2016-07-01:
@sivassrr - You are seeing this after letting the host boot, correct?  It appears that the IPMI inventory commands are broken then.  @adamliyi will investigate.

adamliyi, 2016-07-05:
@sivassrr: I think you did not boot host. You can check "obmcutil state" firstly, to make sure host has booted. Bellow are inventory on Palmetto:
Check system status:
```
root@palmetto:~# obmcutil state
 = HOST_BOOTED
```

```
curl -c cjar -b cjar -k https://192.168.0.60/org/openbmc/inventory/enumerate
{
  "data": {
    "/org/openbmc/inventory/system": {
      "Asset Tag": [], 
      "Custom Field 1": "\thostboot-bc98d0b-d9ee868", 
      "Custom Field 2": "\tocc-0362706", 
      "Custom Field 3": "\tskiboot-5.1.5-456d3df", 
      "Custom Field 4": "\thostboot-binaries-43d5a59", 
      "Custom Field 5": "\tpalmetto-xml-8ce36f3-6bfc689", 
      "Custom Field 6": "\tcapp-ucode-105cb8f", 
      "Custom Field 7": [], 
      "Custom Field 8": [], 
      "FRU File ID": [], 
      "Manufacturer": [], 
      "Model Number": [], 
      "Name": "OpenPOWER Firmware", 
      "Serial Number": [], 
      "Version": "open-power-palmetto-v1.7", 
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis": {
      "Custom Field 1": [], 
      "Custom Field 2": [], 
      "Custom Field 3": [], 
      "Custom Field 4": [], 
      "Custom Field 5": [], 
      "Custom Field 6": [], 
      "Custom Field 7": [], 
      "Custom Field 8": [], 
      "Part Number": "0123456789      ", 
      "Serial Number": "0000000000000000", 
      "Type": 2, 
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan0": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan1": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan2": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
[... 340 lines cut ...]
      "fault": "False", 
      "fru_type": "MAIN_PLANAR", 
      "is_fru": 0, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/motherboard/todclock": {
      "fault": "False", 
      "fru_type": "MAIN_PLANAR", 
      "is_fru": 0, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/systemevent": {
      "fault": "False", 
      "fru_type": "SYSTEM_EVENT", 
      "is_fru": 0, 
      "present": "False", 
      "version": ""
    }
  }, 
  "message": "200 OK", 
  "status": "ok"

```

adamliyi, 2016-07-05:
check OPAL output, only 4 cores are available on the Palmetto I do the test, so the FRU/inventory work as expected:
```
root@ubuntu1504:/home/tyan# cat /sys/firmware/opal/msglog | grep -ie core
[7293305603,7] OCC: Chip 0 Core 2 PPMSR c3c3bc0008000000
[7293823043,7] OCC: Chip 0 Core 5 PPMSR ccccbc0008000000
[7294341260,7] OCC: Chip 0 Core b PPMSR d5d5bc0008000000
[7294857023,7] OCC: Chip 0 Core d PPMSR dedebc0008000000
```
So this issue is not a bug.

adamliyi, 2016-07-05:
Tested on Palmetto, using OpenBMC-v1.0, the CPU, Core, and DIMM present bit work as expected.
So @sivassrr , could you boot host, and verify?

```
url -c cjar -b cjar -k https://192.168.0.60/org/openbmc/inventory/enumerate
{
  "data": {
    "/org/openbmc/inventory/system": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan0": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan1": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan2": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan3": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan4": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/io_board/pcie_slot0": {
      "fault": "False", 
      "fru_type": "PCIE_CARD", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/io_board/pcie_slot1": {
      "fault": "False", 
      "fru_type": "PCIE_CARD", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
[... 156 lines cut ...]
      "fault": "False", 
      "fru_type": "MAIN_PLANAR", 
      "is_fru": 0, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/motherboard/todclock": {
      "fault": "False", 
      "fru_type": "MAIN_PLANAR", 
      "is_fru": 0, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/systemevent": {
      "fault": "False", 
      "fru_type": "SYSTEM_EVENT", 
      "is_fru": 0, 
      "present": "False", 
      "version": ""
    }
  }, 
  "message": "200 OK", 
  "status": "ok"

```

sivassrr, 2016-07-05:
Below output is from 9.3.164.153 (Palmetto) - This has V1.0 and Jun 30 build/level. 
Even at BMC READY (Similar to FSP Standby) state, expect to have right presence data, right.

root@palmetto:~# obmcutil getinventory
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system
fault = False
fru_type = SYSTEM
is_fru = 1
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis
fault = False
fru_type = SYSTEM
is_fru = 1
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/fan0
fault = False
fru_type = FAN
is_fru = 1
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/fan1
fault = False
fru_type = FAN
is_fru = 1
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/fan2
fault = False
fru_type = FAN
is_fru = 1
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/fan3
fault = False
fru_type = FAN
is_fru = 1
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/fan4
fault = False
fru_type = FAN
is_fru = 1
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/io_board/pcie_slot0
fault = False
fru_type = PCIE_CARD
is_fru = 1
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/io_board/pcie_slot1
fault = False
fru_type = PCIE_CARD
is_fru = 1
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard
fault = False
fru_type = MAIN_PLANAR
is_fru = 1
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/apss
fault = False
fru_type = MAIN_PLANAR
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/bmc
fault = False
fru_type = BMC
is_fru = 0
manufacturer = ASPEED
present = False
version = "v1.0-48-g8d43aee-dirty"
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/cpu0
fault = False
fru_type = CPU
is_fru = 1
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/cpu0/core0
fault = False
fru_type = CORE
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/cpu0/core1
fault = False
fru_type = CORE
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/cpu0/core10
fault = False
fru_type = CORE
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/cpu0/core11
fault = False
fru_type = CORE
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/cpu0/core2
fault = False
fru_type = CORE
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/cpu0/core3
fault = False
fru_type = CORE
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/cpu0/core4
fault = False
fru_type = CORE
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/cpu0/core5
fault = False
fru_type = CORE
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/cpu0/core6
fault = False
fru_type = CORE
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/cpu0/core7
fault = False
fru_type = CORE
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/cpu0/core8
fault = False
fru_type = CORE
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/cpu0/core9
fault = False
fru_type = CORE
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/dimm0
fault = False
fru_type = DIMM
is_fru = 1
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/dimm1
fault = False
fru_type = DIMM
is_fru = 1
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/dimm2
fault = False
fru_type = DIMM
is_fru = 1
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/dimm3
fault = False
fru_type = DIMM
is_fru = 1
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/membuf0
fault = False
fru_type = MEMORY_BUFFER
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/pcieclock
fault = False
fru_type = MAIN_PLANAR
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/refclock
fault = False
fru_type = MAIN_PLANAR
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/chassis/motherboard/todclock
fault = False
fru_type = MAIN_PLANAR
is_fru = 0
present = False
version =
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/org/openbmc/inventory/system/systemevent
fault = False
fru_type = SYSTEM_EVENT
is_fru = 0
present = False
version =
root@palmetto:~# cat /etc/version
20160630213138
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="0.1.0 (krogoth)"
VERSION_ID="v1.0-48-g8d43aee-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (krogoth)"
BUILD_ID="v1.0"
root@palmetto:~# obmcutil state
 = HOST_BOOTING

williamspatrick, 2016-07-05:
@sivassrr -
```
Even at BMC READY (Similar to FSP Standby) state, expect to have right presence data, right.
```

Not entirely true.  We only expect to have right presence data if the machine has booted once before.

1. Fresh BMC, at BMC standby. - No data.
2. Previously booted BMC, at BMC standby. - Possibly stale, but valid, data.
3. BMC at runtime. - Fresh, valid data.

The BMC cannot provide DIMM and CPU presence detect at standby because it does not have hardware access directly to the DIMMs and Cores.  This is different from the FSP.

This may change on Power9 systems with the OpenFSI drivers, but for Power8 we do not have an access method.

sivassrr, 2016-07-27:
Sure, Will verify and change the test code.  At the same time, Would like to put one suggestion in this. If  the system is at BMC_READY state then we should not be allowed to issue REST commands related to HOST dependency.  Or populate the value with "HOST off state" information.

sivassrr, 2016-09-02:
Verified and works

closed 2016-09-02 by sivassrr