openbmc/openbmc #430: Design and Implement framework for IPMI backend.

williamspatrick, 2016-06-30:


tomjoseph83, 2016-11-02:
- [ ]  Separate the commands from the libapphandler that should be executed only from host-ipmid path(For example Read Event Message Buffer, Set BMC Global Enables)
- [ ]  Modify the Command Table to add infrastructure for provider functions
- [ ]  Add code in netipmid to read the libraries and register the provider functions in the /usr/lib/phosphor-net-ipmid
- [ ]   Modify the registration functions in phosphor-host-ipmid to suit netipmid (add privilege levels for the command)
- [ ] Modify the registration functions in ipmi-fru-parser
- [ ] Modify the registration functions in ibm-oem
- [ ] Modify the recipe file/makefile to place the libraries in a common path in install image and create necessary softlinks for phosphor-net-ipmid and phosphor-host-ipmid
- [ ] Create a recipe for phosphor-net-ipmid

tomjoseph83, 2017-01-09:
phosphor-host-ipmid(5 patches) - https://gerrit.openbmc-project.xyz/#/c/1365
phosphor-host-ipmid(4 patches) - https://gerrit.openbmc-project.xyz/#/c/1020
ipmi-fru-parser - https://gerrit.openbmc-project.xyz/#/c/399/
openpower-host-ipmi-oem - https://gerrit.openbmc-project.xyz/#/c/511/

closed 2017-01-31 by williamspatrick