openbmc/openbmc #432: Attaching RMCP session to SOL.

williamspatrick, 2016-06-30:


tomjoseph83, 2016-12-02:
- [ ] Adapt the sd_event_loop to handle incoming IPMI traffic, monitor activity on host console socket and  add timers for character accumulate interval and retry interval.

- [ ] Design and implement a SOL engine to process the state of the SOL and sent the data to the IPMI client

- [ ] Design and implement a storage for the host buffer

- [ ] Modify the session data to keep channel client information and SOL metadata

- [ ] SOL Incoming Payload handler to route inbound SOL traffic to the host

- [ ] Modify Message Handler to sent unsolicited SOL payload Type 

- [ ] Implement SOL Activate command

- [ ] Implement SOL Deactivate command

- [ ] Implement Changes to handle multiple payload instances

- [ ] Implement Get SOL Configuration Parameters Command

- [ ] Set SOL Configuration Parameters Command

- [ ] Get Payload Activation Status

geissonator, 2016-12-14:
We've moved this one out a bit so just going to move into the backlog.

williamspatrick, 2017-06-02:
Can this be closed?

closed 2017-07-13 by zahrens