openbmc/openbmc #441: Implement sensor / SDR commands.

williamspatrick, 2016-06-30:


ongchris, 2016-12-21:
Is the plan to implement an IPMI-compliant SDR? Is the current sensor polling done in pyhwmon/hwmon.py?

kc8apf, 2016-12-21:
Zaius needs IPMI SDR so it is definitely on Google's schedule.  There has been some design discussion on the mailing list.

williamspatrick, 2016-12-21:
Yes, that is the plan.  To implement a translation between the local dbus objects and an IPMI SDR.

@bradbishop is working on a replacement for the pyhwmon implementation.

yadron, 2017-03-10:
@williamspatrick , could you please review the code below?
Do I have access for publishing these patches to https://gerrit.openbmc-project.xyz ?

How to contact @bradbishop ?
I need more details: what is the plan and if started already?

Personally implemented one new sensors/SDR command (both in host-ipmid and in DBUS). Describing it below.

### Read first:

1. http://www.intel.com/content/www/us/en/servers/ipmi/second-gen-interface-spec-v2-rev1-4.html - Intel document that describes IPMI commands (and the command being implemented described at page 417).
2. http://www.dsm.fordham.edu/cgi-bin/man-cgi.pl?topic=sd_bus_message_append&ampsect=3 - description of DBUS params for calling function.

### Command get num of sensors ("Get Device SDR Info").

Successfully added "get number of sensors" function both to DBUS and to IPMId-host.
This function is "0x04 0x20" IPMI command (that is "Get Device SDR Info", see example https://computercheese.blogspot.ru/2013/05/ipmi-sensor-device-commands.html?view=timeslide or page 417 in Intel doc).

### Changed components:

1. /usr/sbin/system_manager.py at BMC that is "meta-phosphor/common/recipes-phosphor/system/obmc-mgr-system".
This is DBUS component (that is "/etc/dbus-1/system.d/org.openbmc.managers.System.conf").

Is it right that I've edited system_manager.py DBUS component? I've just seen that it parses hardware configuration already, so it was simple adding the count sensors new DBUS function here.

2. /usr/sbin/ipmid that is "phosphor-ipmi-host".

### Overview

Generally, IPMI command itself added to IPMI-host (ipmid) and DBUS new function to org.openbmc.Sensors.service (system_manager.py).

Used this command at host for testing: "ipmitool sensor list -vvvv"

Used these two commands at BMC for testing:

```
python /usr/sbin/system_manager.py &
/usr/sbin/ipmid_new
```

Used these commands for debugging at build server (updating only ipmid without recompiling of OpenBMC):

```
[build@obmc build]$ rm ./tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/build/.libs/libapphandler.so.0
[build@obmc build]$ rm ./tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/build/ipmid
[build@obmc build]$ bitbake -f -c compile phosphor-ipmi-host
[build@obmc build]$ scp ./tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/build/.libs/libapphandler.so.0 root@172.17.27.77:/usr/lib/host-ipmid/libapphandler.so.0
[build@obmc build]$ scp ./tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/build/ipmid root@172.17.27.77:/usr/sbin/ipmid_new
and manual edition: "root@barreleye:~# vi /usr/sbin/system_manager.py"
```

### Patch for phosphor-ipmi-host

Code added to tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/git/sensorhandler.cpp _(this patch is prototype, because currently it always return the same ID="0x1, 0x2, 0x3, 0x4" and always returns num of sensors, but as of specification it should return "total number of SDRs in device" or "the number of sensors implemented on LUN" depending on requested operation)_:

```
[build@obmc ~]$ diff ./openbmc-prev/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/git/sensorhandler.cpp ~/sensors/sensorhandler_updated.cpp
102a103,151
> // yadron
> // see doc: http://www.dsm.fordham.edu/cgi-bin/man-cgi.pl?topic=sd_bus_message_append&ampsect=3
> // and see ./tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/git/testaddsel.cpp
> int get_sensors_count() {
>     char  *busname = NULL;
>     const char  *iface = "org.openbmc.managers.System";
>     const char  *objname = "/org/openbmc/managers/System";
>     sd_bus_error error = SD_BUS_ERROR_NULL;
>     sd_bus_message *reply = NULL;
>     int32_t num;
>     sd_bus_message *m=NULL;
> 
[... 104 lines cut ...]
For example, command "ipmitool sensor list -vvvv" for testing of "Get num of sensors".
If ipmitool output some error (e.g. "Err in cmd get sensor sdr info"), then read what it means in source code "alarkin@lt-alarkin:~/openbmc/ipmitool/ipmitool-1.8.17$ vi ./lib/ipmi_sdr.c".

The output of ipmid and "/usr/sbin/system_manager.py" at BMC helps debugging too.

Successful result:

```
[root@localhost ~]# ipmitool sensor list -vvvv
...
OpenIPMI Request Message Header:
 netfn = 0x4
 cmd = 0x20
Sending request 0x20 to System Interface
SDR records : 81
```

and both

```
root@barreleye:~# python /usr/sbin/system_manager.py &
[1] 12444
...
Counted 81
```

yadron, 2017-06-29:
Looks like my suggested "IPMI_CMD_GET_LIST_OF_SENSORS = 0x20" and "IPMI_CMD_GET_SDR_INFO       = 0x20," being currently added by emilyshaffer are the same:
https://gerrit.openbmc-project.xyz/#/c/4136/8/sensorhandler.h

(see also original https://github.com/openbmc/openbmc/issues/1089 )

closed 2018-04-11 by rfrandse