openbmc/openbmc #445: Implement DCMI commands

williamspatrick, 2016-06-30:


dkodihal, 2017-06-02:
@tomjoseph83 and I talked about this today. We need to implement commands in the following areas. I've also provided guestimate story points.

- Identification (BMC Id, asset tag, etc), 1 pt.
- Chassis power control and boot options - covered by #440, 2pts.
- Event logging commands - should be mostly covered by the SEL work, so 1 pt placeholder.
- Temperature sensors handling - 2 pt.
- Power Management - 2 pts.

Overall it seems like the DCMI should be an 8 point epic instead of the 5 points that it is now.

williamspatrick, 2017-06-02:
I moved this to Phase 5 because I suspect there is infrastructure pieces we need for xcat (via REST) that would have been done as part of DCMI support.  We need to break this down into backend vs IPMI interface.

closed 2018-08-09 by rfrandse