openbmc/openbmc #446: Implement user management commands

williamspatrick, 2016-06-30:


agangidi53, 2016-07-18:
Are the use management commands referenced here managed via ipmitool ?

@anoo1  some time back indicated that it is inband ipmitool how we are gonna be managing the users but now since ipmi over lan implementation is being considered I want to know if that's going to be changed.

tomjoseph83, 2017-11-16:
IPMI commands related to User account management:-

1) Set user Access command
2) Get user Access command
3) Set user name command
4) Get user name command
5) Set user password command
6) Set user payload access
7) Get user payload access
8) Set channel security access
9) Set channel access
10) Get channel access

rthomaiy, 2017-11-16:
do we have plans on how ipmi will do the user management verification ? It requires clear text password. Are we plan to create a encrypted file in which the clear text password will be stored (say in phosphor-user-manager). 
i.e. d-bus calls to Phosphor-user-manager --> Create new user / change password --> has to update both /etc/shadow, groups file along with a separate encrypted file (to store user-information), from which IPMI will do the authentication ?
Do we have plan of implementation like this or a different one ? 
So other than IPMI. Can we assume other app's say webui / redfish etc. will use PAM authentication directly. 
All apps including IPMI will trigger d-bus method API calls to create / change user / password ?