openbmc/openbmc #450: Unable to Ping on Broadcom 5720 Running OpenBMC

dbuesom, 2016-06-30:
### Problem Description
I am running a PowerVM driver for an embedded Broadcom 5720 on a Barreleye system. The 5720 has both ports configured for BMC NC-SI passthrough. I am attempting to ping with the second port of the adapter, but it reports a transmit timeout. Debugging has shown that data is being written to the transmit buffer and the send ring producer index value is incremented correctly. The timeout occurs because the driver never receives an update that the send consumer index has incremented. Therefore, no packets are being sent out from the adapter onto the network. Also seen is that certain configuration and interrupt registers do not allow their values to be modified. These registers are critical for proper functionality of the driver. They include EMAC Status (0x404), EMAC Mode (0x400) and Host Coalescing Mode (0x3C00). 

#### Logs:
**BCM5720 Port 2 ping test on a Barreleye running OpenBMC**
Status Block:  **<--------- This status block should be updated by host coalescing engine**
000008ac2000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 :................:
000008ac2010: 00 00 00 00 00 00 00 00 :........:

INIT: EMAC Status Reg: Link State Changed OFF

INIT: Status Block: Link State Changed OFF

INIT: EMAC 400: before set 19e04808 **<------ Upper bits here report APE TX/RX paths enabled**

INIT: TX-MAC-STATUS-REG Reports LINK UP

WRITE: current-tdesc: 0 

WRITE: tx-buf virt addr: 89b0000 

WRITE: Buffer 
0000089b0000: ff ff ff ff ff ff 00 01 6c 80 02 2b 08 06 00 01 :........l..+....:
0000089b0010: 08 00 06 04 00 01 00 01 6c 80 02 2b 09 05 5c 6a :........l..+..\j:
0000089b0020: 00 00 00 00 00 00 09 05 5c 6a 00 00 00 00 :........\j....:

WRITE: tdesc virt addr: 8ac0000 

WRITE: tdesc (tx-buf) phys addr: 13000 

WRITE: tdesc flags: 84 

WRITE: tdesc length: 2e 

WRITE: After MB Update: 

WRITE: curr-prod-idx (mb value): 1 

WRITE: current-tdesc: 1 

WRITE: send cons (from reg x3cc0): 0 **<------- Send consumer index never updates**
WRITE: Transmit timeout **<--------- Timeout set to arbitrary value of 5 secs**

WRITE: Status Block
000008ac2000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 :................:
000008ac2010: 00 00 00 00 00 00 00 00 :........:

**BCM5720 Port 2 ping test on a POWER7 _not running_ OpenBMC**
Status Block: **<------- status block updated, link status changed, and enabled tagging**
000008ad1000: 00 00 00 03 00 00 00 01 00 00 00 00 00 00 00 00 :................:
000008ad1010: 00 00 00 00 00 00 00 00 :........:

INIT: EMAC Status Reg: Link State Changed ON

INIT: Status Block: Link State Changed ON

INIT: EMAC 400: before set e04804 **<------ Upper bits for APE TX/RX paths are disabled**

INIT: TX-MAC-STATUS-REG Reports LINK UP

WRITE: current-tdesc: 0 

WRITE: tx-buf virt addr: 8a80000 

WRITE: Buffer 
000008a80000: ff ff ff ff ff ff 00 0a f7 94 7e 2f 08 06 00 01 :..........~/....:
000008a80010: 08 00 06 04 00 01 00 0a f7 94 7e 2f 09 03 b5 c9 :..........~/....:
000008a80020: 00 00 00 00 00 00 09 03 b5 c9 00 00 00 00 :..............:

WRITE: tdesc virt addr: 8ac7000 

WRITE: tdesc (tx-buf) phys addr: 13000 

WRITE: tdesc flags: 84 

WRITE: tdesc length: 2e 

WRITE: After MB Update: 

WRITE: curr-prod-idx (mb value): 1 

WRITE: current-tdesc: 1 

WRITE: send cons (from reg x3cc0): 1 **<------ send consumer index updates properly here**

WRITE: Status Block
000008ad1000: 00 00 00 03 00 00 00 04 00 02 00 00 00 00 00 00 :................:
000008ad1010: 00 01 00 02 00 00 00 00 :........:

**Eventually leads to ping success**

#### Q's:

1. What initialization steps does the OpenBMC driver for 5720 follow? I could not find in the specification where it discussed enabling APE and how to properly route HTX2B (Host TX to BMC/APE) and B2HRX (BMC/APE to Host RX) packets.

2. The OpenBMC driver is running before our 5720 driver is loaded, does the OpenBMC driver disable certain registers after loading?

agangidi53, 2016-07-06:
so @dbuesom   You ping the BMC on BCM port 2 and you see a failure ? I was presuming that port 2 was deliberately disabled to work with BMC so that it can be host dedicated . Is that wrong ?

dbuesom, 2016-07-06:
Hi @agangidi53 thanks for your response. I am not pinging the BMC. I am on a partition but at the Open Firmware prompt (not yet at the OS). From there, I configure port 2 of the BCM 5720 to ping a NIM server. The ping fails.  It is odd because when you boot the partition to the OS (Linux in this case) and configure the port, the ping succeeds. It may be that the BCM 5720 Linux driver accounts for something that we are missing when BMC is operational.

In your response, what did you mean by "host dedicated"? Is the host BMC, Linux, or something else?

williamspatrick, 2016-07-14:
@shenki , @gwshan - You guys asked for this issue to be opened but then haven't participated.  Can you give @dbuesom some feedback on his questions, please?

shenki, 2016-07-14:
I don't recall asking for this bug to be opened.

In answer to the questions:

1) The initialisation is contained in the NCSI stack here https://github.com/openbmc/linux/blob/dev-4.6/net/ncsi/

2) OpenBMC does not disable any registers after loading that I'm aware of

dbuesom, 2016-07-14:
Hi @williamspatrick thanks for drawing attention to this issue. It is starting to gain visibility so I happened to message @shenki this morning on Slack with more background and the same questions.

@shenki did mention the NCSI Spec v1. I will start looking over and see if anything jumps out at me. Unfortunately I cannot look at the initialization NCSI code pointed to at above. I could possibly look at pseudo-code if any exists. 

Some of the background I mentioned to @shenki:
- The driver has previously worked on a 5720, but with BMC NCSI disabled
- Booting a partition to Linux and using the 5720 works (on Barreleye with BMC)
- I asked the Linux team if their 5720 driver does something to account for BMC, it appears to them that they do not

mdmillerii, 2016-07-14:
I am not familiar with the driver but I was able to find a few things browsing this code:

I verified the Linux watchdog for tx timeout on this device is 5 seconds.

It appears the tg3 driver is using the 0x50 byte memory status block to determine the producer and consumer indices.  In fact in IRQ (not MSI) mode the driver polls PCISTATE (0x70) bit mask 2 (INT_NOT_ACTIVE) to flush the hw_status block write to memory.  I find the register offset defined but don't see an access.

I see reset_hw writes HOSTCC_MODE (0x3c00) to 0 then polls every 10us up to 2000 times for ENABLE (bit mask 0x2) to clear, then programs various settings in that register.

Otherwise I would suggest working with Broadcom.

gwshan, 2016-07-15:
I will take a look next week when getting chance. I was busy on v2 NCSI
patches this week
and didn't get time for it, sorry about that.

dbuesom, 2016-07-18:
Hi @mdmillerii, thanks for your comment. Our timeout is also set to 5 seconds. We also cannot support interrupts in Open Firmware, so therefore cannot follow the interrupt flow you mentioned. This driver does follow the same procedure for configuring the host coalescing mode engine that you state later. That configuration takes place in initialization.  

I did read through the NCSI spec. The NCSI commands I saw setup basic configuration and should not be causing the problems I am seeing. I am trying to get in touch with Broadcom. It could that something is going on behind the scenes after setting up BMC that is causing issues (hardware arbitration, etc). I would like to perform one quick test, as mentioned above I noticed that before PHYP loads our driver, both of the BCM 5720 ports are configured and enabled for BMC. Could I possibly get a sandbox from @gwshan or @shenki, where only port 1 for example has BMC passthrough enabled?

shenki, 2016-11-16:
I think this was resolved.

closed 2016-11-16 by shenki

dbuesom, 2016-11-18:
Hi @shenki, yes the issue was resolved after further discussions with Broadcom. We had to implement mutex locks in the APE region (not mentioned in the Broadcom 5720 specification) in order to handle contention of shared resources with the PowerVM FCode driver and OpenBMC's NCSI driver.