openbmc/openbmc #452: Tag:V1: Blink fast the heartbeet LED, Blink Fast the Beep LED fails intermittently

sivassrr, 2016-07-01:
Failure Case: Blink Fast the Beep LED:
KEYWORD BuiltIn . Should Be True ${ON_VALUE} > ${MIN_TOGGLE_VALUE} and ${OFF_VALUE} > ${MIN_TOGGLE_VALUE}
Documentation: 	

Fails if the given condition is not true.
Start / End / Elapsed: 	20160701 09:54:19.422 / 20160701 09:54:19.425 / 00:00:00.003
09:54:19.425 	FAIL 	'5 > 0 and 0 > 0' should be true.

Pass Case:
Documentation: 	

Sets variable based on the given condition.
Start / End / Elapsed: 	20160701 10:48:15.225 / 20160701 10:48:15.228 / 00:00:00.003
10:48:15.228 	INFO 	${OFF_VALUE} = 4 	
00:00:00.003 KEYWORD BuiltIn . Should Be True ${ON_VALUE} > ${MIN_TOGGLE_VALUE} and ${OFF_VALUE} > ${MIN_TOGGLE_VALUE}
Documentation: 	

Fails if the given condition is not true.
Start / End / Elapsed: 	20160701 10:48:15.228 / 20160701 10:48:15.231 / 00:00:00.003

sivassrr, 2016-07-01:
Pass Case: Blink Fast Heartbeat LED:
Documentation: 	

Sets variable based on the given condition.
Start / End / Elapsed: 	20160701 10:47:10.873 / 20160701 10:47:10.876 / 00:00:00.003
10:47:10.876 	INFO 	${ON_VALUE} = 3 	
00:00:00.003 KEYWORD ${OFF_VALUE} = BuiltIn . Set Variable If '${ledstate}'=='Off', ${OFF_VALUE + 1}, ${OFF_VALUE}
Documentation: 	

Sets variable based on the given condition.
Start / End / Elapsed: 	20160701 10:47:10.876 / 20160701 10:47:10.879 / 00:00:00.003
10:47:10.879 	INFO 	${OFF_VALUE} = 2 	
00:00:00.002 KEYWORD BuiltIn . Should Be True ${ON_VALUE} > ${MIN_TOGGLE_VALUE} and ${OFF_VALUE} > ${MIN_TOGGLE_VALUE}
Documentation: 	

Fails if the given condition is not true.
Start / End / Elapsed: 	20160701 10:47:10.880 / 20160701 10:47:10.882 / 00:00:00.002

Fail case: Blink Fast the Heartbeet LED
Documentation: 	

Sets variable based on the given condition.
Start / End / Elapsed: 	20160701 09:53:14.988 / 20160701 09:53:14.991 / 00:00:00.003
09:53:14.991 	INFO 	${ON_VALUE} = 5 	
00:00:00.003 KEYWORD ${OFF_VALUE} = BuiltIn . Set Variable If '${ledstate}'=='Off', ${OFF_VALUE + 1}, ${OFF_VALUE}
Documentation: 	

Sets variable based on the given condition.
Start / End / Elapsed: 	20160701 09:53:14.991 / 20160701 09:53:14.994 / 00:00:00.003
09:53:14.994 	INFO 	${OFF_VALUE} = 0 	
00:00:00.003 KEYWORD BuiltIn . Should Be True ${ON_VALUE} > ${MIN_TOGGLE_VALUE} and ${OFF_VALUE} > ${MIN_TOGGLE_VALUE}
Documentation: 	

Fails if the given condition is not true.
Start / End / Elapsed: 	20160701 09:53:14.994 / 20160701 09:53:14.997 / 00:00:00.003
09:53:14.997 	FAIL 	'5 > 0 and 0 > 0' should be true.

sivassrr, 2016-07-01:
OpenBMC Version Detail:
root@barreleye:~# cat /etc/version
20160630213138
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="0.1.0 (krogoth)"
VERSION_ID="v1.0-48-g8d43aee-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (krogoth)"
BUILD_ID="v1.0"
root@barreleye:~#

williamspatrick, 2016-07-01:
@sivassrr - We need a human understandable issue report here.  This looks like output from Robot to me and I really have no idea what to even do with this.  Can you please give an explanation of what you are trying to do, what your expected result is, and what you are seeing?

gkeishin, 2016-07-02:
Let me help decode this in simpler terms

Robot output:
09:54:19.425 FAIL '5 > 0 and 0 > 0' should be true.
Test code :
should be true   ${ON_VALUE} > ${MIN_TOGGLE_VALUE} and ${OFF_VALUE} > ${MIN_TOGGLE_VALUE}

When the test does a sampling to get the blink led stats, it couldn't get back expected blink expectation.. This means that when the test samples for 5 blinks, the toggle value turns out to be 0 when the test expect 5 toggling blinking.

This is anyway not a good way to test LED's... From the test automation point of view, we can't verify if the LEDs are blinking or not unless someone verifies it in person.

In similar line, there is a defect in discussion going on here https://github.com/openbmc/openbmc-test-automation/issues/22

Hope that helps..

williamspatrick, 2016-07-05:
@gkeishin - We should open a feature for the LED control code to give you the blink state.  'On' , 'Off', 'Blink'.  For the blink we should also have a duty-cycle attribute to tell you what the current blink rate is.  We probably have a few other attributes needed (like color).

sivassrr, 2016-09-26:
@gkeishin We have bink fast and slow already in our test automation, right

gkeishin, 2016-09-26:
This needs the design to be changed from the LED dev code.. @vishwabmc  would be working on this ahead , that much I know about it.

rfrandse, 2017-04-07:
Implemented in current code

closed 2017-04-07 by rfrandse