openbmc/openbmc #453: Tag:V1: Firmware version populated with "\"

sivassrr, 2016-07-01:
Firmware Version TC failing because of following value. 

Convert a string to a JSON object
Start / End / Elapsed: 	20160701 10:43:12.416 / 20160701 10:43:12.416 / 00:00:00.000
10:43:12.416 	INFO 	To JSON using : content={
  "data": {
    "fault": "False", 
    "fru_type": "BMC", 
    "is_fru": 0, 
    "manufacturer": "ASPEED", 
    "present": "False", 
    "version": "\"v1.0-48-g8d43aee-dirty\""   <======
  }, 
  "message": "200 OK", 
  "status": "ok"
} 

root@barreleye:~# cat /etc/version
20160630213138
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="0.1.0 (krogoth)"
VERSION_ID="v1.0-48-g8d43aee-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (krogoth)"
BUILD_ID="v1.0"
root@barreleye:~#

============================

Previous Version of OpenBMC it used to pass and the output is as follows:

17:17:15.853	INFO	To JSON using : content={
  "data": {
    "fault": "False", 
    "fru_type": "BMC", 
    "is_fru": 0, 
    "manufacturer": "ASPEED", 
    "present": "False", 
    "version": "v1.0-11-gf81616d-dirty"
  }, 
  "message": "200 OK", 
  "status": "ok"

williamspatrick, 2016-07-01:
Something wrong with the rest server here?

bradbishop, 2016-07-11:
The dbus object itself has the extra quotes.

williamspatrick, 2016-07-11:
@bradbishop - Do you want to keep doing the investigation or should we move it along to someone else?

gkeishin, 2016-07-21:
https://github.com/openbmc/openbmc-build-scripts/issues/10

closed 2016-08-11 by williamspatrick