openbmc/openbmc #455: Feature Request: Exporting VGA console on BMC

agangidi53, 2016-07-06:
A lot of Barreleye / OpenBMC users seem to run into an issue where they try to install the OS, where they can't see the OS install menu after kexec'ing from Petitboot via the VGA. This is primarily because of missing VGA drivers for OS installers.

To counter this: Can we export the BMC console log to VGA console ? That way users using VGA would see pretty much the same output as the "hconsole". Since ASPEED chip has the VGA controller I believe it's possible to get this to work from the BMC end.

shenki, 2016-07-06:
Paging @jk-ozlabs

sammj, 2016-08-15:
Has this been tested on a vaguely recent version of Petitboot? Since https://github.com/open-power/petitboot/commit/91ba67605de8ff5f15c335db4330594e07b5317e Petitboot updates the device-tree so that the booting kernel can inherit the existing framebuffer if it does not have the driver. I haven't tested this on an OpenBMC machine but I can't think of a reason it wouldn't also work here.

geissonator, 2018-05-23:
This has all been merged into master as of a few weeks ago.

agangidi53, 2018-05-31:
@geissonator  Can you please point to the commits that implemented this ?  Appreciate the help.

bojunyao, 2022-09-20:
@shenki Hi, where is the link to submit this issue? thanks~