openbmc/openbmc #457: Tag V1:20160707143158:Barreleye:Build141: Host is in booting state for more than 18 minutes

sivassrr, 2016-07-09:
root@barreleye:/etc# obmcutil state
= HOST_BOOTING
root@barreleye:/etc# obmcutil bootprogress
units =
value = FW Progress, Baseboard Init
error = 0
root@barreleye:/etc# date
Sat Jul 9 18:12:29 UTC 2016
root@barreleye:/etc# date; obmcutil bootprogress; date
Sat Jul 9 18:12:46 UTC 2016
units =
value = FW Progress, Baseboard Init
error = 0
Sat Jul 9 18:12:48 UTC 2016

root@barreleye:/etc# obmcutil state; obmcutil bootprogress;date
= HOST_BOOTING
units =
value = FW Progress, Baseboard Init
error = 0
Sat Jul 9 18:31:19 UTC 2016
root@barreleye:/etc#
[journaloutput.txt](https://github.com/openbmc/openbmc/files/355687/journaloutput.txt)

sivassrr, 2016-07-09:
Because of this, Both Inventory and OCC PowerCAP related TCs are failing.

sivassrr, 2016-07-10:
on the same system left overnight to boot, still at the same state. 
root@barreleye:~# cat /etc/version
20160707143158
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="0.1.0 (krogoth)"
VERSION_ID="v1.0-67-g2ad67f6-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (krogoth)"
BUILD_ID="v1.0"
root@barreleye:~# obmcutil state
 = HOST_BOOTING
root@barreleye:~# date
Sun Jul 10 04:42:35 UTC 2016
root@barreleye:~# obmcutil bootprogress
units =
value = FW Progress, Baseboard Init
error = 0
root@barreleye:~# journactl -f
-sh: journactl: command not found
root@barreleye:~# journalctl -f
-- Logs begin at Sat 2016-07-09 18:46:01 UTC. --
Jul 10 04:42:03 barreleye system_manager.py[927]: Running System State: HOST_BOOTING
Jul 10 04:42:03 barreleye system_manager.py[927]: GPIO Lookup:  MEZZ0_PRESENT = 432,in
Jul 10 04:42:03 barreleye system_manager.py[927]: Checking Presence: MEZZ1_PRESENT
Jul 10 04:42:03 barreleye system_manager.py[927]: GPIO Lookup:  MEZZ1_PRESENT = 433,in
Jul 10 04:42:08 barreleye system_manager.py[927]: Setting watchdog poll interval to: 120000
Jul 10 04:42:08 barreleye system_manager.py[927]: Stopping watchdog
Jul 10 04:42:08 barreleye system_manager.py[927]: Starting watchdog with poll interval: 120000
Jul 10 04:42:12 barreleye dropbear[3022]: Password auth succeeded for 'root' from 9.199.43.98:52897
Jul 10 04:42:14 barreleye dropbear[3027]: lastlog_perform_login: Couldn't stat /var/log/lastlog: No such file or directory
Jul 10 04:42:14 barreleye dropbear[3027]: lastlog_openseek: /var/log/lastlog is not a file or directory!

mdmillerii, 2016-07-11:
This seems to be a dup of openbmc/openbmc-test-automation#38 which has a full journal posted.

sivassrr, 2016-07-11:
@mdmillerii openbmc/openbmc-test-automation#38 is for Test Automation update, Initially thought our test code waiting time of boot to host partition is not enough. But when I tried multiple time with reboot, found it as openbmc issue. In case if the boot time changes then will use openbmc/openbmc-test-automation#38  for test code changes.

williamspatrick, 2016-07-11:
See my comments in openbmc/openbmc-test-automation#38 .

closed 2016-07-11 by williamspatrick