openbmc/openbmc #466: i2c debug tool - Need 4 byte read support.

williamspatrick, 2016-07-11:
The Witherspoon lab team was using the 'i2c-tools' utilities we have installed and they are insufficient for one particular chip they would like to interact with.

The chip in question requires a 4 byte write to update an offset register within the chip, followed by a 4 byte read to read the register.  Most i2c devices are a single byte write and this seems to be all that `i2cdump` supports.  There is no way to do the following transaction(s):

```
[start] {slave-addr} {write0} {write1} {write2} {write3} [stop]
[start] {slave-addr} {read0} {read1} {read2} {read3} [stop]
```

The first transaction can be emulated today with `i2cset`.  The second transaction cannot.

Examining the code for i2cdump the following appears to be the behavior.
The 'c' mode in `i2cdump` will perform:

```
[start] {slave-addr} {write-reg} [stop]
[start] {slave-addr} {read0} [stop]
...
[start] {slave-addr} {read3} [stop]
```

The 'b' mode in `i2cdump` will perform the same as 'c' except for it will not do the first write of the register-offset.  It still performs N unique start/stop sequences for reading N bytes.

We need to either add an enhancement to `i2cdump` or find another tool.  One option is to opensource and port the i2cmaster tool from the FSP code base but it currently works against the FSP i2c subsystem instead of the "standard" one.

williamspatrick, 2016-07-11:
@shenki , @bradbishop , @eddiejames - Seeking your input.

paulgrobertson, 2016-07-11:
A read count option (-rc N) for i2cget would be a good enhancement.   Presently i2cget will only do 1 or 2 bytes,  mode b or w.   

i2cget -y -rc N  $bus  $slave

      <slave addr><read><read>...<read(N-1)>    

-OR-  add double-word mode to read 4 bytes.

i2cget -y $bus $slave dw

      <slave addr><read><read><read><read>

bradbishop, 2016-07-12:
My vote would be for adding what we need to the existing OSS tools.

shenki, 2016-07-12:
Is this just for hardware bringup, or do we need it in all systems?

williamspatrick, 2016-07-12:
@shenki - The only reason to manually run the i2c-tools should be for lab / debug.  So yes?

@bradbishop - I would prefer that as well I think.  But, the lm-sensors.org people seem to have dropped off the face of the earth.  The site has been down for going on 6 months.  I went to Wayback to find an IRC channel (#linux-sensors @ freenode) and it is basically dead too.

eddiejames, 2016-07-12:
I think modifying the existing i2c tools makes the most sense. iicmaster is kind of buggy anyway.

bradbishop, 2016-07-13:
> @shenki <https://github.com/shenki> - The only reason to manually run the i2c-tools should be for lab / debug. So yes?
>
> @bradbishop <https://github.com/bradbishop> - I would prefer that as well I [...]

maybe we could ask on the hwmon mailing list if they know what is going on?

williamspatrick, 2016-07-13:
@causten pointed me to this issue:  groeck/lm-sensors#3 .  In that, someone implies that this 'groeck' repository is now the "official" repository but there have not been any commits to that in a year either.

lm-sensors:
`
Latest commit 21e2560  on Aug 31, 2015￼ groeck Add detection of Nuvoton NCT6793D
`
i2c-tools:
`
Latest commit 9726bed  on Jul 7, 2015￼ Jean Delvare eeprog: Drop duplicate CFLAGS ￼…
`

The lm-sensors one doesn't have an svn id so maybe somewhere between July and August of last year they did "transition" over to github.

williamspatrick, 2016-07-13:
I also emailed Ross Burton, who made the Yocto change to point the SRC_URI at yoctoproject.org instead of lm-sensors.org and he replied:

```
Sorry, no.  It appears that the owners of that domain have disappeared and
nobody else has forked the development of lm-sensors.

I guess that makes you the new maintainer! :)

Ross
```

williamspatrick, 2016-08-16:
Code is in gerrit, but waiting on review.

closed 2016-09-06 by williamspatrick

amboar, 2016-09-07:
For the record, groeck is Gunther Roeck, a maintainer of the kernel's hwmon subsystem.