openbmc/openbmc #477: Format to document org.openbmc interfaces.

williamspatrick, 2016-07-15:


causten, 2016-07-20:
Please make sure this issue covers status property values.  We need to have a hardened list of OBMC approved values and what they mean documented.  This allows people to write scripts that can handle responses correctly.  Let's use Code Update as an example.  

You can call this method to prepare a BMC for update... /org/openbmc/control/flash/bmc/action/update 
You then check the status of the action via... /org/openbmc/control/flash/bmc/attr/status  This is a case where you can't just use the method response value for status.  There are processes that we know exceed the http request timeout value hence the need for a status property on an interface.

The value of the status property must come from a well known documented list otherwise scripts don't know what a successful is. It use to return 

'Update Success' or 'Unpack Error'.  Now at the time of this writing is says 
'Deferred for mounted filesystem. reboot BMC to complete'

Nice new text but hard to write user scripts for.  I think standardizing on a fixed list would be better. i.e.
SUCCESS, ERROR, BUSY, RETRY, REBOOT, NOT AUTHORIZED, etc

I'm sure there is a list of values in the computer industry that would be well suited for our needs.

williamspatrick, 2016-10-05:
- [ ] Create example interface specfication.
- [ ] Document file format
- [ ] Add support for enumerations.
- [ ] Add support for error identifiers.
- [ ] Generate markdown documentation similar to existing docs repo.

rfrandse, 2017-07-04:
<https://gerrit.openbmc-project.xyz/5086> docs: document interface and error YAMLs
Resolves  #477 Format to document org.openbmc interfaces.

closed 2017-07-18 by williamspatrick