openbmc/openbmc #479: Sync u-boot uuid to /etc/machine-id

anoo1, 2016-07-18:
In order to be able to keep the system uuid in sync between the vpd value and the fw value, a script is needed to copy the uuid u-boot value to /etc/machine-id. @mdmillerii has the details.

whowutwut, 2017-06-09:
Is this UUID going to be exposed in the ReST commands? Which endpoint and what would it look like? (@zet809 FYI)

anoo1, 2017-06-15:
> Is this UUID going to be exposed in the ReST commands?

@chinaridinesh is working on syncing the uuid and seems he's using xyz.openbmc_project.Inventory.Decorator.UUID. I don't see this interface yet in the phosphor-dbus-interfaces yet, so @chinaridinesh do you know who's adding it?

whowutwut, 2017-06-16:
@williamspatrick  Is this the unique identifier for each node that you had mentioned in the slack conversation?

closed 2019-10-28 by stale[bot]

hughsie, 2021-11-12:
I'm also trying to get fwupd running on the BMC -- and we also need `/etc/machine-id` -- could this issue be revived perhaps?

mdmillerii, 2021-11-12:
I am not sure what the design should be, @hughsie but my understanding of machine id has changed in the last 6 years.   Let's not discuss here.

Please state your requirements and reasoning in an email to the openbmc development list . You could also use a issue on openbmc/openbmc but the community prefers email to raise issues and gerrit to submit a design to docs to discuss specific proposals.