openbmc/openbmc #480: Feature request: Need to populate basic details on PCIe card via BMC inventory

agangidi53, 2016-07-18:
For ironic usage in onMetal kind of use-case, I thought it will be good to have a feature where we can query the PCIe devices via the BMC to figure out what exact PCIe device is in the said PCIe slot . Currently we have only have the ability to know present / not present.

willenlee, 2016-08-01:
**Insert the PCIe riser and PCIe cards then checking SLOT present via REST.**

[image]
[image]

agangidi53, 2016-08-01:
Hi @willenlee 

I am not looking for what you have suggested. I need the DETAILS of what SPECIFIC card is present in the slot.

amboar, 2016-08-02:
@willenlee As a general note it's probably more useful to everybody if you copy/paste the text out of the console rather than attaching screenshots. There are also some [nice formatting options ](https://guides.github.com/features/mastering-markdown/) for github issues to help make the output readable.

willenlee, 2016-08-08:
At present, BMC can access PCIE through I2C bus to get below information.
Or BIOS can get these information then writing fru into BMC.

1. Vendor ID
2. Device ID
3. Revision ID
4. Class Code
[image]

closed 2019-10-27 by stale[bot]