openbmc/openbmc #482: Power on fails - /org/openbmc/control/chassis0 not found

micwalsh, 2016-07-19:
I ran the following test:
*** Settings ***

Resource        ../lib/rest_client.robot
Resource        ../lib/ipmi_client.robot
Resource        ../lib/utils.robot

Suite Setup     Open Connection And Log In
Suite Teardown  Close All Connections

*** Test Cases ***
Sample Test Three
   ${state}=  Get Boot Progress
   Log To Console  State is ${state}.
   Power On Host

And the Power On Host fails with the following...
.URI:https://9.3.23.32/org/openbmc/control/chassis0//action/powerOn, method:Post, args:{u'data': {u'data': []}, u'headers': {u'Content-Type': u'application/json'}}
Response code:404, Content:{
 "data": {
   "description": "org.openbmc.ObjectMapper.Error.NotFound: path or object not found: /org/openbmc/control/chassis0"
 }, 
"message": "404 Not Found", 
 "status": "error"
}
Sample Test Three                                                     | FAIL |
404 != 200

Andrew Geissler, George Keishing seem to think this is a legitimate BMC code problem.

williamspatrick, 2016-07-19:
The LED entries are in the device tree and I see them in the /proc file system.

https://github.com/shenki/linux/blob/dev-4.6/arch/arm/boot/dts/aspeed-bmc-opp-palmetto.dts#L63

```
root@palmetto:/proc/device-tree/leds/identify# pwd
/proc/device-tree/leds/identify
root@palmetto:/proc/device-tree/leds/identify# ls
gpios  name
root@palmetto:/proc/device-tree/leds/identify# hexdump -c name
0000000   i   d   e   n   t   i   f   y  \0                            
0000009
root@palmetto:/proc/device-tree/leds/identify# hexdump gpios 
0000000 0000 0700 0000 0200 0000 0100          
000000c
```

williamspatrick, 2016-07-19:
These LED options are missing in our 4.6 defconfig.

https://github.com/openbmc/linux/blob/dev-4.4/arch/arm/configs/aspeed_defconfig#L116
https://github.com/shenki/linux/blob/dev-4.6/arch/arm/configs/aspeed_g4_defconfig

shenki, 2016-07-20:
Those defconfig in the kernel tree are used for kernel development. While I would prefer we use a in-kernel tree defconfig, we are using a defconfig from the openbmc tree for the yocto builds. It lives here, and has the LED drivers enabled:

 https://github.com/openbmc/openbmc/blob/master/meta-openbmc-bsp/meta-aspeed/meta-ast2400/recipes-kernel/linux/linux-obmc/defconfig#L120

williamspatrick, 2016-07-20:
Any idea why LEDs are not working with 4.6 then?

shenki, 2016-07-20:
Not sure.

Is there anything in `/sys/class/leds/`?

Anything in the kernel logs about gpio or leds?

williamspatrick, 2016-07-20:
@shenki - Every Barreleye and Palmetto is failing the same way with your latest kernel.  Please claim one of the machines and take a look.  I did not see anything obvious in dmesg.  As I posted earlier `/sys/class/leds` is empty.

shenki, 2016-07-20:
I just did a build. It looks like the correct options are enabled in the kernel config:

```
CONFIG_NEW_LEDS=y
CONFIG_LEDS_CLASS=y
CONFIG_LEDS_GPIO=y
```
I will do some testing on a local machine to work out why the gpio-leds driver is not being loaded.

shenki, 2016-07-21:
The call chain looks like this:

1. `gpio_led_probe`
2. `gpio_leds_create`
3. `devm_get_gpiod_from_child`
4. `fwnode_get_named_gpiod`
5. `gpiod_request`
6. `chip->request(chip, gpio_chip_hwgpio(desc))`

request fails with status -517

shenki, 2016-07-21:
`->request` is `pinctrl_request_gpio`. I suspect we have a partially integrated pinctrl driver that does not have the GPIOs set up.

@amboar, can you confirm this theory? We're looking at the dev-4.6 branch.

shenki, 2016-07-21:
I reverted the half-baked pinmuxification of the gpio driver for now. We will move to pinmux when we rebase on 4.7 next week.

amboar, 2016-07-21:
@shenki I didn't follow the development of the 4.6 branch closely but if you backported the pinmux patches from dev-4.7 onto 4.6 then yes, you likely have a half-baked pinmux implementation that might not support the required pins. If reverting "the half baked pinmuxification of the gpio driver" means removing the pinmux callbacks, that should be fine as long as we're not using GPIO on pins multiplexed with ROM, RG?MII, VGA or the ADC, or if we are, that we are whacking the pinmux registers in the board file to configure GPIO explicitly.

closed 2016-07-22 by williamspatrick