openbmc/openbmc #483: DISTRO_VERSION should match our tag

shenki, 2016-07-20:
yocto has a DISTRO_VERSION variable that is currently set to 0.1.0. I believe this is what we see when logging into a BMC.

```
DISTRO_VERSION    = "0.1.0"
```

It would be preferable if we kept it in sync with our tags, ie, it would currently show v1.0.3+.

agangidi53, 2016-07-20:
+1 on this. It's been a source of some confusion internally at RAX too.

williamspatrick, 2016-07-27:
```
VERSION="0.1.0 (krogoth)"
VERSION_ID="v0.6-157-gd011f68-dirty"
```

Lets make VERSION="v0.6-157".

williamspatrick, 2016-07-27:
Need to backport this to the v1.0.x stream and also put into master.

amboar, 2016-07-29:
> Lets make VERSION="v0.6-157".

This might reflect my lack of yocto knowledge, but where does that come from? Wasn't @shenki  discussing DISTRO_VERSION rather than VERSION? Shouldn't we be using the tag value? If we're talking about tag values we shouldn't be backporting anywhere as the patch will be different wherever we apply it.

shenki, 2016-07-29:
+1 for tag value

williamspatrick, 2016-07-29:
DISTRO_VERSION in Yocto ends up being VERSION out of /etc/os-release.  That was where I copied those values from.

We already have another variable that uses git-magic to determine the tag.  We'll be doing the same here.  I don't want to have to manually update the DISTRO_VERSION with a commit before each tag.  So, we'll backport the support for this variable to be determined from the same git commands as the other one.

amboar, 2016-07-29:
@williamspatrick right, that makes more sense.

> Lets make VERSION="v0.6-157"

So just to clear up the quote, was it instead (roughly) saying "VERSION should hold the value that is in VERSION_ID" in the abstract, rather than the specific v0.6-157 value (as that doesn't seem to make much sense)?

closed 2016-08-05 by williamspatrick