openbmc/openbmc #485: Sometimes identify led doesn't turn off when press power button to power off system

willenlee, 2016-07-27:
**Test Procedure**
1. Add debug message in file led_controller.c.
2. Enable journalctl -f on bmc console.
3. Press power button to turn on the system then pressing power button to turn off system.
4. We will see the identify led still on but the system is off.
5. Check debug message, we can see when system power off then identify led off, but after 2 secs identify led on and no any "Running System State: HOST_POWERED_ON" message.

>diff` --git a/ledctl/led_controller.c b/ledctl/led_controller.c
>index 9eda9ee..b6621ea 100644
>--- a/ledctl/led_controller.c
>+++ b/ledctl/led_controller.c
> 
>@@ -167,6 +166,7 @@ led_stable_state_function(const char *led_name, const char *led_function)
>        /* Generic error reporter. */
>        int rc = -1;
>
>\+        fprintf(stderr,"led_stable_state_function led_name:[%s] led_function [%s]\n",led_name,led_function);
>        const char *value = NULL;
>        if(strcmp(led_function, "setOff") == 0)
>        {
>

[26/07/16 - 18:01:42:747] Jul 19 04:03:37 barreleye system_manager.py[961]: Turn off power
[26/07/16 - 18:01:42:801] Jul 19 04:03:37 barreleye system_manager.py[961]: Stopping watchdog
[26/07/16 - 18:01:42:876] Jul 19 04:03:37 barreleye system_manager.py[961]: **Running System State: HOST_POWERING_OFF**
[26/07/16 - 18:01:44:172] Jul 19 04:03:39 barreleye system_manager.py[961]: **Running System State: HOST_POWERED_OFF**
[26/07/16 - 18:01:44:253] Jul 19 04:03:39 barreleye system_manager.py[961]: **led_stable_state_function led_name:[identify] led_function:[setOff]**
[26/07/16 - 18:01:44:301] Jul 19 04:03:39 barreleye system_manager.py[961]: Deleting OCC device
[26/07/16 - 18:01:44:348] Jul 19 04:03:39 barreleye system_manager.py[961]: Deleting OCC device
[26/07/16 - 18:01:44:490] ftgmac100: NCSI interface up
[26/07/16 - 18:01:44:533] Jul 19 04:03:39 barreleye kernel: ftgmac100: NCSI interface up
[26/07/16 - 18:01:44:538] Jul 19 04:03:39 barreleye kernel[960]: ftgmac100: NCSI interface up
[26/07/16 - 18:01:44:890] Jul 19 04:03:39 barreleye system_manager.py[961]: ERROR (name_properties): "name" property in /bmc/inventory/hdd_backplane is incorrect ("BARRELEYE Exander" instead of base node name)
[26/07/16 - 18:01:44:939] Jul 19 04:03:40 barreleye system_manager.py[961]: ERROR (name_properties): "name" property in /bmc/inventory/membuf0 is incorrect ("0000000000000000" instead of base node name)
[26/07/16 - 18:01:45:003] Jul 19 04:03:40 barreleye system_manager.py[961]: ERROR (name_properties): "name" property in /bmc/inventory/membuf3 is incorrect ("0000000000000000" instead of base node name)
[26/07/16 - 18:01:45:053] Jul 19 04:03:40 barreleye system_manager.py[961]: ERROR (name_properties): "name" property in /bmc/inventory/dimm4 is incorrect ("0x0b" instead of base node name)
[26/07/16 - 18:01:45:100] Jul 19 04:03:40 barreleye system_manager.py[961]: ERROR (name_properties): "name" property in /bmc/inventory/dimm5 is incorrect ("0x0b" instead of base node name)
[26/07/16 - 18:01:45:148] Jul 19 04:03:40 barreleye system_manager.py[961]: ERROR (name_properties): "name" property in /bmc/inventory/dimm0 is incorrect ("0x0b" instead of base node name)
[26/07/16 - 18:01:45:159] Jul 19 04:03:40 barreleye system_manager.py[961]: ERROR (name_properties): "name" property in /bmc/inventory/dimm1 is incorrect ("0x0b" instead of base node name)
[26/07/16 - 18:01:45:204] Jul 19 04:03:40 barreleye system_manager.py[961]: ERROR (name_properties): "name" property in /bmc/inventory/bios is incorrect ("OpenPOWER Firmware" instead of base node name)
[26/07/16 - 18:01:45:250] Jul 19 04:03:40 barreleye system_manager.py[961]: ERROR (name_properties): "name" property in /bmc/inventory/membuf2 is incorrect ("0000000000000000" instead of base node name)
[26/07/16 - 18:01:45:333] Jul 19 04:03:40 barreleye system_manager.py[961]: ERROR (name_properties): "name" property in /bmc/inventory/cpu0 is incorrect ("PROCESSOR MODULE" instead of base node name)
[26/07/16 - 18:01:45:341] Jul 19 04:03:40 barreleye system_manager.py[961]: ERROR (name_properties): "name" property in /bmc/inventory/sas_expander is incorrect ("BARRELEYE BPx15" instead of base node name)
[26/07/16 - 18:01:45:392] Jul 19 04:03:40 barreleye system_manager.py[961]: ERROR (name_properties): "name" property in /bmc/inventory/misc is incorrect ("0000000000000000" instead of base node name)
[26/07/16 - 18:01:45:437] Jul 19 04:03:40 barreleye system_manager.py[961]: ERROR (name_properties): "name" property in /bmc/inventory/membuf1 is incorrect ("0000000000000000" instead of base node name)
[26/07/16 - 18:01:45:475] Jul 19 04:03:40 barreleye system_manager.py[961]: ERROR: Input tree has errors, aborting (use -f to force output)
[26/07/16 - 18:01:45:520] Jul 19 04:03:40 barreleye system_manager.py[961]: About to erase 0x01e75000..0x01e7e000 !
[26/07/16 - 18:01:45:569] Jul 19 04:03:40 barreleye system_manager.py[961]: Erasing...
[26/07/16 - 18:01:45:572] Jul 19 04:03:40 barreleye system_manager.py[961]: [569B blob data]
[26/07/16 - 18:01:45:576] Jul 19 04:03:40 barreleye system_manager.py[961]: About to program "/tmp/tmpLQYzWC" at 0x01e75000..0x01e7e000 !
[26/07/16 - 18:01:45:617] Jul 19 04:03:40 barreleye system_manager.py[961]: Programming & Verifying...
[26/07/16 - 18:01:45:624] Jul 19 04:03:40 barreleye system_manager.py[961]: [                                                  ] 0%
[26/07/16 - 18:01:45:662] Jul 19 04:03:40 barreleye system_manager.py[961]: Updating actual size in partition header...
[26/07/16 - 18:01:45:670] Jul 19 04:03:40 barreleye system_manager.py[961]: Uploading inventory to PNOR in dtb format...
[26/07/16 - 18:01:45:713] Jul 19 04:03:40 barreleye system_manager.py[961]: Running System State: INVENTORY_UPLOADED
[26/07/16 - 18:01:45:759] Jul 19 04:03:40 barreleye system_manager.py[961]: Starting process: /usr/sbin/pcie_slot_present.exe: pcie_present
[26/07/16 - 18:01:45:801] Jul 19 04:03:40 barreleye system_manager.py[961]: Setting fans to max
[26/07/16 - 18:01:45:939] Jul 19 04:03:41 barreleye system_manager.py[961]: Booting host
[26/07/16 - 18:01:46:035] Jul 19 04:03:41 barreleye system_manager.py[961]: **led_stable_state_function led_name:[identify] led_function:[setOn]**
[26/07/16 - 18:01:46:259] Jul 19 04:03:41 barreleye system_manager.py[961]: Checking Presence: SLOT0_RISER_PRESENT
[26/07/16 - 18:01:46:308] Jul 19 04:03:41 barreleye system_manager.py[961]: GPIO Lookup:  SLOT0_RISER_PRESENT = 424,in
[26/07/16 - 18:01:46:387] Jul 19 04:03:41 barreleye system_manager.py[961]: Checking Presence: SLOT1_RISER_PRESENT
[26/07/16 - 18:01:46:515] Jul 19 04:03:41 barreleye system_manager.py[961]: GPIO Lookup:  SLOT1_RISER_PRESENT = 425,in
[26/07/16 - 18:01:46:564] Jul 19 04:03:41 barreleye system_manager.py[961]: Checking Presence: SLOT2_RISER_PRESENT
[26/07/16 - 18:01:46:674] Jul 19 04:03:41 barreleye system_manager.py[961]: GPIO Lookup:  SLOT2_RISER_PRESENT = 426,in
[26/07/16 - 18:01:46:755] Jul 19 04:03:41 barreleye system_manager.py[961]: Checking Presence: SLOT0_PRESENT
[26/07/16 - 18:01:46:865] Jul 19 04:03:41 barreleye system_manager.py[961]: GPIO Lookup:  SLOT0_PRESENT = 427,in
[26/07/16 - 18:01:46:930] Jul 19 04:03:42 barreleye system_manager.py[961]: Checking Presence: SLOT1_PRESENT
[26/07/16 - 18:01:47:041] Jul 19 04:03:42 barreleye system_manager.py[961]: GPIO Lookup:  SLOT1_PRESENT = 428,in
[26/07/16 - 18:01:47:089] Jul 19 04:03:42 barreleye system_manager.py[961]: Checking Presence: SLOT2_PRESENT
[26/07/16 - 18:01:47:233] Jul 19 04:03:42 barreleye system_manager.py[961]: Using primary side of the bios flash
[26/07/16 - 18:01:47:270] Jul 19 04:03:42 barreleye system_manager.py[961]: GPIO Lookup:  SLOT2_PRESENT = 429,in
[26/07/16 - 18:01:47:320] Jul 19 04:03:42 barreleye system_manager.py[961]: Checking Presence: MEZZ0_PRESENT
[26/07/16 - 18:01:47:422] Jul 19 04:03:42 barreleye system_manager.py[961]: GPIO Lookup:  MEZZ0_PRESENT = 432,in
[26/07/16 - 18:01:47:486] Jul 19 04:03:42 barreleye system_manager.py[961]: Checking Presence: MEZZ1_PRESENT
[26/07/16 - 18:01:47:598] Jul 19 04:03:42 barreleye system_manager.py[961]: Running System State: HOST_BOOTING
[26/07/16 - 18:01:47:646] Jul 19 04:03:42 barreleye system_manager.py[961]: GPIO Lookup:  MEZZ1_PRESENT = 433,in

anoo1, 2016-08-10:
After the power off, the system immediately powers back on, but as posted, there's no 'power on' trace, which indicates the system performed a 'reboot' instead of a 'power off':

[27/07/16 - 15:34:14:942] Jul 19 03:56:34 barreleye system_manager.py[968]: Running System State: HOST_POWERED_OFF
[27/07/16 - 15:34:24:054] Jul 19 03:56:43 barreleye system_manager.py[968]: Booting host

This could happen if the power button was pushed for over 3 seconds (long press), which will trigger a reboot instead of power off.
Could that be the case here, that when attempting to power off the system, inadvertently a long press of the power button was issued?

anoo1, 2016-08-11:
Having some discussion on the power button behavior. We might change it to do a power off instead of a reboot so that the behavior is the same as x86 and other BMCs.

closed 2016-08-17 by williamspatrick

anoo1, 2016-08-17:
Per agreement, the behavior was changed on the stable branch to:
Short press: Remove the power off action, only Power on with the short press.
Long press (>3 seconds): Hard power off.

johnhcwang, 2016-08-18:
Hi @anoo1 , may I know why we need to remove the power off action while short press?
This behavior is not like as the industry norm of "Short press triggers a soft power off, long press triggers a hard power off".

williamspatrick, 2016-08-18:
@johnhcwang - This was the requested behavior from @KennethWilke .  I thought your team was involved in that decision?

anoo1, 2016-08-18:
Correct, on Monday's call, RAX team (Aaron, Adi, Kenneth) requested this behavior. Max Tsai represented FOX team on the call, please check with him for confirmation.

johnhcwang, 2016-08-18:
Sorry that I probably miss this information. Thanks for your clarification.

anoo1, 2016-08-18:
@johnhcwang we understand this is a rax-specific request and not fully industry standard, so we'll plan on a future feature to make the button action configurable via a setting or property so that the default behavior can be changed for other systems.

agangidi53, 2016-08-18:
Hi john . This was in fact Rax request , specifically Aaron's request .  Patrick / Adriana . Thanks for working through the confusion