openbmc/openbmc #486: Getting system inventory fails

shenki, 2016-07-27:
According the [REST API document](https://github.com/openbmc/docs/blob/master/rest-api.md#http-get-operations--url-structure) when we do a get on `/org/openbmc/inventory/system` we should get back the version information. I find this very useful for retrieving the PNOR version info.

However, on my Palmetto, this does not work:

```
$ curl -b cjar -k https://palm5-bmc//org/openbmc/inventory/system
{
  "data": {
    "fault": "False", 
    "fru_type": "SYSTEM", 
    "is_fru": 1, 
    "present": "False", 
    "version": ""
  }, 
  "message": "200 OK", 
  "status": "ok"
```

I'm running v1.0-87-g542d3a5.

williamspatrick, 2016-07-27:
There was an issue with this but it was fixed by 75dcd9e3 , which you seem to have.  Are you sure you booted the machine once?  You have to boot the machine so that Hostboot pushes down this inventory object.

shenki, 2016-07-27:
No, I had not booted.

Requiring a boot is a bug in my mind. The data that's returned it's a copy
of the version partition. In this case my host wouldn't boot, and I wanted
to know what was in the pnor.

Is the contents of the version partition exposed though a different path
that does not require a boot?

williamspatrick, 2016-07-27:
No.  It isn't.  Parsing PNOR content would be a new feature.

williamspatrick, 2016-07-27:
And a Power-specific feature.  You can't always get this information on all systems because you're not guaranteed that the "BIOS" chip is even attached to the Aspeed.

amboar, 2016-07-29:
Potentially dumb question: What's the motivation for not baking the inventory into the BMC? Space constraints? Dynamic state? Other?

williamspatrick, 2016-08-01:
@amboar - Persistence of inventory just wasn't implemented for 1.x.  It is also not "easy".  It is trivial to serialize the object into the file system.  What is difficult is:
  1. Communicating to the user which cases the data is 'stale' and cannot be fully trusted.
  2. Identifying all of the cases where it is appropriate to erase the persisted data.

I think the shortcut for 1.x was to just not do persistence.  Then you are always only getting "the current data available to the BMC and nothing more."

williamspatrick, 2016-08-01:
See #487 also.  These sound related.

gtmills, 2018-08-20:
This is a really old build, inventory is now `/xyz/openbmc_project/inventory`, and no activity in 2 years. Closing. Please reopen if still an issue. Thanks!

closed 2018-08-20 by gtmills