openbmc/openbmc #487: Incorrect state transfer after reboot BMC when system power is on

johnhcwang, 2016-07-27:
It will cause that the inventory didn't be re-created even if the system power is on.

Step 1: Make sure that the system located in **HOST_BOOTED** state
````
# obmcutil state
ftgmac100: NCSI interface up
 = HOST_BOOTED
````
Step 2: Login to BMC by curl command
````
$ curl -c cjar -b cjar -k -X POST -H "Content-Type: application/json" -d '{"data": [ "root", "0penBmc" ] }' https://10.32.9.38/login
{
  "data": "User 'root' logged in", 
  "message": "200 OK", 
  "status": "ok"
}
$ curl -c cjar -b cjar -k https://10.32.9.38/org/openbmc/inventory/system/chassis/motherboard/bmc
{
  "data": {
    "fault": "False", 
    "fru_type": "BMC", 
    "is_fru": 0, 
    "manufacturer": "ASPEED", 
    "present": "False", 
    "version": "v1.0.3-dirty"
  }, 
  "message": "200 OK", 
  "status": "ok"
````
**Step 3: Create 5 sessions to get inventory continually and make sure that all of them can get the information correctly**
````
$for x in `seq 1 100`;do curl -b cjar -k https://10.32.9.38/org/openbmc/inventory/system/chassis/motherboard/enumerate ;done &
$for x in `seq 1 100`;do curl -b cjar -k https://10.32.9.38/org/openbmc/inventory/system/chassis/motherboard/enumerate ;done &
$for x in `seq 1 100`;do curl -b cjar -k https://10.32.9.38/org/openbmc/inventory/system/chassis/motherboard/enumerate ;done &
$for x in `seq 1 100`;do curl -b cjar -k https://10.32.9.38/org/openbmc/inventory/system/chassis/motherboard/enumerate ;done &
$for x in `seq 1 100`;do curl -b cjar -k https://10.32.9.38/org/openbmc/inventory/system/chassis/motherboard/enumerate ;done &
````
Step 4: Reboot BMC and the state **jump to HOST_BOOTED from BMC_READY directly**.
````
Jul 21 00:37:31 barreleye system_manager.py[918]: Running ChassisControlService
Jul 21 00:37:31 barreleye system_manager.py[918]: New object: /org/openbmc/control/chassis0 (:1.31)
Jul 21 00:37:31 barreleye system_manager.py[918]: Initializing fan: /org/openbmc/sensors/speed/fan0
Jul 21 00:37:31 barreleye system_manager.py[918]: Initializing fan: /org/openbmc/sensors/speed/fan1
Jul 21 00:37:31 barreleye system_manager.py[918]: Initializing fan: /org/openbmc/sensors/speed/fan2
Jul 21 00:37:31 barreleye system_manager.py[918]: Initializing fan: /org/openbmc/sensors/speed/fan3
Jul 21 00:37:31 barreleye system_manager.py[918]: Initializing fan: /org/openbmc/sensors/speed/fan4
Jul 21 00:37:32 barreleye system_manager.py[918]: Initializing fan: /org/openbmc/sensors/speed/fan5
Jul 21 00:37:32 barreleye system_manager.py[918]: Starting fan control
Jul 21 00:37:32 barreleye system_manager.py[918]: Setting fans to max
Jul 21 00:37:32 barreleye system_manager.py[918]: New object: /org/openbmc/control/fans (:1.32)
Jul 21 00:37:32 barreleye system_manager.py[918]: All required objects started for BMC_STARTING2
Jul 21 00:37:32 barreleye system_manager.py[918]: SystemManager Goto System State: BMC_READY
Jul 21 00:37:32 barreleye system_manager.py[918]: Running System State: BMC_READY
Jul 21 00:37:32 barreleye system_manager.py[918]: Starting process: /usr/sbin/discover_system_state.py: restore
Jul 21 00:37:35 barreleye system_manager.py[918]: Starting flasher: bios,,/org/openbmc/control/flash/bios,
Jul 21 00:37:35 barreleye system_manager.py[918]: flasher: , BMC = 0, address = 0x0
Jul 21 00:37:35 barreleye system_manager.py[918]: Setting up BIOS flash
Jul 21 00:37:35 barreleye system_manager.py[918]: Flash tuned
TCP: request_sock_TCP: Possible SYN flooding on port 443. Dropping request.  Check SNMP counters.
Jul 21 00:37:36 barreleye kernel: TCP: request_sock_TCP: Possible SYN flooding on port 443. Dropping request.  Check SNMP counters.
Jul 21 00:37:36 barreleye kernel[923]: TCP: request_sock_TCP: Possible SYN flooding on port 443. Dropping request.  Check SNMP counters.
Jul 21 00:37:37 barreleye system_manager.py[918]: Starting flasher: bios,,/org/openbmc/control/flash/bios,
Jul 21 00:37:37 barreleye system_manager.py[918]: flasher: , BMC = 0, address = 0x0
Jul 21 00:37:37 barreleye system_manager.py[918]: Setting up BIOS flash
Jul 21 00:37:37 barreleye system_manager.py[918]: Flash tuned
Jul 21 00:37:44 barreleye system_manager.py[918]: Running System State: HOST_BOOTED
Jul 21 00:37:44 barreleye system_manager.py[918]: Installing OCC device
Jul 21 00:37:44 barreleye system_manager.py[918]: Installing OCC device
````

williamspatrick, 2016-07-27:
I don't believe the inventory is currently persisted through a reboot of the BMC.  The inventory is only pushed down while the Hostboot code is running.  Unless you also reboot the system I don't think there is any way to get a new set of inventory.

johnhcwang, 2016-07-28:
No, it won't be persisted after reboot BMC. In this case, we also can't get any inventory after upgrade BMC FW until reboot the system.

amboar, 2016-07-29:
> Unless you also reboot the system I don't think there is any way to get a new set of inventory.

Just to clarify, if we reboot the BMC but not the host we subsequently don't receive anything in response to inventory queries?

williamspatrick, 2016-08-01:
Much of the system inventory is passed down to the BMC via IPMI from the Host.  This data is only available to be sent to the BMC while the system is booting.  Once the system is running, the firmware involved in creating this information is no longer available (hostboot).

The 1.x stream does not currently persist inventory data from the host.  If the BMC goes through a reset for any reason, including code update, the inventory data will be unavailable until the next system boot.

willenlee, 2016-08-12:
**1. Using curl command to get Inventory when system power on**
```
{
  "data": {
    "/org/openbmc/inventory/system": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "True", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/bios": {
      "Asset Tag": "", 
      "Custom Field 1": "\thostboot-bc98d0b-34310c7", 
      "Custom Field 2": "\tocc-0362706", 
      "Custom Field 3": "\tskiboot-5.1.5-f13d6e4", 
      "Custom Field 4": "\thostboot-binaries-43d5a59", 
      "Custom Field 5": "\tbarreleye-xml-fd8e8c8", 
      "Custom Field 6": "\tcapp-ucode-105cb8f", 
      "Custom Field 7": "", 
      "Custom Field 8": "", 
      "FRU File ID": "", 
      "Manufacturer": "", 
      "Model Number": "", 
      "Name": "OpenPOWER Firmware", 
      "Serial Number": "", 
      "Version": "OpenPower-Barreleye-v0.2.0", 
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "True", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "True", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan0": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan1": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan2": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan3": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan4": {
      "fault": "False", 
[... 1648 lines cut ...]
      "Custom Field 8": "", 
      "FRU File ID": "", 
      "Manufacturer": "INGRASYS", 
      "Name": "BARRELEYE", 
      "Part Number": "1A4218800-600-G", 
      "Serial Number": "62M001FX02", 
      "fault": "True", 
      "fru_type": "SYSTEM", 
      "is_fru": 0, 
      "present": "True", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/systemevent": {
      "fault": "False", 
      "fru_type": "SYSTEM_EVENT", 
      "is_fru": 0, 
      "present": "False", 
      "version": ""
    }
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```
**3. We could see some inventory status that didn't update.**

anoo1, 2016-08-15:
The 'present' and 'fault' properties were not being cached, so restoring the inventory restores all the data like the custom fields, but not the 'present' and 'fault' values. Will add a change to cache the present and fault properties.

johnhcwang, 2016-08-16:
Hi @anoo1, will the cached data be cleared if BMC reboot by update procedure?
It seems that all inventory data still can't be persisted after BMC FW update.

anoo1, 2016-08-16:
Hi @mdmillerii , could you comment on the persistency of '/var/cache/obmc/' during BMC code update?

closed 2016-08-17 by williamspatrick

johnhcwang, 2016-08-17:
Sorry that if I miss the information. Did we fix this issue during BMC update?

anoo1, 2016-08-17:
The /var/cache/ directory (where the inventory is stored) is erased during a BMC update. Preserving the inventory during a BMC update would be a different feature.

agangidi53, 2016-08-17:
Yes,  I was just testing if this gets cached during new BMC update method .

johnhcwang, 2016-08-18:
Hi @anoo1, as I mentioned on https://github.com/openbmc/openbmc/issues/487#issuecomment-235795089, I believe that it is more critical if the inventory can't be persisted after BMC update than reboot.
Do you need I to create another issue on GitHub to discuss this feature?

anoo1, 2016-08-18:
Yes, please open a new openbmc/openbmc issue. I talked to Norm today about this and will bring up with the team on the options to make it happen.

anoo1, 2016-08-18:
I'll be adding the cache directory to the whiltelist list so that it gets preserved during BMC FW updates. This change and the fix to cache the present and fault properties will be part of the next tag.