openbmc/openbmc #489: Correction need in Dbus interface document

sivassrr, 2016-07-28:
Brief Description: 
org.openbmc.HostIpmi interface signal method should be ReceivedMessage instead of ReceivedMsg as in currently in 
https://github.com/openbmc/docs/blob/master/dbus-interfaces.md

Currrent Behavior:
root@palmetto:~# dbus-send --system --print-reply --dest=org.openbmc.HostIpmi /org/openbmc/HostIpmi/1 org.freedesktop.DBus.Introspectable.Introspect
method return time=1467264884.248911 sender=:1.1 -> destination=:1.74 serial=9 reply_serial=2
   string "<!DOCTYPE node PUBLIC "-//freedesktop//DTD D-BUS Object Introspection 1.0//EN"
"http://www.freedesktop.org/standards/dbus/1.0/introspect.dtd"> (http://www.freedesktop.org/standards/dbus/1.0/introspect.dtd%27%3E) 
<node>
 <interface name="org.freedesktop.DBus.Peer">
  <method name="Ping"/>
  <method name="GetMachineId">
   <arg type="s" name="machine_uuid" direction="out"/>
  </method>
 </interface>
 <interface name="org.freedesktop.DBus.Introspectable">
  <method name="Introspect">
   <arg name="data" type="s" direction="out"/>
  </method>
 </interface>
 <interface name="org.freedesktop.DBus.Properties">
  <method name="Get">
   <arg name="interface" direction="in" type="s"/>
   <arg name="property" direction="in" type="s"/>
   <arg name="value" direction="out" type="v"/>
  </method>
  <method name="GetAll">
   <arg name="interface" direction="in" type="s"/>
   <arg name="properties" direction="out" type="a{sv}"/>
  </method>
  <method name="Set">
   <arg name="interface" direction="in" type="s"/>
   <arg name="property" direction="in" type="s"/>
   <arg name="value" direction="in" type="v"/>
  </method>
  <signal name="PropertiesChanged">
   <arg type="s" name="interface"/>
   <arg type="a{sv}" name="changed_properties"/>
   <arg type="as" name="invalidated_properties"/>
  </signal>
 </interface>
 <interface name="org.openbmc.HostIpmi">
  <method name="sendMessage">
   <arg type="y" direction="in"/>
   <arg type="y" direction="in"/>
   <arg type="y" direction="in"/>
   <arg type="y" direction="in"/>
   <arg type="y" direction="in"/>
   <arg type="ay" direction="in"/>
   <arg type="x" direction="out"/>
  </method>
  <method name="setAttention">
   <arg type="x" direction="out"/>
  </method>
  <signal name="ReceivedMessage">      <====================
   <arg type="y"/>
   <arg type="y"/>
   <arg type="y"/>
   <arg type="y"/>
   <arg type="ay"/>
  </signal>
 </interface>
</node>
"

williamspatrick, 2016-07-28:
@sivassrr - Please just make a pull request against the docs repository to update.

sivassrr, 2016-08-04:
https://github.com/openbmc/docs/pull/35 - Done.

closed 2016-08-04 by sivassrr