openbmc/openbmc #494: Changes to timeout handling on IPMI commands.

vishwabmc, 2016-07-29:
Currently, here is how any IPMI Netfn/Cmd is handled.

1/. Host sends a IPMI command with SeqNo-NetFn-Cmd-data
2/. BT driver receives it and broadcasts a message on HostIpmi/1 dbus
3/. BT starts a "1 second" timer for a response from IPMID.
4/. If BT receives a response, then its sent to Host.
5/. If BT does *not* see a response in *1 second" then it times out and sends an 'error response" to Host.

So in both the cases #4 and #5, Host has received a response for the command that was sent and is now free to reuse the SeqNo.

With this structure in place, we have an issue which looks like this :

1/. Host sends a IPMI command with SeqNo( 10 )-NetFn-Cmd-data
2/. BT driver receives it and broadcasts a message on HostIpmi/1 dbus
3/. BT starts a "1 second" timer for a response from IPMID.
4/. IPMID is taking a lot of time to process the request ( for whatever reason ).
5/. BT driver fails to receive the response and times out.
6/. BT driver sends an error response for SeqNo:10.
7/. Host reuses the Seq No: 10 and sends some other command
8/.  BT driver receives it and broadcasts a message on HostIpmi/1 dbus
9/. BT starts a "1 second" timer for a response from IPMID.
10/. IPMID now sends the response to SeqNo:10 which was actually sent in step #1.
11/. BT receives the response.
12/. Response is sent to host for SeqNo:10 which is not what was needed to be sent for the command sent in step #7.

So we need some ways to solve this problem.. Current thinking is this.

1/. BT driver and IPMID will use a 32 bit SeqNo, of which, the top uint8_t would contain the SeqNo that is sent by host and the bottom uint24_t would be some random number. 

2/. Doing this way will make sure that the BT does not send a response to in-appropriate SeqNo that is mentioned in above problem scenario.

jk-ozlabs, 2016-07-29:
> So we need some ways to solve this problem

No, we don't. We *need* to send the response with the same sequence number that the host sent us; there is no getting around that.

Regardless of whatever internal sequence number scheme you try to implement here, we still only have one byte of sequence number space to respond with. That's defined by the IPMI protocol, and will not change.

The way to avoid this is for the host not to re-use sequence numbers for outstanding IPMI messages. If the host re-uses a sequence number that is pending, there is *nothing* that the BMC can do to correct that.

vishwabmc, 2016-07-29:
@jk-ozlabs In this case, the transaction is not outstanding..  host has already received a response for a particular command but just that the response was a `head->rsp.cc = 0xce;` so it can then re use the SeqNo since its now transaction complete on that SeqNo.

Are we suggesting that Host should not use the same SeqNo if they receive the Timeout response ?.

jk-ozlabs, 2016-07-29:
Ah, so you're looking at the case where the message is still "outstanding" on the BMC side, but not on the host side. In this case, perhaps we should have a facility to cancel the request on the IPMI-daemon side, so that it doesn't attempt to send another response?

vishwabmc, 2016-07-29:
@jk-ozlabs Are we suggesting a timer in IPMID ?. I feel, we can leverage on timer that is set by BT anyway than having 2 timers. Do you see any issues with what we proposed on random number ?.

jk-ozlabs, 2016-07-29:
No issues with that, no. However, it seems a little wasteful to continue processing the (now useless) IPMI response.

I wouldn't suggest a new timer in the IPMI daemon; either the BT daemon would send a signal when it expires a message, or we move the timeout into the IPMI daemon (and so making the BT daemon purely a bridge, which only forwards between BT and dbus).

vishwabmc, 2016-07-29:
@jk-ozlabs: IMO, the timeout need to be maintained by the host since they own the usecase and we are just honouring the request.

williamspatrick, 2016-07-29:
The host maintains a timeout already.  We have a timeout in the btbridged because:

1. It is better for the BMC to give a timeout than the host to eventually assume a timeout.
2. There were cases when the ipmid crashed and was restarted by systemd, and therefore the message was lost, but we needed a way to clean up the outstanding resources in btbridged.

The current structure of the ipmid is single-threaded with the provider plugins being non-reentrant.  There isn't a good way with the current structure to "cancel" a request.  Right now none of the commands are long-running, so the most likely reason for a timeout is a restart of the process.  As we start adding more functionality we do need to come up with a 'callback' mechanism (which we might actually slightly have) but we would also need to write some mechanism to tie outstanding dbus requests to a btbridged request within the daemon.  This isn't currently done and based on the depth of our backlog I would rather not tackle that for Witherspoon.

The simplest solution to avoid mistakenly responding to an already timed-out message, to me, is to add the random 24 bits of salt onto the sequence ID within the btbridged<->ipmid interface.  This gets us by for Witherspoon.  I suggest we do that and then open an issue for the 3.x backlog to revisit.

rfrandse, 2018-08-13:
tracked ibm-openbmc/def#52

closed 2018-08-13 by rfrandse