openbmc/openbmc #498: Integrate qemu with palmetto-bmc machine model

amboar, 2016-08-08:
Currently bitbake can cook up an image for a versatilepb to be be run under qemu for virtualised testing. This requires a number of differences (such as kernel configuration) to the ast2x00s, and so problems don't always reflect reality on hardware.

Integrate a newer qemu (at least 2.6.0) into openbmc via runqemu to load a palmetto-bmc machine instead of versatilepb.

williamspatrick, 2016-08-08:
@amboar - Were you expecting to own this or someone else?  The first part of qemu update should be a straight-forward .bbappend (or new .bb for the version) in meta-phosphor.  The second part I think you were previously working on and we were having trouble figuring out how to mesh it with what was going on upstream in Yocto pertaining to QEMU.

amboar, 2016-08-08:
@williamspatrick I still expect to do the work myself, however if anyone is keen to take it up I wouldn't mind! I added this as an issue to give it some visibility given that there has been some confusion in other issues with  respect to test suite behaviour and differences in kernel configuration.

> The first part of qemu update should be a straight-forward .bbappend (or new .bb for the version) in meta-phosphor. The second part I think you were previously working on and we were having trouble figuring out how to mesh it with what was going on upstream in Yocto pertaining to QEMU.

Not so much having trouble figuring it out, rather converging on a solution we both agreed was acceptable. I think we have a way forward with the patches that have stagnated on the openembedded-core list - the plan was to pick them up, polish them, send them upstream for inclusion and carry them in openbmc/openbmc in the mean time.

williamspatrick, 2016-08-09:
Ack.  My wording isn't always the most precise.  We are on the same page with respect to the upstream work.

amboar, 2019-04-02:
Closing as a duplicate of #1057

closed 2019-04-02 by amboar