openbmc/openbmc #499: [Barreleye Build - v1.0-116-g0c0d27e-dirty]   BMC not booting to READY state

gkeishin, 2016-08-08:
The latest driver doesn't boot to BMC READY state.

Version info 

root@barreleye:~# cat /etc/version
20160805162917

root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.0-116"
VERSION_ID="v1.0-116-g0c0d27e-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.0-116"
BUILD_ID="v1.0"
root@barreleye:~#

root@barreleye:~# obmcutil state
 = BMC_STARTING
root@barreleye:~#

williamspatrick, 2016-08-08:
Looks like we are missing this dbus object:
`'/org/openbmc/sensors/speed/fan5': 0,`

williamspatrick, 2016-08-08:
adm1278 content is correctly in the device tree.

```
./ahb/apb/i2c@1e78a000/i2c-bus@140/adm1278@10
./ahb/apb/i2c@1e78a000/i2c-bus@140/adm1278@10/reg
./ahb/apb/i2c@1e78a000/i2c-bus@140/adm1278@10/name
./ahb/apb/i2c@1e78a000/i2c-bus@140/adm1278@10/compatible
./ahb/apb/i2c@1e78a000/i2c-bus@140/adm1278@10/sense-resistor
./ahb/apb/i2c@1e78a000/i2c-bus@180/adm1278@10
./ahb/apb/i2c@1e78a000/i2c-bus@180/adm1278@10/reg
./ahb/apb/i2c@1e78a000/i2c-bus@180/adm1278@10/name
./ahb/apb/i2c@1e78a000/i2c-bus@180/adm1278@10/compatible
./ahb/apb/i2c@1e78a000/i2c-bus@180/adm1278@10/sense-resistor
```

williamspatrick, 2016-08-08:
```
root@barreleye:/sys/class/hwmon# cat hwmon*/name
ds3231
lm75
root@barreleye:/sys/class/hwmon# ls
hwmon0  hwmon1
```

williamspatrick, 2016-08-08:
dmesg output from i2c drivers:
```
i2c /dev entries driver
i2c_aspeed 1e78a000.i2c: i2c controller registered, irq 20
at24 0-0050: 32768 byte 24c256 EEPROM, writable, 64 bytes/write
rtc-ds1307 0-0068: rtc core: registered ds3231 as rtc0
i2c_aspeed i2c-0: i2c bus 0 registered, irq 213
i2c_aspeed i2c-3: i2c bus 3 registered, irq 216
at24 4-0054: 32768 byte 24c256 EEPROM, writable, 64 bytes/write
i2c_aspeed i2c-4: i2c bus 4 registered, irq 217
i2c_aspeed i2c-5: i2c bus 5 registered, irq 218
at24 6-0051: 256 byte 24c02 EEPROM, writable, 8 bytes/write
at24 6-0055: 256 byte 24c02 EEPROM, writable, 8 bytes/write
i2c_aspeed i2c-6: i2c bus 6 registered, irq 219
lm75 0-004a: hwmon1: sensor 'lm75'
nct7904: probe of 6-002d failed with error -5
nct7904: probe of 6-002e failed with error -5
occ-i2c 3-0050: occ i2c driver ready: i2c addr@0x50
occ-i2c 3-0051: occ i2c driver ready: i2c addr@0x51
adm1275 4-0010: Failed to read Manufacturer ID
adm1275: probe of 4-0010 failed with error -5
adm1275 5-0010: Failed to read Manufacturer ID
adm1275: probe of 5-0010 failed with error -5
```

williamspatrick, 2016-08-08:
Based on the skeleton code we are expecting a device on bus 6-002e:

```
>-------'6-002e' : {
>------->-------'names' : {
>------->------->-------'pwm1' : { 'object_path' : 'speed/fan3','poll_interval' : 10000,'scale' : 1,'units' : '' },
>------->------->-------'pwm2' : { 'object_path' : 'speed/fan4','poll_interval' : 10000,'scale' : 1,'units' : '' },
>------->------->-------'pwm3' : { 'object_path' : 'speed/fan5','poll_interval' : 10000,'scale' : 1,'units' : '' },
   >---->------->-------'in1_input' : { 'object_path' : 'voltage/P1V35_CPU1_BUF4','poll_interval' : 10000,'scale' : 1,'units' : '' },
                        'in2_input' : { 'object_path' : 'voltage/P0V9_CPU1_BUF1','poll_interval' : 10000,'scale' : 1,'units' : '' },
                        'in3_input' : { 'object_path' : 'voltage/P0V9_CPU1_BUF2','poll_interval' : 10000,'scale' : 1,'units' : '' },
                        'in4_input' : { 'object_path' : 'voltage/P0V9_CPU1_BUF3','poll_interval' : 10000,'scale' : 1,'units' : '' },
                        'in5_input' : { 'object_path' : 'voltage/P0V9_CPU1_BUF4','poll_interval' : 10000,'scale' : 1,'units' : '' },
                        'in6_input' : { 'object_path' : 'voltage/P1V09_CPU1_BUF1','poll_interval' : 10000,'scale' : 1,'units' : '' },
                        'in7_input' : { 'object_path' : 'voltage/P1V09_CPU1_BUF2','poll_interval' : 10000,'scale' : 1,'units' : '' },
                        'in8_input' : { 'object_path' : 'voltage/P1V09_CPU1_BUF3','poll_interval' : 10000,'scale' : 1,'units' : '' },
                        'in9_input' : { 'object_path' : 'voltage/P1V09_CPU1_BUF4','poll_interval' : 10000,'scale' : 1,'units' : '' },
                        'in10_input' : { 'object_path' : 'voltage/P0V97_CPU1','poll_interval' : 10000,'scale' : 1,'units' : '' },
                        'in11_input' : { 'object_path' : 'voltage/P1V1_MEM1','poll_interval' : 10000,'scale' : 1,'units' : '' },
                        'in12_input' : { 'object_path' : 'voltage/P1V35_CPU1_BUF1','poll_interval' : 10000,'scale' : 1,'units' : '' },
                        'in13_input' : { 'object_path' : 'voltage/P1V35_CPU1_BUF2','poll_interval' : 10000,'scale' : 1,'units' : '' },
                        'in14_input' : { 'object_path' : 'voltage/P1V35_CPU1_BUF3','poll_interval' : 10000,'scale' : 1,'units' : '' },
>------->-------}
         },
```

williamspatrick, 2016-08-08:
Ah, it is actually a nct7904 at that location and not an adm1278.

https://github.com/openbmc/linux/blob/dev-4.6/arch/arm/boot/dts/aspeed-bmc-opp-barreleye.dts#L147

williamspatrick, 2016-08-08:
```
root@barreleye:/proc/device-tree# find | grep nct
./ahb/apb/i2c@1e78a000/i2c-bus@1c0/nct7904@2d
./ahb/apb/i2c@1e78a000/i2c-bus@1c0/nct7904@2d/reg
./ahb/apb/i2c@1e78a000/i2c-bus@1c0/nct7904@2d/name
./ahb/apb/i2c@1e78a000/i2c-bus@1c0/nct7904@2d/compatible
./ahb/apb/i2c@1e78a000/i2c-bus@1c0/nct7904@2e
./ahb/apb/i2c@1e78a000/i2c-bus@1c0/nct7904@2e/reg
./ahb/apb/i2c@1e78a000/i2c-bus@1c0/nct7904@2e/name
./ahb/apb/i2c@1e78a000/i2c-bus@1c0/nct7904@2e/compatible
```

williamspatrick, 2016-08-08:
NCT7904 seems to be enabled in the defconfig.

```
$ grep NCT ./meta-aspeed/meta-ast2400/recipes-kernel/linux/linux-obmc/defconfig
CONFIG_SENSORS_NCT7904=y
```

williamspatrick, 2016-08-08:
```
nct7904: probe of 6-002d failed with error -5
nct7904: probe of 6-002e failed with error -5
```

williamspatrick, 2016-08-08:
Need Joel to investigate kernel stuff at this point.

shenki, 2016-08-08:
`Linux barreleye 4.6.4-openbmc-20160722-1 #1 Wed Jul 27 10:09:13 UTC 2016 armv5tejl GNU/Linux`

```
root@barreleye:~# find  /sys/bus/i2c/devices
/sys/bus/i2c/devices
/sys/bus/i2c/devices/i2c-0
/sys/bus/i2c/devices/i2c-3
/sys/bus/i2c/devices/i2c-4
/sys/bus/i2c/devices/i2c-5
/sys/bus/i2c/devices/i2c-6
/sys/bus/i2c/devices/0-0050
/sys/bus/i2c/devices/0-0068
/sys/bus/i2c/devices/0-004a
/sys/bus/i2c/devices/3-0050
/sys/bus/i2c/devices/3-0051
/sys/bus/i2c/devices/4-0010
/sys/bus/i2c/devices/4-0054
/sys/bus/i2c/devices/5-0010
/sys/bus/i2c/devices/6-0051
/sys/bus/i2c/devices/6-0055
/sys/bus/i2c/devices/6-002d
/sys/bus/i2c/devices/6-002e
```

shenki, 2016-08-08:
i2c4 is not working. My theory is the EEPROM driver appears to work as it does not touch the bus when probing.

```
root@barreleye:~# tail -c 1 /sys/class/i2c-dev/i2c-4/device/4-0054/eeprom 
tail: read error: Connection timed out
```

Two theories:
  * APB reset is required. I tried to do this on the u-boot command line but I have not verified that method works
 * bus is not pinmuxed

shenki, 2016-08-08:
Under my 4.7 wip tree:

```
# sensors 
ds3231-i2c-0-68
Adapter: Aspeed i2c-0
temp1:        +29.5°C  

lm75-i2c-0-4a

Adapter: Aspeed i2c-0
temp1:        +27.5°C  (high = +80.0°C, hyst = +75.0°C)

nct7904-i2c-6-2d
Adapter: Aspeed i2c-6
in1:          +0.00 V  
in2:          +0.00 V  
in3:          +0.00 V  
in4:          +0.00 V  
in5:          +0.00 V  
in6:          +0.00 V  
in7:          +0.00 V  
in8:          +0.00 V  
in9:          +0.00 V  
in10:         +0.00 V  
in11:         +0.00 V  
in12:         +0.00 V  
in13:         +0.00 V  
in14:         +0.00 V  
in15:         +3.07 V  
in20:         +3.07 V  
fan1:           0 RPM
fan2:           0 RPM
fan3:           0 RPM
fan4:           0 RPM
fan5:           0 RPM
fan6:           0 RPM
fan7:           0 RPM
fan8:           0 RPM
fan9:           0 RPM
fan10:          0 RPM
temp1:         +2.6°C  
temp2:         +0.0°C  
temp3:         +0.0°C  
temp4:         +0.0°C  
temp5:         +0.0°C  
temp6:         +0.0°C  
temp7:         +0.0°C  
temp8:         +0.0°C  
temp9:         +0.0°C  

nct7904-i2c-6-2e
Adapter: Aspeed i2c-6
in1:          +0.00 V  
in2:          +0.00 V  
in3:          +0.00 V  
in4:          +0.00 V  
in5:          +0.00 V  
in6:          +0.00 V  
in7:          +0.00 V  
in8:          +0.00 V  
in9:          +0.00 V  
in10:         +0.00 V  
in11:         +0.00 V  
in12:         +0.00 V  
in13:         +0.00 V  
in14:         +0.00 V  
in15:         +3.07 V  
in20:         +3.07 V  
fan1:           0 RPM
fan2:           0 RPM
fan3:           0 RPM
fan4:           0 RPM
fan5:           0 RPM
fan6:           0 RPM
fan7:           0 RPM
fan8:           0 RPM
fan9:           0 RPM
fan10:          0 RPM
temp1:         +5.8°C  
temp2:         +0.0°C  
temp3:         +0.0°C  
temp4:         +0.0°C  
temp5:         +0.0°C  
temp6:         +0.0°C  
temp7:         +0.0°C  
temp8:         +0.0°C  
temp9:         +0.0°C  

adm1278-i2c-4-10
Adapter: Aspeed i2c-4
vin:         +12.05 V  (min =  +0.00 V, max = +20.89 V)
                       (highest = +12.44 V)
vout1:        +0.01 V  (min =  +0.00 V, max = +20.89 V)
                       (highest = +12.37 V)
pin:         293.97 mW (highest = -111.32 mW, max = -58.80 mW)
iout1:        +0.01 A  (max = +25.59 A, highest =  +6.03 A)

adm1278-i2c-5-10
Adapter: Aspeed i2c-5
vin:         +12.06 V  (min =  +0.00 V, max = +20.89 V)
                       (highest = +17.87 V)
vout1:        +0.01 V  (min =  +0.00 V, max = +20.89 V)
                       (highest = +12.37 V)
pin:         -47.35 mW (highest =   7.93 mW, max = -58.80 mW)
iout1:        +0.14 A  (max = +25.59 A, highest =  +6.37 A)

```

sivassrr, 2016-08-20:
BMC VERSION_ID="v1.0-140-g718613f-dirty" found booted to HOST. 
root@barreleye:~# obmcutil  state
 = HOST_BOOTED

gkeishin, 2016-08-21:
This issue is no longer seen in latter driver.. closing this..

closed 2016-08-21 by gkeishin

williamspatrick, 2016-08-23:
#513 is the continuation of the same kernel problems.  We still don't have them resolved.

sivassrr, 2016-09-26:
@gkeishin Based on the loop test, Still some of our loop test iteration fails because not able to see the system in HOST booted state. Shall we reopen?