openbmc/openbmc #500: Fail to compile with Ubuntu 16.04 on ppc64le

geissonator, 2016-08-08:
There are 2 issues with compiling openbmc on Ubuntu ppc64le.

1. The first is this message string: https://lists.yoctoproject.org/pipermail/yocto/2016-June/030784.html

Symptom is this error:
uninative selected but not configured, please set UNINATIVE_CHKSUM[ppc64le]

Workaround is to add this to your local.conf:
INHERIT_remove = "uninative" 

2. The second is this error:
| ERROR: User requested feature sdl
|        configure was not able to find it.
|        Install SDL devel

It seems related to this message chain and bug:
https://lists.yoctoproject.org/pipermail/poky/2016-May/010552.html
https://bugzilla.yoctoproject.org/show_bug.cgi?id=8553

I'm currently testing the proposed workaround of commenting out ASSUME_PROVIDED from local.conf.

geissonator, 2016-08-08:
Confirmed the workaround for issue 2, was able to compile and boot the generated image in QEMU on Ubuntu 16.04 with ppc64le.

shenki, 2016-08-08:
I think we want to remove the SDL qemu bits from our local.conf.

We can document how to build qemu with the SDL GUI for those who need it. For the majority of use cases it will not be required.

Are you building on 14.04 for a reason? It would be nice to test on the latest LTS.

kc8apf, 2016-08-08:
The first issue is related to Yocto uninative which isn't strictly
necessary. Adding INHERIT_ remove to local.conf as you did just disabled
it. You can also run the BitBake command to generate the uninative tarball.

The second issue is that you do not have the libsdl headers installed.
Install libsdl-dev or equivalent. This is in the top-level readme.

geissonator, 2016-08-08:
@shenki - I'm using Ubuntu 16.04 LTS
@kc8apf - I had the lib installed, there was a change recently in yocto to try and build the sdl natively (I think).  Commenting out ASSUME_PROVIDED += "libsdl-native" from local.conf fixed this.

kc8apf, 2016-08-08:
Yocto has always assumed the host has libsdl available. That's what
ASSUME_PROVIDED += "libsdl-native" does.  The error indicates binaries
can't be built against libsdl on the host.  Could be the host missing
headers or the test in qemu is broken.

geissonator, 2016-08-11:
Patrick resolved the ASSUME_PROVIDED part of this issue with https://gerrit.openbmc-project.xyz/#/c/158/.  So now we're just tracking the uninative aspect.

KennethWilke, 2016-08-22:
I hit this trying to run a build today, after commenting out the ASSUME_PROVIDED line i get:
```
ERROR: Nothing PROVIDES 'libsdl-native' (but virtual:native:/home/kwilke/builds/openbmc/meta/recipes-devtools/qemu/qemu_2.4.0.bb DEPENDS on or otherwise requires it). Close matches:
  libxslt-native
  libsm-native
  mklibs-native
```

Is there another step I missed to get the build going?

geissonator, 2016-08-23:
@KennethWilke was that against stable or master?  The workaround doesn't seem to work for stable (I think the yocto version in stable doesn't have rules on how to generate the sdl library).

KennethWilke, 2016-08-23:
@geissonator I'm trying to build the latest tag for Barreleye provided by our vendor which based on the openbmc v1.0.4 tag: https://github.com/ken1029/openbmc/tree/v10.40.256

geissonator, 2016-08-23:
Yeah, you're going t have to use Ubuntu 14.04 then to build that for now.

mdmillerii, 2016-08-23:
I would guess that 16.04 has similar packages that can build the required dependencies.

The point was the older yocto in the `v1.0-stable` branch does not have code to build the `sdl` even if its removed from the `ASSUME_PROVIDED`, which was the solution on the `master` branch, and so the build dependencies will need to be pre-installed.

mdmillerii, 2016-08-23:
@KennethWilke The [README](https://github.com/openbmc/openbmc/blob/master/README.md) file in this repository lists several packages required on the host for the ubuntu and fedora distributions.  That applied to the 1.0.x releases.

* this had been drafted before the prior comment about 16.04 but aparently not posted

geissonator, 2016-08-23:
I had installed the required packages into my 16.04 instance but still no dice on compiling the stable branch.  They must be incompatible with the older yocto.

KennethWilke, 2016-08-24:
@mdmillerii I did ensure those packages were installed prior to building, and also installed all libsdl packages to see if any would satisfy the dependency but that resulted in the same errors. 

I'd previously built openbmc on 15.10, so will spin up a VM to build it on for now.

KennethWilke, 2016-12-07:
Commit 6044148e9177c95c6ad62a1cdf52a9ac3954535c has resolved this issue on my end for building the `v1.0-stable` branch for running in qemu and for running on Barreleye.

When I try to build on the master branch I get this result:
```
kwilke@barreleye-1:/tmp/openbmc$ . openbmc-env
### Initializing OE build env ###
You had no conf/local.conf file. This configuration file has therefore been
created for you with some default values. You may wish to edit it to, for
example, select a different MACHINE (target hardware). See conf/local.conf
for more information as common configuration options are commented.

You had no conf/bblayers.conf file. This configuration file has therefore been
created for you with some default values. To add additional metadata layers
into your configuration please add entries to conf/bblayers.conf.

The Yocto Project has extensive documentation about OE including a reference
manual which can be found at:
    http://yoctoproject.org/documentation

For more information about OpenEmbedded see their website:
    http://www.openembedded.org/

### Shell environment set up for builds. ###

You can now run 'bitbake <target>'

Common targets are:
     obmc-phosphor-image

You can also run generated qemu images with a command like:
     runqemu qemuarm
kwilke@barreleye-1:/tmp/openbmc/build$ bitbake obmc-phosphor-image
NOTE: Your conf/bblayers.conf has been automatically updated.
Parsing recipes: 100% |########################################################################| Time: 00:00:12
Parsing of 1918 .bb files complete (0 cached, 1918 parsed). 2486 targets, 385 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "ppc64le-linux"
NATIVELSBSTRING   = "Ubuntu-16.04"
TARGET_SYS        = "arm-openbmc-linux-gnueabi"
MACHINE           = "qemuarm"
DISTRO            = "openbmc-phosphor"
DISTRO_VERSION    = "0.1.0"
TUNE_FEATURES     = "arm armv5 thumb dsp"
TARGET_FPU        = "soft"
meta              
meta-poky         
meta-oe           
meta-networking   
meta-python       
meta-virtualization 
meta-phosphor     = "master:ed181d504f1f622f8e19ab9d83b4470c191fc5c9"

ERROR: Uninative selected but not configured correctly, please set UNINATIVE_CHECKSUM[ppc64le]

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

kc8apf, 2016-12-07:
Yocto 2.1 introduced uninative which is an attempt to isolate the build
system from the host OS.  Looks like they haven't published a uninative
tarball for ppc64le.  You can build one yourself.  See the details at
http://www.yoctoproject.org/docs/latest/ref-manual/ref-manual.html#ref-classes-uninative

adamliyi, 2017-03-15:
A quick fix to error:
```
ERROR: Uninative selected but not configured correctly, please set UNINATIVE_CHECKSUM[ppc64le]
```
is

```
--- a/import-layers/yocto-poky/meta-poky/conf/distro/poky.conf
+++ b/import-layers/yocto-poky/meta-poky/conf/distro/poky.conf
@@ -107,4 +107,4 @@ ERROR_QA_append = " ${WARN_TO_ERROR_QA}"
 require conf/distro/include/poky-world-exclude.inc
 require conf/distro/include/no-static-libs.inc
 require conf/distro/include/yocto-uninative.inc
-INHERIT += "uninative"
+#INHERIT += "uninative"
```

geissonator, 2017-03-15:
Thanks for the reminder Yi, yeah we now have master building on ppc64le Ubuntu for all systems.

closed 2017-03-15 by geissonator