openbmc/openbmc #501: REST server should not depend on systemd-networkd

bradbishop, 2016-08-08:
As a workaround for openbmc/phosphor-rest-server#24 PartOf=systemd-networkd was added to the obmc-rest service unit.

The network stack implementation should not be hardcoded.  Find a way to restart the REST server when the IP address changes irrespective of the network implementation.

williamspatrick, 2016-08-08:
What we saw was that, under load, the rocket threads seemed to be getting blocked on their connections that were left open from the old network address.  This ended up hanging the worker threads so that they could not handle new connections.

We'll probably want to run 'strace' on the process under this condition to get a better picture of what is going on.  I'm surprised that this is something userspace applications really need to be aware of.  I would have thought that changing ip addresses was a fairly common occurrence on wifi.

bradbishop, 2016-08-08:
Maybe python-gevent will handle a little better....

geissonator, 2017-10-13:
We fixed this with latest web server I believe Brad?

closed 2017-10-13 by geissonator

mdmillerii, 2017-10-16:
If you change address connections that were established with the old address will no longer get any incoming packets.  The workaround is to establish a new web connection; or otherwise timeout the connections.  @geissonator @bradbishop should we reopen, and have the servers kill open connections?

@williamspatrick on wifi the address doesn't often change while a connection is open.  Typically the dhcp server will renew the address, at home the assignments are often 24+ hours, and in corporate environments with managed access points a subnet often extends across a wide range.