openbmc/openbmc #502: Witherspoon dts update for APSS UART.

williamspatrick, 2016-08-08:
Pass2 of the board is adding a dedicated UART connection between the ast2500 and the APSS.  We need to make sure this is in the device-tree correctly in case we need to use it for debug purposes.  The main purpose of it would be, if we need in the lab, to push a code update into the APSS.

williamspatrick, 2016-08-08:
Would like to have this by mid-September.

shenki, 2016-08-08:
This is a 30 second job. It requires two bits of information:

1. Which UART pints are being used?
2. What is the desired baud rate?

It would be good to document the firmware update process, or at least perform some kind of handshake with APSS so we can test the change.

williamspatrick, 2016-08-08:
We aren't going to be responsible for implementing a code update at this point.  Jordan implied they have some scripting that can turn an image into a bunch of 'echo' commands onto a UART.  We can get the baud rate from him (copied you via email).

amboar, 2016-08-09:
> ... UART pints ...

Untappd doesn't even list this beer, must be exclusive!

closed 2019-10-27 by stale[bot]