openbmc/openbmc #503: [QEMU Build - v1.0-116-g0c0d27e-dirty]: dbus-send  to HostIpmi is not working

sivassrr, 2016-08-08:
Brief Description of the issue:  
Issued the command "dbus-send --system --type=signal --dest=org.openbmc.HostIpmi /org/openbmc/HostIpmi/1 org.openbmc.HostIpmi.ReceivedMessage byte:0x00 byte:0x04 byte:0x00 byte:0x30 array:byte:9,0x01,0x00,0x35,0x00,0x00,0x00,0x00,0x00,0x00 " at BMC prompt, ipmid didnt receive any message.  

**Expected Behavior:**
ipmid should pick up and set the equivalent sensor value of BootCount.

**Actual Behavior:**
ipmid didnt received the command:
Here is the ipmid -d7 & output
Registering NetFn:[0x2C], Cmd:[0x3]
Registering NetFn:[0xC], Cmd:[0xFF]
Registering NetFn:[0xC], Cmd:[0x1]
Registering NetFn:[0xC], Cmd:[0x2]
Registering NetFn:[0x6], Cmd:[0x2]
Registering NetFn:[0x2C], Cmd:[0x0]   <===this is the last output

**Part of the dbus-monitor ouput:**
signal time=1470670925.372659 sender=:1.152 -> destination=org.openbmc.HostIpmi serial=2 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=ReceivedMessage
   byte 0
   byte 4
   byte 0
   byte 48
   array of bytes [
      09 01 00 35 00 00 00 00 00 00
   ]
error time=1470670925.379604 sender=org.freedesktop.DBus -> destination=:1.152 error_name=org.freedesktop.DBus.Error.ServiceUnknown reply_serial=2
   string "The name org.openbmc.HostIpmi was not provided by any .service files"
signal time=1470670925.385888 sender=org.freedesktop.DBus -> destination=:1.152 serial=12 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameLost
   string ":1.152"
signal time=1470670925.388142 sender=org.freedesktop.DBus -> destination=(null destination) serial=328 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameOwnerChanged
   string ":1.152"
   string ":1.152"
   string ""

**Recreations Steps:**
Execute the following command in QEMU, Barreleye, Palmetto (Tried it on all)
dbus-send --system --type=signal --dest=org.openbmc.HostIpmi /org/openbmc/HostIpmi/1 org.openbmc.HostIpmi.ReceivedMessage byte:0x00 byte:0x04 byte:0x00 byte:0x30 array:byte:9,0x01,0x00,0x35,0x00,0x00,0x00,0x00,0x00,0x00

Tried varies other commands using HostIpmi none worked.

sivassrr, 2016-08-08:
[journalctlHostIpmi.txt](https://github.com/openbmc/openbmc/files/407193/journalctlHostIpmi.txt)

williamspatrick, 2016-08-08:
This looks problematic...
`string "The name org.openbmc.HostIpmi was not provided by any .service files"`

@bradbishop - Any ideas on this?

vishwabmc, 2016-08-12:
@williamspatrick I had told this to @sivassrr .. on a QEMU, there is no one that is hosting  org.openbmc.HostIpmi service. The only way we could do that in QEMU would be to use the ipmi_debug.py and supply commands in 'stdin'.

dbus-send to IPMI daemon would only make sense in a real HW since BT would have advertised it.

vishwabmc, 2016-08-12:
-or- use the ipmitool -I dbus

sivassrr, 2016-08-12:
@vishwabmc As mentioned during our discussion, Could able to run it in barreleye system with version-id : v1.0-121-g0a4549a-dirty successfully. 

QEMU with palmetto, We should still able to see that service, right.

sivassrr, 2016-08-16:
@vishwabmc , @bradbishop - On Real HW tried and here is the output.

**Executed real HW (Barreleye) version-id: v1.0-140-g718613f-dirty (output of part of dbus-monitor):**
method call time=1471356329.251866 sender=:1.159 -> destination=org.freedesktop.DBus serial=1 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=Hello
method return time=1471356329.253426 sender=org.freedesktop.DBus -> destination=:1.159 serial=1 reply_serial=1
   string ":1.159"
signal time=1471356329.255530 sender=org.freedesktop.DBus -> destination=(null destination) serial=315 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameOwnerCh
   string ":1.159"
   string ""
   string ":1.159"
signal time=1471356329.259137 sender=org.freedesktop.DBus -> destination=:1.159 serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameAcquired
   string ":1.159"
signal time=1471356329.282473 sender=:1.159 -> destination=(null destination) serial=2 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=ReceivedMessage
   byte 0
   byte 0
   byte 0
   byte 8
   array of bytes [
      05 80 14 00 00 00
   ]
method call time=1471356329.320373 sender=:1.5 -> destination=org.openbmc.ObjectMapper serial=591 path=/org/openbmc/ObjectMapper; interface=org.openbmc.ObjectMapper; member=GetObjec
   string "/org/openbmc/settings/host0"
signal time=1471356329.326419 sender=org.freedesktop.DBus -> destination=:1.159 serial=5 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameLost
   string ":1.159"
signal time=1471356329.328689 sender=org.freedesktop.DBus -> destination=(null destination) serial=316 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameOwnerCh
   string ":1.159"
   string ":1.159"
   string ""
method return time=1471356329.426921 sender=:1.7 -> destination=:1.5 serial=702 reply_serial=591
   array [
      dict entry(
         string ":1.6"
         array [
            string "org.openbmc.settings.Host"
            string "org.openbmc.Object.Properties"
         ]
      )
   ]
method call time=1471356329.525118 sender=:1.5 -> destination=:1.6 serial=592 path=/org/openbmc/settings/host0; interface=org.freedesktop.DBus.Properties; member=Set
   string "org.openbmc.settings.Host"
   string "boot_flags"
   variant       string "CDROM"
method return time=1471356329.529014 sender=:1.6 -> destination=:1.5 serial=523 reply_serial=592
method call time=1471356329.536078 sender=:1.5 -> destination=org.openbmc.ObjectMapper serial=593 path=/org/openbmc/ObjectMapper; interface=org.openbmc.ObjectMapper; member=GetObjec
   string "/org/openbmc/settings/host0"
method return time=1471356329.539287 sender=:1.7 -> destination=:1.5 serial=703 reply_serial=593
   array [
      dict entry(
         string ":1.6"
         array [
            string "org.openbmc.settings.Host"
            string "org.openbmc.Object.Properties"
         ]
      )
   ]
method call time=1471356329.546391 sender=:1.5 -> destination=:1.6 serial=594 path=/org/openbmc/settings/host0; interface=org.freedesktop.DBus.Properties; member=Set
   string "org.openbmc.settings.Host"
   string "boot_policy"
   variant       string "ONETIME"
method return time=1471356329.550336 sender=:1.6 -> destination=:1.5 serial=524 reply_serial=594
method call time=1471356329.551667 sender=:1.5 -> destination=:1.159 serial=595 path=/org/openbmc/HostIpmi/1; interface=org.openbmc.HostIpmi; member=sendMessage
   byte 0
   byte 1
   byte 0
   byte 8
   byte 0
   array [
   ]
error time=1471356329.555650 sender=org.freedesktop.DBus -> destination=:1.5 error_name=org.freedesktop.DBus.Error.ServiceUnknown reply_serial=595
   string **"The name :1.159 was not provided by any .service files"     <======**

causten, 2016-08-16:
In short I don't think the --dest=org.openbmc.HostIpmi is needed.  Remove it from your dbus-send and it works.  The basic versatilepb and calling the code works.  Your exact command causes an ipmi crash which is actually a defect (ipmid needs to handle gracefully invalid arguments) but calling ipmi with dbus-send which is what this issue is about works fine.  

1) there should be a space between the -d and the 7.  Not ipmid -d7 
2) Killing a process with systemd will simply restart it.  Before running the "ipmid -d 7" you must run systemctl stop host-ipmid
3) This works...
dbus-send --system /org/openbmc/HostIpmi/1 org.openbmc.HostIpmi.ReceivedMessage byte:0x00 byte:0x04 byte:0x00 byte:0x30 array:byte:9,0x01,0x00,0x35,0x00,0x00,0x00,0x00,0x00,0x00

closed 2016-08-16 by causten

sivassrr, 2016-08-17:
I am not using -dest path in my code. Is there any defect for ipmid crash ? 
And objective of the defect is to evaluate the error
"error time=1471356329.555650 sender=org.freedesktop.DBus -> destination=:1.5 error_name=org.freedesktop.DBus.Error.ServiceUnknown reply_serial=595
   string **"The name :1.159 was not provided by any .service files"     <======**"

williamspatrick, 2016-08-18:
@sivassrr - Why do you say "I am not using -dest path in my code"?  Every example above you have it.

Is ipmid crashing?