openbmc/openbmc #505: netman_watch_dns fails to update resolv.conf

shenki, 2016-08-10:
when booting openbmc master on my palmetto on a network with a DHCP server:

```
Aug 10 09:04:18 palmetto netman_watch_dns[676]: Error updating resolv.conf with:[DHCP_AUTO= 192.231.203.132 210.5.35.1
```

When booted, `/etc/resolv.conf` is empty.

vishwabmc, 2016-08-10:
Is org.openbmc.NetworkManager running ?.

shenki, 2016-08-10:
```
root@palmetto:~# ps | grep -i net
  673 root     17868 S    python /usr/sbin/netman.py
  676 root      3768 S    /usr/sbin/netman_watch_dns /run/systemd/netif/state
  677 systemd- 11372 S    /lib/systemd/systemd-networkd
```

shenki, 2016-08-10:
How do I tell if it is running?

vishwabmc, 2016-08-10:
hmm.. netman.py must have hosted the necessary services.. Can you paste the journal output for this ?

shenki, 2016-08-10:
```
-- Logs begin at Wed 2016-08-10 09:03:53 UTC, end at Wed 2016-08-10 09:29:20 UTC. --
Aug 10 09:03:53 palmetto systemd-journald[404]: Runtime journal (/run/log/journal/) is 3.0M, max 24.3M, 21.3M free.
Aug 10 09:03:53 palmetto kernel: Booting Linux on physical CPU 0x0
Aug 10 09:03:53 palmetto kernel: Linux version 4.7.0-00052-g758c30c81ef2 (joel@aurora) (gcc version 6.1.1 20160705 (Ubuntu 6.1.1-9ubuntu11) ) #796 Wed Aug 10 18:32:56 ACST 2016
Aug 10 09:03:53 palmetto kernel: CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
Aug 10 09:03:53 palmetto kernel: CPU: VIVT data cache, VIVT instruction cache
Aug 10 09:03:53 palmetto kernel: Machine model: Palmetto BMC
Aug 10 09:03:53 palmetto kernel: Memory policy: Data cache writeback
Aug 10 09:03:53 palmetto kernel: SOC Rev: 02010303
Aug 10 09:03:53 palmetto kernel: On node 0 totalpages: 126976
Aug 10 09:03:53 palmetto kernel: free_area_init_node: node 0, pgdat c061b80c, node_mem_map dec19000
Aug 10 09:03:53 palmetto kernel:   Normal zone: 992 pages used for memmap
Aug 10 09:03:53 palmetto kernel:   Normal zone: 0 pages reserved
Aug 10 09:03:53 palmetto kernel:   Normal zone: 126976 pages, LIFO batch:31
Aug 10 09:03:53 palmetto kernel: pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Aug 10 09:03:53 palmetto kernel: pcpu-alloc: [0] 0 
Aug 10 09:03:53 palmetto kernel: Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 125984
Aug 10 09:03:53 palmetto kernel: Kernel command line: console=ttyS4,115200n8 root=/dev/ram rw
Aug 10 09:03:53 palmetto kernel: PID hash table entries: 2048 (order: 1, 8192 bytes)
Aug 10 09:03:53 palmetto kernel: Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Aug 10 09:03:53 palmetto kernel: Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Aug 10 09:03:53 palmetto kernel: Memory: 496404K/507904K available (3176K kernel code, 130K rwdata, 688K rodata, 1024K init, 110K bss, 11500K reserved, 0K cma-reserved)
Aug 10 09:03:53 palmetto kernel: Virtual kernel memory layout:
                                     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                                     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                                     vmalloc : 0xdf800000 - 0xff800000   ( 512 MB)
                                     lowmem  : 0xc0000000 - 0xdf000000   ( 496 MB)
                                       .text : 0xc0008000 - 0xc04c60c8   (4857 kB)
                                       .init : 0xc0500000 - 0xc0600000   (1024 kB)
                                       .data : 0xc0600000 - 0xc0620970   ( 131 kB)
                                        .bss : 0xc0620970 - 0xc063c288   ( 111 kB)
Aug 10 09:03:53 palmetto kernel: NR_IRQS:16 nr_irqs:16 16
Aug 10 09:03:53 palmetto kernel: clocksource: moxart_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 39817925974 ns
Aug 10 09:03:53 palmetto kernel: sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
Aug 10 09:03:53 palmetto kernel: Calibrating delay loop... 190.87 BogoMIPS (lpj=954368)
Aug 10 09:03:53 palmetto kernel: pid_max: default: 32768 minimum: 301
Aug 10 09:03:53 palmetto kernel: Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Aug 10 09:03:53 palmetto kernel: Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Aug 10 09:03:53 palmetto kernel: CPU: Testing write buffer coherency: ok
Aug 10 09:03:53 palmetto kernel: Setting up static identity map for 0x40100000 - 0x4010003c
Aug 10 09:03:53 palmetto kernel: devtmpfs: initialized
Aug 10 09:03:53 palmetto kernel: clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Aug 10 09:03:53 palmetto kernel: pinctrl core: initialized pinctrl subsystem
Aug 10 09:03:53 palmetto kernel: NET: Registered protocol family 16
Aug 10 09:03:53 palmetto kernel: DMA: preallocated 256 KiB pool for atomic coherent allocations
Aug 10 09:03:53 palmetto kernel: clocksource: Switched to clocksource moxart_timer
Aug 10 09:03:53 palmetto kernel: NET: Registered protocol family 2
Aug 10 09:03:53 palmetto kernel: TCP established hash table entries: 4096 (order: 2, 16384 bytes)
Aug 10 09:03:53 palmetto kernel: TCP bind hash table entries: 4096 (order: 2, 16384 bytes)
Aug 10 09:03:53 palmetto kernel: TCP: Hash tables configured (established 4096 bind 4096)
Aug 10 09:03:53 palmetto kernel: UDP hash table entries: 256 (order: 0, 4096 bytes)
Aug 10 09:03:53 palmetto kernel: UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
Aug 10 09:03:53 palmetto kernel: NET: Registered protocol family 1
Aug 10 09:03:53 palmetto kernel: RPC: Registered named UNIX socket transport module.
Aug 10 09:03:53 palmetto kernel: RPC: Registered udp transport module.
Aug 10 09:03:53 palmetto kernel: RPC: Registered tcp transport module.
Aug 10 09:03:53 palmetto kernel: RPC: Registered tcp NFSv4.1 backchannel transport module.
Aug 10 09:03:53 palmetto kernel: Trying to unpack rootfs image as initramfs...
Aug 10 09:03:53 palmetto kernel: Freeing initrd memory: 1628K (c0800000 - c0997000)
Aug 10 09:03:53 palmetto kernel: futex hash table entries: 256 (order: -1, 3072 bytes)
Aug 10 09:03:53 palmetto kernel: workingset: timestamp_bits=29 max_order=17 bucket_order=0
Aug 10 09:03:53 palmetto kernel: squashfs: version 4.0 (2009/01/31) Phillip Lougher
Aug 10 09:03:53 palmetto kernel: jffs2: version 2.2. (NAND) (SUMMARY)  © 2001-2006 Red Hat, Inc.
[... cut ...]
Aug 10 09:27:20 palmetto system_manager.py[688]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
Aug 10 09:27:50 palmetto system_manager.py[688]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
Aug 10 09:28:20 palmetto system_manager.py[688]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
Aug 10 09:28:50 palmetto system_manager.py[688]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
Aug 10 09:29:20 palmetto system_manager.py[688]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
```

vishwabmc, 2016-08-10:
Aug 10 09:04:18 palmetto systemd-networkd[677]: eth0: DHCPv4 address 10.104.0.83/24 via 10.104.0.1
Aug 10 09:04:18 palmetto systemd-timesyncd[660]: Network configuration changed, trying to establish connection.
Aug 10 09:04:18 palmetto netman_watch_dns[676]: ERROR updating DNS entries:[The name org.openbmc.NetworkManager was not provided by any .service files]

Aug 10 09:04:18 palmetto netman_watch_dns[676]: Error updating resolv.conf with:[DHCP_AUTO= 192.231.203.132 210.5.35.1
Aug 10 09:04:18 palmetto netman_watch_dns[676]:   ]

An issue with serializing.. This would need a change. I have put a Requires= but not sure why it would still start.  Need to see more

mdmillerii, 2016-08-10:
If you require another service to be running you need to put that in the
service file.  According to the [documentation](https://www.freedesktop.org/software/systemd/man/systemd.unit.html#Requires=):

> Note  that requirement dependencies do not influence the order in which services are started or stopped. This has to be configured independently with the After= or Before= options. If a unit foo.service requires a unit bar.service as configured with Requires= and no ordering is configured with After= or Before=, then both units will be started simultaneously and without any delay between them if foo.service is activated.

You may want to review other clauses listed there.

vishwabmc, 2016-08-11:
I already have this.

[Unit]
Description=DNS update handler
Requires=network.service
After=network.service

[Service]
ExecStart=/usr/sbin/netman_watch_dns /run/systemd/netif/state

[Install]
WantedBy=multi-user.target

And while booting BMC, I see that 

Network DBUS object is started first and only then DNS Update Handler starts.

May be the python script is taking some time launching services and before its done, netman_watch_dns seems to be doing things...  I would need to wait on something else now.. May be the standard network services that come late..

bradbishop, 2016-08-11:
The Type= in network.service should be set to dbus.  That way, systemd won’t count the job as started until netman has actually grabbed its busname.

closed 2019-10-28 by stale[bot]