openbmc/openbmc #507: Failed to run on AST2500 EVB

rednewguy, 2016-08-12:
I am not sure if I am doing it right. Used conf at meta-openbmc-machines/meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf and flashed "image-bmc" using tftp from u-boot on AST2500 EVB. 
U-Boot and Kernel seem to load and run properly, but fails in init. It couldn't find mtd for some reason. 
Is this a known issue or am I missing something?

`U-Boot 2016.07 (Aug 05 2016 - 10:14:49 -0700)

DRAM:  496 MiB
Flash: 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:   MAC0 : RGMII
MAC1 : RGMII
FTGMAC100#0, FTGMAC100#1
Hit any key to stop autoboot:  0
Booting kernel from Legacy Image at 20080000 ...
   Image Name:   Phosphor OpenBMC (Phosphor OpenB
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1989762 Bytes = 1.9 MiB
   Load Address: 80001000
   Entry Point:  80001000
   Verifying Checksum ... OK
Loading init Ramdisk from Legacy Image at 20300000 ...
   Image Name:   obmc-phosphor-image
   Image Type:   ARM Linux RAMDisk Image (lzma compressed)
   Data Size:    1668196 Bytes = 1.6 MiB
   Load Address: 81000000
   Entry Point:  81000000
   Verifying Checksum ... OK
   Loading Kernel Image ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Booting Linux on physical CPU 0x0
Linux version 4.6.4-openbmc-20160722-1 (docker@docker-Virtual-Machine) (gcc vers                                                                                                                                                             ion 5.3.0 (GCC) ) #1 Fri Aug 5 09:47:51 PDT 2016
CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Machine model: AST2500 EVB
Memory policy: Data cache writeback
SOC Rev: 04010303
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 125984
Kernel command line: console=ttyS4,115200n8 root=/dev/ram rw
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Memory: 496276K/507904K available (3698K kernel code, 184K rwdata, 1344K rodata,                                                                                                                                                              180K init, 107K bss, 11628K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0x9f800000 - 0xff800000   (1536 MB)
    lowmem  : 0x80000000 - 0x9f000000   ( 496 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x804f4a98   (5043 kB)
      .init : 0x804f5000 - 0x80522000   ( 180 kB)
      .data : 0x80522000 - 0x80550058   ( 185 kB)
       .bss : 0x80550058 - 0x8056ae64   ( 108 kB)
NR_IRQS:16 nr_irqs:16 16
clocksource: moxart_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns:                                                                                                                                                              77222644334 ns
sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 2147483647500                                                                                                                                                             0000ns
Calibrating delay loop... 789.70 BogoMIPS (lpj=3948544)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x80008220 - 0x8000827c
devtmpfs: initialized
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911                                                                                                                                                             2604462750000 ns
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
clocksource: Switched to clocksource moxart_timer
NET: Registered protocol family 2
TCP established hash table entries: 4096 (order: 2, 16384 bytes)
TCP bind hash table entries: 4096 (order: 2, 16384 bytes)
TCP: Hash tables configured (established 4096 bind 4096)
UDP hash table entries: 256 (order: 0, 4096 bytes)
UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
NET: Registered protocol family 1
RPC: Registered named UNIX socket transport module.
RPC: Registered udp transport module.
RPC: Registered tcp transport module.
RPC: Registered tcp NFSv4.1 backchannel transport module.
Trying to unpack rootfs image as initramfs...
Freeing initrd memory: 1632K (81000000 - 81198000)
futex hash table entries: 256 (order: -1, 3072 bytes)
workingset: timestamp_bits=29 max_order=17 bucket_order=0
squashfs: version 4.0 (2009/01/31) Phillip Lougher
jffs2: version 2.2. (NAND) (SUMMARY)  © 2001-2006 Red Hat, Inc.
io scheduler noop registered (default)
Serial: 8250/16550 driver, 6 ports, IRQ sharing enabled
console [ttyS4] disabled
1e784000.serial: ttyS4 at MMIO 0x1e784000 (irq = 20, base_baud = 1500000) is a 1                                                                                                                                                             6550A
console [ttyS4] enabled
brd: module loaded
loop: module loaded
bt-host 1e789140.ibt: Found bt host device
bt-host 1e789140.ibt: Using IRQ 18
ftgmac100 1e660000.ethernet: Read MAC address from chip 00:0c:cd:dc:ce:b0
libphy: ftgmac100_mdio: probed
ftmac100: Loading version 0.2 ...
ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
usbcore: registered new interface driver cdc_acm
cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
usbcore: registered new interface driver usb-storage
usbcore: registered new interface driver usbserial
usbcore: registered new interface driver usbserial_generic
usbserial: USB Serial support registered for generic
udc-core: couldn't find an available UDC - added [g_serial] to list of pending d                                                                                                                                                             rivers
mousedev: PS/2 mouse device common for all mice
i2c /dev entries driver
i2c_aspeed 1e78a000.i2c: i2c controller registered, irq 21
i2c_aspeed i2c-0: i2c bus 0 registered, irq 214
i2c_aspeed i2c-1: i2c bus 1 registered, irq 215
i2c_aspeed i2c-2: i2c bus 2 registered, irq 216
i2c_aspeed i2c-3: i2c bus 3 registered, irq 217
i2c_aspeed i2c-4: i2c bus 4 registered, irq 218
i2c_aspeed i2c-5: i2c bus 5 registered, irq 219
i2c_aspeed i2c-6: i2c bus 6 registered, irq 220
i2c_aspeed i2c-7: i2c bus 7 registered, irq 221
i2c_aspeed i2c-8: i2c bus 8 registered, irq 222
i2c_aspeed i2c-9: i2c bus 9 registered, irq 223
i2c_aspeed i2c-10: i2c bus 10 registered, irq 224
i2c_aspeed i2c-11: i2c bus 11 registered, irq 225
i2c_aspeed i2c-12: i2c bus 12 registered, irq 226
i2c_aspeed i2c-13: i2c bus 13 registered, irq 227
usbcore: registered new interface driver usbhid
usbhid: USB HID core driver
ip_tables: (C) 2000-2006 Netfilter Core Team
NET: Registered protocol family 17
hctosys: unable to open rtc device (rtc0)
Freeing unused kernel memory: 180K (804f5000 - 80522000)
This architecture does not have kernel memory protection.
grep: /sys/class/mtd/*/name: No such file or directory
cp: omitting directory '/run/mtd:u-boot-env'
grep: /sys/class/mtd/*/name: No such file or directory
grep: /sys/class/mtd/*/name: No such file or directory
cat: can't open '/run/fw_env': No such file or directory
cat: can't open '/run/fw_env': No such file or directory
rofs = squashfs rwfs = jffs2
mount: mounting /dev/mtdblock on run/initramfs/ro failed: No such file or direct                                                                                                                                                             ory
mount: mounting /dev/mtdblock on run/initramfs/rw failed: No such file or direct                                                                                                                                                             ory

Mounting read-write /dev/mtdblock filesystem failed.  Please fix and run
        mount /dev/mtdblock run/initramfs/rw -t jffs2 -o rw
`

amboar, 2016-08-12:
@rednewguy Try the commit from https://github.com/openbmc/linux/pull/93, which adds the flash layout description to the ast2500-evb dts. You can add a patch to openbmc/openbmc's linux-obmc package via `SRC_URI` in the `.bbappend` file.

There's some contention as to whether https://github.com/openbmc/linux/pull/93 is the approach we want to take, so it hasn't yet been merged.

rednewguy, 2016-08-12:
@amboar I did try making this change to my local sandbox couple days back. I figured this out by comparing dts of ast2400 and ast2500, and I wasn't aware of this pending merge request until now (read as new to git, yocto and openbmc). 
I also modified INITRD_IMAGE_ENTRYPOINT from 0x40800000 to 0x80800000 in meta-phosphor/classes/obmc-phosphor-image_types_uboot.bbclass. Both didn't help. 
I only ran bitbake obmc-phosphor-image after making these changes. Should I be doing something else to take these changes into effect?

mdmillerii, 2016-08-12:
Actually I see /sys/class/mtd/name does not exist according to grep.  That indicates the kernel is not finding any mtd devices including the whole chip.  I don't see message from spi-nor indicating it found a chip or attempted to probe.  Please check the dtb has the flash node, and that the kernel has the driver configured.

Your initrd appears to be unpacking so your entry point is ok.  I think it uses the load address but not the entry point.

amboar, 2016-08-14:
@rednewguy if you haven't yet tried the linked patch to can you please do so? It has worked for others (presumably for GHF who sent the pull-req, I've boot-tested it in qemu, and at least one other developer was asking about this problem). It's not clear to me from your response how you made your changes, or whether the changes you made were correct (but as the flash still wasn't found I imagine something wasn't right). If you need clearer instruction on integrating GHF's patch into yocto then I can probably cook up a patch for that as well.

@mdmillerii suggests the initrd entry point is okay, so lets change one thing at a time. Regardless of the value in `meta-phosphor/classes/obmc-phosphor-image_types_uboot.bbclass` the boot log shows it is being loaded at a sane location (`Load Address: 81000000` and `Entry Point: 81000000`)

GHF, 2016-08-16:
FYI the latest master is already using the dev-4.7 branch of the kernel (https://github.com/openbmc/openbmc/commit/b3530fb14f1314f5ef92751ceb31ef5b853d8e2a) so the change from openbmc/linux#93 should already be in the DTS (https://github.com/openbmc/linux/commit/cc9cc99dfdbcff1c0001fe4de16430ba12ec642c#diff-dc07dbf5b275d6c049e69fff597c2e5dR32).

I just tried building from master and the image does boot on my AST2500 EVB after flashing the entire `image-bmc` over the boot SPI flash.

amboar, 2016-08-17:
@GHF good point, looks like @williamspatrick committed the version-bump yesterday. @rednewguy so if your openbmc tree is at least b3530fb14f1314f5ef92751ceb31ef5b853d8e2a it should be resolved.

closed 2016-08-17 by williamspatrick