openbmc/openbmc #508: obmc fail to boot after AC cycle on Palm309

williamli80, 2016-08-12:
Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 palmetto ttyS4

palmetto login: root
Password:
root@palmetto:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 42:F9:C3:01:C0:5F
          inet addr:9.3.164.177  Bcast:9.3.164.255  Mask:255.255.255.128
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1775 errors:0 dropped:0 overruns:0 frame:0
          TX packets:95 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:94333 (92.1 KiB)  TX bytes:6606 (6.4 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@palmetto:~#
root@palmetto:~# random: nonblocking pool is initialized
ftgmac100: NCSI interface down
ftgmac100: NCSI interface up
ftgmac100: NCSI interface down
ftgmac100: NCSI interface up
ftgmac100: NCSI interface up
Unable to handle kernel NULL pointer dereference at virtual address 00000084
pgd = dd9d4000
[00000084] *pgd=5f653831, *pte=00000000, *ppte=00000000
Internal error: Oops: 17 [#1] ARM
Modules linked in:
CPU: 0 PID: 638 Comm: inarp Not tainted 4.6.4-openbmc-20160722-1 #1
Hardware name: ASpeed SoC
task: df530b60 ti: dd9c8000 task.ti: dd9c8000
PC is at sock_poll+0x14/0xa0
LR is at do_sys_poll+0x210/0x450
pc : [<c01f5bd4>]    lr : [<c00a4288>]    psr: a0000013
sp : dd9c9bc0  ip : 00004000  fp : df5622c0
r10: beba2cbc  r9 : dd9c9c04  r8 : 00000000
r7 : dd9c9d04  r6 : df5622c0  r5 : df0d7160  r4 : 00000000
r3 : 00000000  r2 : c01f5bc0  r1 : dd9c9d04  r0 : 00000000
Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 0005317f  Table: 5d9d4000  DAC: 00000051
Process inarp (pid: 638, stack limit = 0xdd9c8190)
Stack: (0xdd9c9bc0 to 0xdd9ca000)
9bc0: 00000000 00000000 00000000 dd9c9c0c 00000000 c00a4288 00000000 00000000
9be0: 00000000 00000000 00000000 00000000 dd9c9c1c df5622c0 00000000 00000002
9c00: 00000000 00000000 00000002 00000003 00000001 00000004 00000001 83890309
9c20: 00000000 00000000 dd981ce0 00000000 dd981ce0 00000000 df4c2000 00000000
9c40: 00000000 dd981ce0 ddbf24b8 c025edd4 df4c2000 c00673a0 00000141 c025edd4
9c60: df4c2000 00000001 c03f9538 00000100 01890309 83890309 dfbe800c dfbe8008
9c80: 00000000 c03fd73c c03fdcb4 00000000 00000000 c03fd73c c03fdcb4 dd981ce0
9ca0: 00000040 ddbf24b0 00000032 dd981ce0 df4c2000 df4c2048 dd981ce0 00000608
9cc0: df4c205c c03e98c8 c027c16c c025ef2c df689c20 c027c430 dd981ce0 df689c20
9ce0: dd90a310 df4c2000 00000000 c03e98c8 dd981ce0 00000608 df4c2000 c025ee28
9d00: df4c2048 00000000 00000019 00000000 df530b60 00000001 00000000 00000002
9d20: df5622c0 00000019 00000000 dd9c9d04 c00a2e90 df657960 df657960 df657960
9d40: df562d60 00000019 00000000 dd9c9d04 c00a2e90 df6579a0 df6579a0 df6579a0
9d60: df4c2890 00000000 00000001 00000000 df4c2890 c01d1184 00000000 ffffffff
9d80: 00000001 00000040 20000013 df4c2890 c01d0c88 dd9c9dc0 dd9c9dc8 0000012c
9da0: c03fdf51 c03fdd40 00000040 c020bae0 c03fdf51 ffffb63a c03ee520 c03e83b0
9dc0: dd9c9dc0 dd9c9dc0 dd9c9dc8 dd9c9dc8 00000000 dd9c8000 40000003 00400100
9de0: 00000008 c0402be0 0000000a c0402c0c dd9c8000 c0018e20 df41e0c0 df41e0c0
9e00: df41e0d0 ffffb639 dd9c9eec 0000002a 0000002a ddbf24cc dd9c9edc beba2d08
9e20: 0000002a 00000000 00000051 c016b310 00000000 c016f180 0000002a 00000000
9e40: 00000040 0000002a dd9c9eec df689c20 0000002a c02028e0 df689c20 0000002a
9e60: dd90a000 0000002a dd90a000 dd9c9ee4 00000000 c02026ac df689c20 c027faf0
9e80: 00000003 00000000 0000002a beba2cb8 dd9c9f10 00000000 00000000 beba2ce0
9ea0: 00000012 00000000 00000051 c016b310 00000012 c01f6580 00000000 beba2ce0
9ec0: df0d7160 c01f7bac dd9c9eec dd9c8000 40000003 0000002a 00000000 beba2d08
9ee0: 0000002a dd9c9f10 00000012 00000000 00000000 00000000 dd9c9ee4 00000000
9f00: 00000000 00000000 c0402c20 00000000 06080011 00000002 06010001 005e0000
9f20: df403b01 00000001 00000000 00000000 ffffffff 00000000 df530b60 df530b60
9f40: df530b60 c0032620 df440018 df530b60 00000053 00000000 dd952c20 c002c8a8
9f60: 00000000 ffffffff beba2ccc 00000002 beba2cbc c000a424 dd9c8000 00000000
9f80: 000115bc c00a456c beba2cbc 00000002 ffffffff beba2cbc beba2ccc 00022024
9fa0: 000000a8 c000a280 beba2cbc beba2ccc beba2cbc 00000002 ffffffff 00000001
9fc0: beba2cbc beba2ccc 00022024 000000a8 beba2d08 beba2d34 00011584 000115bc
9fe0: 00000000 beba2c8c 000109fc 47be9cac 60000010 beba2cbc 11a02808 01020044
[<c01f5bd4>] (sock_poll) from [<c00a4288>] (do_sys_poll+0x210/0x450)
[<c00a4288>] (do_sys_poll) from [<c00a456c>] (SyS_poll+0x58/0xc0)
[<c00a456c>] (SyS_poll) from [<c000a280>] (ret_fast_syscall+0x0/0x38)
Code: e5905078 e1a06000 e5950014 e1a07001 (e5904084)
---[ end trace c7b27769547ce6bb ]---

mdmillerii, 2016-08-12:
@gwshan @shenki this seems to be a NULL pointer (`00000084`) in `sock_poll` from `sys_poll` on the 4.6 branch.  `4.6.4-openbmc-20160722-1`.

A quick search finds similar poll oops at 00000000000000e4 on 64 bit architectures 
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1596070
https://lists.launchpad.net/kernel-packages/msg183037.html

It looks like it doesn't happen in 4.7 but bisect has not yet been completed:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1589837/comments/9

gtmills, 2018-05-23:
@williamli80 @mine260309 Can this be closed?

shenki, 2018-05-24:
We no longer run this kernel.

closed 2018-05-24 by shenki