openbmc/openbmc #509: Host soft-reboot puts BMC in a PINGable but NOT SSHable stage

agangidi53, 2016-08-15:
We ran into this weird bug on our 2 Barreleye PVT servers in Data Center. Host soft-reboot puts BMC in a PINGable but NOT SSHable stage. It is to be noted that after the soft-reboot when the hypervisor comes back up it issues ipmitool commands to make sure static IP is set . This ipmitool command to do a cold reset or assign a different static IP times out after host is up. 

This server is running an older version of firmware
BMC: 6.15 
BIOS: 0.2.0
CPLD : 1.0.2

I am trying to replicate this on a newer combination of firmware. It's hard for me to get BMC logs since server is remote.  

If someone has seen an issue like this, please let us know.

gtmills, 2018-05-23:
@agangidi53 @geissonator Can we close? Barreleye is no longer a supported machine.

geissonator, 2018-05-23:
Yeah, lets close this one out and reopen if we can recreate with latest software. I haven't heard of an issue like this on barreleyes follow-on, Zaius.

closed 2018-05-23 by geissonator