openbmc/openbmc #510: Documentation for User Management

anoo1, 2016-08-15:
After the migration to Gerrit, pending pull requests were closed like the documentation update for User Management: https://github.com/openbmc/docs/pull/19

Opening an issue to get this pull request (and any other documentation one) re-submitted since there's already users asking how to do user management on openbmc.

amboar, 2016-08-15:
Should we just send the patches to gerrit instead? Were the changes requested in https://github.com/openbmc/docs/pull/19 ever made?

Also, do you know of any other documentation pull-reqs that got missed? If so I think it would be good to list them, that way we know when the issue can be closed.

anoo1, 2016-08-16:
Checking the closed pull requests, the user management one is the only one that didn't get merged on time. A new patch to gerrit would be needed with the contents of the closed pulled request.

anoo1, 2018-05-23:
There's a user management proposal doc here: https://gerrit.openbmc-project.xyz/#/c/8440/

gtmills, 2018-06-07:
Will close this after  https://gerrit.openbmc-project.xyz/#/c/8440/ is merged

gtmills, 2018-06-21:
https://gerrit.openbmc-project.xyz/#/c/8440/ was merged. Closing

closed 2018-06-21 by gtmills

rfrandse, 2018-08-17:
move to https://github.com/ibm-openbmc/dev/issues/88 for completeness