openbmc/openbmc #512: Network Service become unavailable during AC-Cycle stress test

johnhcwang, 2016-08-16:
When this issue occurred, we can't restart eth0 device with SIOCSIFFLAGS error.
The details can be referred to the original issue [phosphor-networkd/issues/18](https://github.com/openbmc/phosphor-networkd/issues/18).

Configurations under the test:
- obmc 1.0.4
- static IP
- kernel: openbmc-4.4-20160804-1

anoo1, 2016-08-16:
Hi @gwshan , Joel mentioned you could help us look at this issue. Their systems are running the network driver patch, but they still see an issue after about 90 AC cycles. The latest log is in the issue mentioned above. Thanks!

shenki, 2016-08-30:
@gwshan wrote in phosphor-networkd/issues/18:
> I had a talk with @johnhcwang. It's likely the story: No NCSI channels are probed
> from BCM5719 when bringing up the network interface for the first time. As the
> NCSI channel enumeration is done for once, the network interface doesn't have
> workable NCSI device assocated, meaning the network interface won't work later on.
> 
> I will provide tentative patch for @johnhcwang to try as discussed, thanks!

staneyshen, 2016-09-08:
@gwshan provides the workaround and I'll provide it for SIT to do the stress test.

staneyshen, 2016-09-12:
@gwshan following is the test result for your reference.
````
==AC power cycle Test Procedure:==
1. AC on
2. SSH BMC by LAN to check BMC status.
3. UUT OS on(random auto power on)
4. Have a counter on OS.
4. Power off by UUT self.
*5. SSH BMC by LAN to check OS power off status.
6. AC off
(9 UUT for the test, once anyone UUT has error the test will stop)

==Test Result:==
1 UUT BMC LAN occurred dis-link status. (occurred at 213th cycle and step 5 in UUT1) 
````

gwshan, 2016-09-12:
@staneyshen, It's notable we had below commands included in uboot "bootcmd"
environmental variable to do PCI slot reset. Those commands are doing PCI
slot reset through GPIO-B5 PIN. Bit#13 in SCU[80] is cleared by default,
meaning GPIO-B5 is a GPIO PIN by default.

setenv bootcmd 'mw.l 0x1e780004 0x00002000; \
                         mw.l 0x1e780000 0x204fe5b3;  \
                         sleep 1;                                  \
                         mw.l 0x1e780000 0x204fc5b3;  \
                         bootm 20080000 20300000'
saveenv

@staneyshen, please confirm: the NCSI interface is down in step (2) without
above workaround? If so, the above around helps and needed by uboot. The
problem reported from last update would be different issue and I will talk
to @staneyshen to add more debugging logs to understand the last issue.

staneyshen, 2016-09-12:
@gwshan We'll test the workaround again to see whether the workaround always failed at step 6. If yes, we'll remove workaround then test again to see the failed situation is at step 2 or step 6.

staneyshen, 2016-09-14:
@gwshan Update the workaround test result for your reference.
3rd test, UUT2 occurred at step 2 at 118th cycle on 9/14.

It seems the workaround still happen the same fail situation.

gwshan, 2016-09-14:
Thanks to @staneyshen for his kindly helps on this issue. According to the information and logs collected from one of the failing BMC, it seems the BCM5718 stopped responsing on DP (Deselect Package) and subsequent commands. The packet statistics from "ifconfig eth0" shows 24 packets were sent and 2 packets were received before hitting the failure. Those 24 Tx packets are 8 DPs, 8 SPs and 8 fllowup DPs that were sent during the probing period. Those 2 Rx packets are responses to DPs as we have only one package on the system. Anyway, the BCM5718 didn't response to the SP sent during the probing time, meaning the only one available package wasn't discovered successfully.

On the failing BMC, @staneyshen helps run below commands and it returns 1 (error). It indicates the available NCSI packages/channels are not discovered successfully. I'm coming up with an enhancement to reset BCM5718 when no available package/channel are probed and retry the probing for once, in order to improve the reliability.

failing_bmc# ifconfig eth0 down; ifconfig eth0 up; echo $?
1

==========
Related logs
==========

Booting Linux on physical CPU 0x0
Linux version 4.4.16-openbmc-4.4-20160804-1 (ken@ken-PowerEdge-R210-II) (gcc version 4.9.3 (GCC) ) #1 Wed Aug 24 15:44:52 CST 2016
      ..
      ..
platform 1e630000.spi:flash: n25q512ax3 (65536 Kbytes)
ftgmac100 1e660000.ethernet: Using NCSI interface
ftgmac100 1e660000.ethernet: Read MAC address from chip 52:a6:f6:0e:6c:02
ftmac100: Loading version 0.2 ...
       ..
       ..
root@barreleye:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 52:A6:F6:0E:6C:02  
         inet addr:10.62.2.102  Bcast:10.62.2.255  Mask:255.255.255.0
         UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
         RX packets:2 errors:0 dropped:0 overruns:0 frame:0
         TX packets:24 errors:0 dropped:0 overruns:0 carrier:0
         collisions:0 txqueuelen:1000
         RX bytes:100 (100.0 B)  TX bytes:1488 (1.4 KiB)

lo        Link encap:Local Loopback  
         inet addr:127.0.0.1  Mask:255.0.0.0
         UP LOOPBACK RUNNING  MTU:65536  Metric:1
         RX packets:111120 errors:0 dropped:0 overruns:0 frame:0
         TX packets:111120 errors:0 dropped:0 overruns:0 carrier:0
         collisions:0 txqueuelen:1
         RX bytes:8445120 (8.0 MiB)  TX bytes:8445120 (8.0 MiB)

staneyshen, 2016-10-13:
Hi @gwshan 
Update the test status for your reference.
We used your patches that you provided last week with v1.0.5 for AC Cycle stress test. it passed over 300 cycles, I think the solution works for this issue, we're going to do the test with formal release version again.

gwshan, 2016-10-13:
@staneyshen, thank you very much for the continuous testing which is very
time consuming. Sometimes, we just hit the issue for once in one week's
testing. The last kernel (dev-4.7) also needs the fix. I will push it
upstream and then backport it to 4.7 and 4.4 branches. Thanks again for
your efforts on this!

gwshan, 2016-10-21:
A series backporting fixes from upstream or linux.net git repo has been posted. Great thanks again to Staney and John for their continuous testing:

https://patchwork.ozlabs.org/patch/684813/

staneyshen, 2016-11-02:
We're verifying @gwshan 's NCSI patches with our v10.40.A version, and no problem found after 300+ continuous rounds of testing. The situation is improved greatly and we always stopped after 70+ rounds of testing before.

gwshan, 2016-11-02:
@staneyshen, thanks for updating your testing result. I guess it can be closed, but I'd received John Wang's confirm before closing it.

johnhcwang, 2016-11-02:
Hi @gwshan , thanks for your great support.
I believe the issue was fixed in kernel 4.4-20160804-1 plus these patches on Barreleye.

Hi @anoo1 , as I mentioned in the meeting, there's another issue using latest kernel 4.4-20161021-1.
I'll collect the information and create a new issue for it.

closed 2016-11-02 by johnhcwang