openbmc/openbmc #513: pgood gpio has incorrect value.

williamspatrick, 2016-08-18:
With the 4.7.x kernel, the PGOOD sensor for both Palmetto and Barreleye seems "stuck" on.  Even if we AC cycle the box, when we get up to BMC_READY state the GPIO value appears on.

```
root@palmetto:/sys/class/gpio/gpio343# uname -a
Linux palmetto 4.4.16-openbmc-4.4-20160804-1 #1 Thu Aug 4 05:13:31 UTC 2016 armv5tejl GNU/Linux
root@palmetto:/sys/class/gpio/gpio343# cat value active_low direction edge
0
0
in
none
```
```
# uname -a
Linux barreleye 4.7.0-a5a7a520d4945df3f5ffa7fb9ed5856f472b9fc1 #1 Mon Aug 15 14:17:34 CDT 2016 armv5tejl GNU/Linux
root@barreleye:/sys/class/gpio/gpio343# cat value active_low direction edge
1
0
in
none
```

amboar, 2016-08-19:
Okay, that shoots a hole in the active-low hypothesis. I'll dig into it.

amboar, 2016-08-22:
As a data-point, for the palmetto I have with me in the office:

```
root@palmetto:/sys/class/gpio/gpio343# uname -a
Linux palmetto 4.4.16-00003-gcffb62f94044 #7 Mon Aug 22 16:24:56 ACST 2016 armv5tejl GNU/Linux
root@palmetto:/sys/class/gpio/gpio343# cat /etc/os-release 
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=v0.8.1-6-g0910c3fe90a3-dirty
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=v0.8.1
root@palmetto:/sys/class/gpio/gpio343# cat value active_low direction edge
1
0
in
none
```

amboar, 2016-08-23:
Okay, as a follow up to the observations above I have rebuilt and (re)booted both `cffb62f94044` (`shenki/dev-4.4`) and `dd134cf30a5` (`openbmc-4.4-20160819-1`) 5 times, and each time observed the undesirable `pgood` behaviour for both cases.

Not yet sure what to make of the inconsistent behaviour with `openbmc-4.4-20160819-1`. I will work my way back through earlier dev-4.4 tags to find a functioning tree.

amboar, 2016-08-23:
Okay. This looks like we're stuck between a rock and a hard place! `openbmc-4.4-20160804-1` behaves correctly with respect to PGOOD. Between that and `shenki/dev-4.4` there are only 3 commits, one of them relevant to GPIO in general.

But before we dive deeper, it's not necessarily PGOOD that's problematic. PGOOD is set high here because the host is genuinely powered on. The host gets switched on by BMC_PWRBTN_OUT_N which is GPIOE1, a.k.a GPIO pin 33, a.k.a GPIO_SYSFS export value 353.

Reverting openbmc/linux@aa65fa6cf3c3a558e009fb2c934b85974c2733db `gpio/aspeed: Write value on switch to output` gives us the expected behaviour (of not powering on the host): 

```
root@palmetto:/sys/class/gpio/gpio343# uname -a
Linux palmetto 4.4.18-00001-g929a1cc79783 #552 Tue Aug 23 15:52:15 ACST 2016 armv5tejl GNU/Linux
root@palmetto:/sys/class/gpio/gpio343# cat /etc/os-release 
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=v0.8.1-6-g0910c3fe90a3-dirty
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=v0.8.1
(reverse-i-search)`': ^C
root@palmetto:/sys/class/gpio/gpio343# cat value active_low direction edge 
0
0
in
none
```
But, that commit resolved issue openbmc/linux#98. Undoing the revert and adding WARN() in the right spot shows:

```
------------[ cut here ]------------
WARNING: CPU: 0 PID: 647 at drivers/gpio/gpio-aspeed.c:130 __aspeed_gpio_set+0x50/0x74()
Setting gpio 33 to 0, was 1
Modules linked in:
CPU: 0 PID: 647 Comm: power_control.e Not tainted 4.4.18-dirty #551
Hardware name: ASpeed SoC
[<c000f770>] (unwind_backtrace) from [<c000d368>] (show_stack+0x10/0x14)
[<c000d368>] (show_stack) from [<c0016a6c>] (warn_slowpath_common+0x80/0xa8)
[<c0016a6c>] (warn_slowpath_common) from [<c0016ad0>] (warn_slowpath_fmt+0x3c/0x64)
[<c0016ad0>] (warn_slowpath_fmt) from [<c0176988>] (__aspeed_gpio_set+0x50/0x74)
[<c0176988>] (__aspeed_gpio_set) from [<c0176aa8>] (aspeed_gpio_dir_out+0x60/0x74)
[<c0176aa8>] (aspeed_gpio_dir_out) from [<c01746e8>] (_gpiod_direction_output_raw+0xc0/0xf8)
[<c01746e8>] (_gpiod_direction_output_raw) from [<c0175c1c>] (direction_store+0x58/0xbc)
[<c0175c1c>] (direction_store) from [<c0104e7c>] (kernfs_fop_write+0x12c/0x190)
[<c0104e7c>] (kernfs_fop_write) from [<c00afb78>] (__vfs_write+0x28/0xf4)
[<c00afb78>] (__vfs_write) from [<c00b0284>] (vfs_write+0xa8/0x130)
[<c00b0284>] (vfs_write) from [<c00b09c4>] (SyS_write+0x50/0xa0)
[<c00b09c4>] (SyS_write) from [<c000a3a0>] (ret_fast_syscall+0x0/0x38)
---[ end trace b0c3e41a4ead88c2 ]---
ftgmac100: NCSI interface down
------------[ cut here ]------------
```
Tracing backwards from `__aspeed_gpio_set()` we find the `value` parameter comes from `direction_store()` from `gpiolib-sysfs.c`:

```
static ssize_t direction_store(struct device *dev,
		struct device_attribute *attr, const char *buf, size_t size)
{
	struct gpiod_data *data = dev_get_drvdata(dev);
	struct gpio_desc *desc = data->desc;
	ssize_t			status;

	mutex_lock(&data->mutex);

	if (sysfs_streq(buf, "high"))
		status = gpiod_direction_output_raw(desc, 1);
	else if (sysfs_streq(buf, "out") || sysfs_streq(buf, "low"))
		status = gpiod_direction_output_raw(desc, 0);
	else if (sysfs_streq(buf, "in"))
		status = gpiod_direction_input(desc);
	else
		status = -EINVAL;

	mutex_unlock(&data->mutex);

	return status ? : size;
}
static DEVICE_ATTR_RW(direction);
```
So we find `direction_store()` picks 0 to poke into BMC_PWRBTN_OUT_N via `gpiod_direction_output_raw()`, which consequently turns the host on and as I understand, asserts the GPIO we've mapped to PGOOD (CFAM_RESET_N_B in the datasheet).

Now, the documentation for `gpiod_direction_output_raw()` says:

```
/**
 * gpiod_direction_output_raw - set the GPIO direction to output
 * @desc:	GPIO to set to output
 * @value:	initial output value of the GPIO
 *
 * Set the direction of the passed GPIO to output, such as gpiod_set_value() can
 * be called safely on it. The initial value of the output must be specified
 * as raw value on the physical line without regard for the ACTIVE_LOW status.
 *
 * Return 0 in case of success, else an error code.
 */
```
Which contrasts with `gpiod_direction_output()`:

```
/**
 * gpiod_direction_output - set the GPIO direction to output
 * @desc:	GPIO to set to output
 * @value:	initial output value of the GPIO
 *
 * Set the direction of the passed GPIO to output, such as gpiod_set_value() can
 * be called safely on it. The initial value of the output must be specified
 * as the logical value of the GPIO, i.e. taking its ACTIVE_LOW status into
 * account.
 *
 * Return 0 in case of success, else an error code.
 */
```

So, this reqiures userspace to change to write "high" to the sysfs `direction` property rather than "out". We cannot play tricks with `active_low` because `gpiod_direction_output_raw()` is used rather than `gpiod_direction_output()`, which would have allowed for a slightly different userspace solution (set `active_low`, continue to write "out" to `direction`, and flip the values written to `value`).

amboar, 2016-08-23:
Given the above, it looks like we'll also need to patch openbmc/skeleton's `gpio_init()` in `libopenbmc_intf/gpio.c`:

```
$ git diff
diff --git a/libopenbmc_intf/gpio.c b/libopenbmc_intf/gpio.c
index 9e8164dd5353..a27e0248bb55 100644
--- a/libopenbmc_intf/gpio.c
+++ b/libopenbmc_intf/gpio.c
@@ -147,7 +147,10 @@ int gpio_init(GDBusConnection *connection, GPIO* gpio)
                        }
                }
                const char* file = "edge";
-               if (strcmp(gpio->direction,"in")==0 || strcmp(gpio->direction,"out")==0)
+               if (strcmp(gpio->direction,"in")==0 ||
+                               strcmp(gpio->direction,"out")==0 ||
+                               strcmp(gpio->direction,"low")==0 ||
+                               strcmp(gpio->direction,"high")==0)
                {
                        file = "direction";
                }
```

geissonator, 2016-08-23:
Pushed a patch up here https://gerrit.openbmc-project.xyz/#/c/261/ but based on testing from @anoo1 we seemed to see the same issue.  @amboar, it wasn't clear to us from your comments, is more then the userspace change to skeleton required here?

williamspatrick, 2016-08-23:
There is no 'direction' of low / high in the config files.  The POWER_BUTTON has a direction of 'both'.  Is this what was intended to be changed?

williamspatrick, 2016-08-23:
https://github.com/openbmc/skeleton/blob/master/configs/Barreleye.py#L569

amboar, 2016-08-23:
@williamspatrick @geissonator yes, the system configuration script must be changed, but not POWER_BUTTON, rather POWER_PIN (GPIOE1). At least that's the case for Palmetto, I haven't checked the Barreleye schematic.

amboar, 2016-08-23:
@williamspatrick the fact that we specify 'direction' as 'both' for POWER_BUTTON is a poor and confusing overloading of concepts. It simultaneously sets the pin to 'in' AND writes 'both' to 'edge', configuring the pin for riding and falling edge interrupts. This allows us to implement the long/short press behaviours. We really must patch things so a separate key is used for interrupt configuration; this 17 character shortcut gives behavior that isn't obvious and could lead to hard-to-diagnose behavior in the face of typos or other mistakes.

geissonator, 2016-08-24:
I tried a quick hack in master where I did the following:

git diff c90eb8bc09eb8cbae27f8e3f2104f975d4ac2496^ c90eb8bc09eb8cbae27f8e3f2104f975d4ac2496
diff --git a/drivers/gpio/gpio-aspeed.c b/drivers/gpio/gpio-aspeed.c
index e48bd1e..4c52414 100644
--- a/drivers/gpio/gpio-aspeed.c
+++ b/drivers/gpio/gpio-aspeed.c
@@ -181,7 +181,7 @@ static int aspeed_gpio_dir_out(struct gpio_chip *gc,
        reg = ioread32(bank_val_reg(gpio, bank, GPIO_DIR));
        iowrite32(reg | GPIO_BIT(offset), bank_val_reg(gpio, bank, GPIO_DIR));
 
-       __aspeed_gpio_set(gc, offset, val);
+       //__aspeed_gpio_set(gc, offset, val); 
        spin_unlock_irqrestore(&gpio->lock, flags);

Still getting the "bad" reading after a power cycle.

root@palmetto:/sys/class/gpio/gpio343# cat value active_low direction edge
1
0
in
none
root@palmetto:/sys/class/gpio/gpio343# random: nonblocking pool is initialized
root@palmetto:/sys/class/gpio/gpio343# obmcutil state
 = HOST_BOOTED

amboar, 2016-08-24:
@geissonator I suspect there's an issue with GPIO and pinmux but I'm yet to confirm.

amboar, 2016-08-24:
@williamspatrick @geissonator 

Okay, as seems to be the way, several things are at play here.

`git diff obmc/dev-4.4 obmc/dev-4.7 -- drivers/gpio/gpio-aspeed.c` is uninteresting, showing pinctrl integration, some comments and other minor fixes. However, `git diff obmc/dev-4.4 HEAD -- arch/arm/mach-aspeed/aspeed.c` is much more fruitful, particularly:

```udiff
@@ -172,12 +115,6 @@ static void __init do_palmetto_setup(void)
        writel(0x30000E00, AST_IO(AST_BASE_LPC | 0x88));
        writel(0xFE0001FF, AST_IO(AST_BASE_LPC | 0x8C));
 
-       /* GPIO setup */
-       writel(0x13008CE7, AST_IO(AST_BASE_GPIO | 0x00));
-       writel(0x0370E677, AST_IO(AST_BASE_GPIO | 0x04));
-       writel(0xDF48F7FF, AST_IO(AST_BASE_GPIO | 0x20));
-       writel(0xC738F202, AST_IO(AST_BASE_GPIO | 0x24));
-
        /* SCU setup */
        writel(0x01C0007F, AST_IO(AST_BASE_SCU | 0x88));
 }
```

amboar, 2016-08-24:
Okay, there are a few things at play here.

The first is that the board-file configuration of the host power GPIO bank went missing from `do_palmetto_setup()` in the shift to openbmc/linux dev-4.7:

`git diff obmc/dev-4.4 obmc/dev-4.7 -- arch/arm/mach-aspeed/aspeed.c`:
```udiff
@@ -172,12 +115,6 @@ static void __init do_palmetto_setup(void)
        writel(0x30000E00, AST_IO(AST_BASE_LPC | 0x88));
        writel(0xFE0001FF, AST_IO(AST_BASE_LPC | 0x8C));
 
-       /* GPIO setup */
-       writel(0x13008CE7, AST_IO(AST_BASE_GPIO | 0x00));
-       writel(0x0370E677, AST_IO(AST_BASE_GPIO | 0x04));
-       writel(0xDF48F7FF, AST_IO(AST_BASE_GPIO | 0x20));
-       writel(0xC738F202, AST_IO(AST_BASE_GPIO | 0x24));
-
        /* SCU setup */
        writel(0x01C0007F, AST_IO(AST_BASE_SCU | 0x88));
 }
```
Specifically, the magic values for offsets 0x20 and 0x24. I'm not sure why it went missing.

Second, there is the patch for openbmc/linux#98 which fixed the LED behaviour, but broke assumptions made by userspace.

Third, I have a proof-of-concept openbmc/skeleton patch:

```udiff
diff --git a/configs/Palmetto.py b/configs/Palmetto.py
index f13cc4e59ae7..e035e77cf709 100644
--- a/configs/Palmetto.py
+++ b/configs/Palmetto.py
@@ -281,7 +281,7 @@ GPIO_CONFIG = {}
 GPIO_CONFIG['FSI_CLK']    =   { 'gpio_pin': 'A4', 'direction': 'out' }
 GPIO_CONFIG['FSI_DATA']   =   { 'gpio_pin': 'A5', 'direction': 'out' }
 GPIO_CONFIG['FSI_ENABLE'] =   { 'gpio_pin': 'D0', 'direction': 'out' }
-GPIO_CONFIG['POWER_PIN']  =   { 'gpio_pin': 'E1', 'direction': 'out'  }
+GPIO_CONFIG['POWER_PIN']  =   { 'gpio_pin': 'E1', 'direction': 'high'  }
 GPIO_CONFIG['CRONUS_SEL'] =   { 'gpio_pin': 'A6', 'direction': 'out'  }
 GPIO_CONFIG['PGOOD']      =   { 'gpio_pin': 'C7', 'direction': 'in'  }
 GPIO_CONFIG['BMC_THROTTLE'] = { 'gpio_pin': 'J3', 'direction': 'out' }
diff --git a/libopenbmc_intf/gpio.c b/libopenbmc_intf/gpio.c
index 9e8164dd5353..a27e0248bb55 100644
--- a/libopenbmc_intf/gpio.c
+++ b/libopenbmc_intf/gpio.c
@@ -147,7 +147,10 @@ int gpio_init(GDBusConnection *connection, GPIO* gpio)
                        }
                }
                const char* file = "edge";
-               if (strcmp(gpio->direction,"in")==0 || strcmp(gpio->direction,"out")==0)
+               if (strcmp(gpio->direction,"in")==0 ||
+                               strcmp(gpio->direction,"out")==0 ||
+                               strcmp(gpio->direction,"low")==0 ||
+                               strcmp(gpio->direction,"high")==0)
                {
                        file = "direction";
                }
```
The following configurations boot with the expected behaviour with respect to PGOOD:
* openbmc/linux@6d93f625315a1e9b51d017a78f8c0938a097ea85 with the openbmc/skeleton PoC
* openbmc/linux@6d93f625315a1e9b51d017a78f8c0938a097ea85 with openbmc/linux#98 reverted and the GPIO bank board-file initialisation re-applied (though the LEDs will not function correctly)

The follow configurations boot with the undesirable behaviour with respect to PGOOD:
* openbmc/linux@6d93f625315a1e9b51d017a78f8c0938a097ea85
* openbmc/linux@6d93f625315a1e9b51d017a78f8c0938a097ea85 with openbmc/linux#98 reverted

amboar, 2016-08-24:
The above is relevant to Palmetto. I still need to understand what is going on with Barreleye.

geissonator, 2016-08-24:
I took master, applied the arch/arm/mach-aspeed/aspeed.c patch (putting the changes back in), and did the skeleton POC and got all the way to the OS login prompt on my Palmetto!  Good stuff Andrew.

williamspatrick, 2016-08-24:
I have a proposed "permanent" solution in skeleton available here: https://gerrit.openbmc-project.xyz/#/c/272/

If this works on master we can backport to stable and then try to re-introduce the 4.4.x kernel update on stable.

geissonator, 2016-08-24:
I tried Patrick's official fix (+ the kernel fix from Andrew) today on our Palmetto.  Things were good on the initial reboot (i.e. we went to BMC_READY state) but a BMC reboot while the host OS was running caused a power off.  Patrick jumped on and saw the following:

The kernel is not maintaining the value / direction through a reset.
 ```
root@palmetto:/sys/class/gpio/gpio343# cat value active_low direction edge
0
0
in
none
root@palmetto:/sys/class/gpio/gpio343# cd ../gpio353
root@palmetto:/sys/class/gpio/gpio353# cat value active_low direction edge
1
0
in
none
```
pgood off, power off (in).  Even though right before the reboot I manually pushed power on and watched pgood.

amboar, 2016-08-25:
@geissonator - with @williamspatrick's skeleton fix you shouldn't need the patch for the kernel. Possibly that's interfering during the reboot.

amboar, 2016-08-25:
@geissonator I've reproduced the behaviour you observed with @williamspatrick's skeleton patch and the GPIO initialsation patch to the board file. I've also confirmed that *not* applying the board file patch leaves the host booted across BMC reboots. So the upshot: Don't apply the board-file change, only use the skeleton patch and a kernel built from the tip of dev-4.7.

geissonator, 2016-08-25:
Yep, confirmed the reboot now works and the host stays up with just skeleton patch and dev-4.7!

geissonator, 2016-08-25:
Also confirmed the same code worked on barreleye.  We def have a secondary IPL issue on both systems with master but that's for another issue...

williamspatrick, 2016-08-25:
Fix via 5b348caf5c45d875cbf077c4d54e9312abf72330

amboar, 2016-08-26:
> Yep, confirmed the reboot now works and the host stays up with just skeleton patch and dev-4.7!
Hmm, I think I have made some false assumptions above. Are you sure you got this to work? @legoater claimed he needed the board-file patch and I sure he's right. I didn't poke at the host console at all when I was testing, and I may have been tripped up by the host watchdog.

I've pushed a commit to amboar/linux@bfae690787ec639b746aa6ed69b126f1ed17a1ba that might resolve host boot and bmc-reboot-after-host-boot issues; it needs more testing.

Reopening to resolve the issues on master.

geissonator, 2016-08-30:
I'm a little confused on this last patch.  I've verified multiple times that with the latest Kernel + the skeleton patch from Patrick (https://github.com/openbmc/skeleton/commit/ed1368dab8568edabb6aa8baacfb90e661997cbd), that the host stays up when I issue a "reboot" on the BMC.  

Now after that, if I power down, I am unable to successfully power back on (https://github.com/openbmc/openbmc/issues/527).  Does this patch potentially address issue 527?

geissonator, 2016-08-30:
@amboar That patch (amboar/linux@bfae690) fixed the 2nd IPL issue!  I verified multiple power on/off's with it on my Palmetto using master + the patch.  

I tried a reboot of the BMC while petitboot was active, when I came back to the session, it appeared like petitboot had crashed or in some way become corrupted.  So that needs some more investigation.  A power off/on works fine from that state.

root@palmetto:~# obmc-console-client 

*** Error writing address 000000000000001e
*** Error writing address 000000000000000e
*** Error writing address 000000000000000e\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500
*** Error writing address 000000000000000enguage, h=help
*** Error writing address 000000000000000e
*** Error writing address 000000000000000e
*** Error writing address 000000000000000e
*** Error writing address 000000000000000e
*** Error writing address 000000000000000e
*** Error writing address 000000000000000e
*** Error writing address 000000000000000e
*** Error writing address 000000000000000e
*** Error writing address 000000000000000e
*** Error writing address 000000000000000e
*** Error writing address 000000000000000e
*** Error writing address 000000000000000e
000000000000000e    
000000000000000e  
0000000000000012  dmesg
*** Error writing address 0000000000000012
000000000000000e 
000000000000004e **************** ****************  |                |
13:mon> 
000000000000008e **************** ****************  |                |
13:mon> exit
cpu 0x13: Vector: 700 (Program Check) at [c0000003f9507790]
    pc: c000000000963d58: tg3_napi_enable+0x38/0x70
    lr: c00000000098466c: tg3_io_error_detected+0x2bc/0x300
    sp: c0000003f9507a10
   msr: 9000000000029033
  current = 0xc0000003f9480d80
  paca    = 0xc00000000fe83900	 softe: 0	 irq_happened: 0x01
    pid   = 242, comm = eehd
kernel BUG at include/linux/netdevice.h:492!
13:mon> 

Do we need a similar patch for barreleye (and the other platforms)?

shenki, 2016-08-30:
Thanks for verifying that the issue is GPIO setup related.

These patches are not the way to set GPIOs. We have a GPIO driver in the tree, and we need to use that to set them up.

Here's the values the code sets. These need to be converted into userspace requests for the GPIO.

I expect that it is only a subset of the values being written here that are actually required.

GPIO   | Value | Direction |  | GPIO   | Value | Direction
------ | ----- | --------- | ---- | ------ | ----- | ----------
GPIOD7 | 0x0 | input       | | GPIOH7 | 0x1 | output
GPIOD6 | 0x0 | input       | | GPIOH6 | 0x1 | output
GPIOD5 | 0x0 | input       | | GPIOH5 | 0x0 | input
GPIOD4 | 0x1 | input       | | GPIOH4 | 0x1 | input
GPIOD3 | 0x0 | input       | | GPIOH3 | 0x1 | input
GPIOD2 | 0x0 | input       | | GPIOH2 | 0x1 | output
GPIOD1 | 0x1 | output      | | GPIOH1 | 0x1 | output
GPIOD0 | 0x1 | output      | | GPIOH0 | 0x1 | output
GPIOC7 | 0x0 | input       | | GPIOG7 | 0x0 | input
GPIOC6 | 0x0 | output      | | GPIOG6 | 0x1 | input
GPIOC5 | 0x0 | output      | | GPIOG5 | 0x0 | output
GPIOC4 | 0x0 | output      | | GPIOG4 | 0x0 | output
GPIOC3 | 0x0 | input       | | GPIOG3 | 0x1 | output
GPIOC2 | 0x0 | input       | | GPIOG2 | 0x0 | input
GPIOC1 | 0x0 | input       | | GPIOG1 | 0x0 | input
GPIOC0 | 0x0 | input       | | GPIOG0 | 0x0 | input
GPIOB7 | 0x1 | output      | | GPIOF7 | 0x1 | output
GPIOB6 | 0x0 | output      | | GPIOF6 | 0x1 | output
GPIOB5 | 0x0 | output      | | GPIOF5 | 0x1 | output
GPIOB4 | 0x0 | input       | | GPIOF4 | 0x1 | output
GPIOB3 | 0x1 | input       | | GPIOF3 | 0x0 | input
GPIOB2 | 0x1 | output      | | GPIOF2 | 0x1 | input
GPIOB1 | 0x0 | output      | | GPIOF1 | 0x1 | output
GPIOB0 | 0x0 | input       | | GPIOF0 | 0x1 | input
GPIOA7 | 0x1 | input       | | GPIOE7 | 0x1 | input
GPIOA6 | 0x1 | output      | | GPIOE6 | 0x1 | input
GPIOA5 | 0x1 | output      | | GPIOE5 | 0x1 | input
GPIOA4 | 0x0 | output      | | GPIOE4 | 0x1 | input
GPIOA3 | 0x0 | input       | | GPIOE3 | 0x0 | input
GPIOA2 | 0x1 | output      | | GPIOE2 | 0x1 | input
GPIOA1 | 0x1 | output      | | GPIOE1 | 0x0 | output
GPIOA0 | 0x1 | output      | | GPIOE0 | 0x1 | input

amboar, 2016-08-31:
@shenki I agree, the kernel patches mentioned are not the ultimate solution to the issue. They are however steps in the process of understanding the problem. Once our understanding solidifies we can implement it all in userspace, or even before if it turns out to be easier. At least for me it's easier to patch the kernel and netboot with an existing initrd than to patch skeleton, patch openbmc, bitbake then bless the resulting cpio with mkimage for each test iteration.

Our GPIO register whacking in the patches above configures 8 banks, but your table only describes 4; I think bank E is relevant here. Further, is your table a simple expansion of the magic values for GPIO000/GPIO004, or was the intent to augment it with the signal names from the schematic? One of the columns is empty. Useful information would be the set of GPIOS that userspace isn't configuring that were being touched by the kernel in dev-4.4

shenki, 2016-08-31:
There are two tables side by side, for banks A, B, C, D, E, F, G, H. There's an empty column because I couldn't work out how to put a divider between the two tables in github; feel free to edit to fix this.

You're correct; we don't need all of these GPIOs set. The task is to work out which are important, document them, and set them in userspace. The same work will need to happen for Barreleye and other platforms as @geissonator suggested.

Give this a userspace task I will leave it to @williamspatrick to assign this work.

amboar, 2016-08-31:
> There are two tables side by side, for banks A, B, C, D, E, F, G, H. There's an empty column because I couldn't work out how to put a divider between the two tables in github; feel free to edit to fix this.

I apologise! It looks like I missed the invisible horizontal scroll bar in the mobile browsing view. Will check for all the invisible things next time.

geissonator, 2016-08-31:
Hey guys, thanks for all the analysis.   What's changed so this doesn't work any longer in master?  (i.e. everything works fine in the stable branch).  I'm all for doing things right, but I'm also for getting thing back and working ASAP so we can take the time to do it right.

shenki, 2016-08-31:
Gather round the 🔥 for some OpenBMC history.

When bringing up OpenBMC on the Palmetto and Barreleye we needed to set the multi-function pin state of the chip. This is the bit that assigns IP blocks (i2c, ethernet, gpio, etc) to external pins (actually balls) on the SoC. Norm hacked this into the skeleton code.

At the same time, on the other side of the world, we were bringing up the kernel. Mysterious behaviour abounded, one of the more annoying bugs was a failure to correctly reboot the BMC.

Having a bunch of userspace code poking at MMIO registers at the same time as the kernel was a warning sign. There's no ordering or mutal exclusion between the kernel drivers and userspace changes. Also, it meant different kernel behaviour depending on the userspace version. Norm and I chose to add the hacks to the board file, at `arch/arm/mach-aspeed/aspeed.c`, using the machine compatible property to set up the Barreleye and Palmetto specific state. Now the hardware bitbanging would live in the kernel, and userspace would use the kernel APIs to do it's thing. This meant we could debug kernel failures in the kernel, independent of userspace issues.

Post Barreleye, the goal was to write proper drivers for all of bits of IP allowing the board file would go slowly lose lines of code until eventually it would be deleted. The move from dev-4.4 (stable) to dev-4.7 (master) introduced @amboar's pinmux driver, which handles the multi-function pin table configuration. This means we no longer need to set those registers, nor the GPIO registers, in the board file. They can be moved out to the device tree and/or request by userspace as required.

I removed them, testing that the BMC still worked correctly as I went. I then submitted the tree for testing and review, which happened on the list (https://lists.ozlabs.org/pipermail/openbmc/2016-August/004372.html). When everyone agreed it was good to go, Patrick reved the kernel to the dev-4.7 tree.

What we all missed was verifying that booting worked with the openbmc userspace both before and after reving the kernel. So we ended up with a broken mess for a time.

We could have avoided the mess by doing the correct thing in the first place, but then we might have never shipped Barreleye on time.

So now it's time to pay out the technical debt and have the system correctly request it's GPIOs, either through the device tree or in userspace. I think this will make it easier to keep working, and if it does break, easier to fix the failures.

geissonator, 2016-08-31:
Thanks for the overview Joel.  The end goal with the MRW work from Matt is that this is all auto-magically done via that process.

For now, based on your input above and some discussion with the team, it seems like the best move is for me to update the device tree files (i.e. ./arch/arm/boot/dts/aspeed-bmc-opp-palmetto.dts) with the appropriate default settings, being careful to not touch the GPIO pins that we put out in sysfs (i.e. openbmc/skeleton/configs/Palmetto.py - GPIO_CONFIG)?

shenki, 2016-08-31:
Afaict the device tree can be used to configure GPIOs that you wish to remain a single value, as I have done with the FUNC_MODE[0:2] pins: https://github.com/openbmc/linux/blob/dev-4.7/arch/arm/boot/dts/aspeed-bmc-opp-palmetto.dts#L149

However, doing this "hogs" the pins for the kernel, so they cannot be changed at runtime by userspace through the sysfs api.

This means you need to modify the skeleton/config/Palmetto.py in order to set up the correct GPIOs (or use whatever mechanism is currently considered best practice in our userspace).

You don't have to be too careful as if you hog a pin in the device tree, the kernel stop you from being able to export it in sysfs, so you will find the bug very quickly.

geissonator, 2016-08-31:
Well, got some more data today.  I grabbed the 4.7.2 kernel, and started adding GPIO's to the device tree.  I created a google spreadsheet with a plan on filling in more info to the table Joel started as I went out here - https://docs.google.com/spreadsheets/d/1M5lDgeUbTBb2mK1tuoUgVxp-uPLzNUy1OiMAAej77o0/edit?usp=sharing

But, as I was incrementally adding things, I noticed that things were working ok with the re-ipl on the Palmetto I was using today.  I went back to the base 4.7.2 (i.e. no changes from me) and things were working fine.  I went back to the 4.7.1 and things continued to work fine.  I verified 4.7.1 was working fine on our barreleye as well (re-ipl's and reboots while host os running).

So looks like the Palmetto I was using yesterday (palm205) is special.  Maybe some floating GPIO's that require the specific config?  Def makes this a lesser priority issue unless people are seeing this on other systems?

amboar, 2016-09-01:
@geissonator my gut feeling is it shouldn't lower the priority until we understand what it is about your palmetto that is special. @legoater, @shenki  and myself were having issues with our Palmettos as well, so it's not limited to palm205.

If you are going to add all of the GPIOs  listed in @shenki's table to either userspace or the devicetree you'll need some help from the pinmux driver. As it stands the pinmux driver in dev-4.7 doesn't implement all of the pin descriptions but I've cooked up a patch: amboar/linux@85cb36fe172b7380a35325ffec887d0d7de97e61. Can you please test it? I've given it a spin under QEMU and on the Palmetto I have with me in the office. Trying to allocate all GPIOs in banks A-H gives:

```
root@palmetto:/sys/class/gpio# for g in `seq 320 383`; do echo $g > export; done
[  157.500000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 0 (D6) for 1e780000.gpio:320
[  157.500000] SCU80[0x00000001]=0x0, got 0x0 from 0x00000000
[  157.560000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 1 (B5) for 1e780000.gpio:321
[  157.560000] SCU80[0x00000002]=0x0, got 0x0 from 0x00000000
[  157.600000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 2 (A4) for 1e780000.gpio:322
[  157.600000] SCU80[0x00000004]=0x0, got 0x0 from 0x00000000
[  157.690000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 3 (E6) for 1e780000.gpio:323
[  157.690000] SCU80[0x00000008]=0x0, got 0x0 from 0x00000000
-sh: echo: write error: Device or resource busy
-sh: echo: writ[  157.800000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 7 (D5) for 1e780000.gpio:327
e error: Device [  157.800000] SCU90[0x00000004]=0x0, got 0x0 from 0x003fa008
or resource busy[  157.800000] SCU80[0x00000080]=0x0, got 0x0 from 0x00000000

-sh: echo: write error: Device or resource busy
[  157.800000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 8 (J21) for 1e780000.gpio:328
[  157.800000] SCU80[0x00000100]=0x0, got 0x0 from 0x00000000
[  157.800000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 9 (J20) for 1e780000.gpio:329
[  157.870000] SCU80[0x00000200]=0x0, got 0x0 from 0x00000000
[  158.030000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 10 (H18) for 1e780000.gpio:330
[  158.030000] SCU80[0x00000400]=0x0, got 0x0 from 0x00000000
[  158.030000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 11 (F18) for 1e780000.gpio:331
[  158.030000] SCU80[0x00000800]=0x0, got 0x0 from 0x00000000
[  158.230000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 12 (E19) for 1e780000.gpio:332
[  158.230000] SCU80[0x00001000]=0x0, got 0x0 from 0x00000000
[  158.230000] SCU70[0x00004000]=0x0, got 0x1 from 0x120ce416
[  158.230000] SCU70[0x00004000]=0x0, got 0x1 from 0x120ce416
[  158.230000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request() failed for pin 12
[  158.230000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-12 (1e780000.gpio:332) status -1
-sh: echo: write error: Operation not permitted
-sh: echo: write error: Device [  158.230000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 15 (E18) for 1e780000.gpio:335
or resource busy[  158.230000] SCU80[0x00008000]=0x0, got 0x0 from 0x00000000

-sh: echo: wri[  158.230000] SCU90[0x80000000]=0x1, got 0x0 from 0x003fa008
te error: Device[  158.230000] SCU80[0x00008000]=0x0, got 0x0 from 0x00000000
 or resource bus[  158.230000] SCU90[0x80000000]=0x1, got 0x1 from 0x803fa008
y
[  158.230000] SCU3c[0x00000008]=0x0, got 0x0 from 0x00000003
[  158.230000] SCU80[0x00008000]=0x0, got 0x0 from 0x00000000
[  158.230000] SCU90[0x80000000]=0x0, got 0x1 from 0x803fa008
[  158.230000] SCU80[0x00008000]=0x0, got 0x0 from 0x00000000
[  158.340000] SCU90[0x80000000]=0x0, got 0x0 from 0x003fa008
[  158.460000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 16 (C4) for 1e780000.gpio:336
[  158.460000] SCU90[0x00000001]=0x0, got 0x0 from 0x003fa008
[  158.460000] SCU90[0x00800000]=0x0, got 0x0 from 0x003fa008
[  158.580000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 17 (B3) for 1e780000.gpio:337
[  158.580000] SCU90[0x00000001]=0x0, got 0x0 from 0x003fa008
[  158.580000] SCU90[0x00800000]=0x0, got 0x0 from 0x003fa008
[  158.580000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 18 (A2) for 1e780000.gpio:338
[  158.580000] SCU90[0x00000001]=0x0, got 0x0 from 0x003fa008
[  158.580000] SCU90[0x01000000]=0x0, got 0x0 from 0x003fa008
[  158.580000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 19 (E5) for 1e780000.gpio:339
[  158.580000] SCU90[0x00000001]=0x0, got 0x0 from 0x003fa008
[  158.580000] SCU90[0x01000000]=0x0, got 0x0 from 0x003fa008
-sh: echo: write error: Device or resource busy
-sh: echo: write error: Device or resource busy
-sh: echo: wri[  158.580000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 25 (D16) for 1e780000.gpio:345
te error: Device[  158.580000] SCU90[0x00000002]=0x0, got 0x0 from 0x003fa008
 or resource bus[  158.580000] SCU8c[0x00000100]=0x0, got 0x0 from 0xc1c000ff
y
-sh: echo: wr[  158.580000] SCU70[0x00200000]=0x0, got 0x0 from 0x120ce416
ite error: Device or resource busy
-sh: echo: write error: Device or resource busy
[  158.990000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 26 (B17) for 1e780000.gpio:346
[  158.990000] SCU90[0x00000002]=0x0, got 0x0 from 0x003fa008
[... cut ...]
3. Trying to export a pin that has fixed functionality via the hardware strapping register(s) (`-sh: echo: write error: Operation not permitted`)

If you choose to add GPIO hogs to the devicetree (rather than in the skeleton userspace configuration files) the nodes should be named something relevant to the pin (i.e. not e.g. `pin_func_mode3`).

I will be sending the pinmux patch I linked above to @shenki in the near future, so your input - testing and any other feedback - would be appreciated (e.g. we can add your Tested-by: tag to the patch).

textaddict, 2016-09-01:
I probably have one of the "special" palmettos.  I have been powering it on reliably with OpenBmc 1.0-stable build for awhile, recently migrated to a master build after getting it to power on with a netboot.  

This system will not power on after an AC power cycle - it just recycles system power after about 30 seconds when a watchdog times out (guessing, it's around a 30-35 second timeout).  But if I netboot it back into that earlier 1.0-stable build, I can then power on reliably.  If I do that once then reboot the BMC back into the master image (from flash) it can power on reliably over and over until I AC power cycle it again.

So that suggests it is something that master isn't initalizing (at least on this machine), prevents the system from starting up, but once it's initialized by the older code it always works.  Probably some gpios that aren't initialized (floating) that the older code initializes but the new version doesn't touch.  I will try to gather some more data today about the passing and failing cases and add it here if I can.  I have also subscribed to this defect so I can test any fixes/changes that come out of this.

madscientist159, 2016-09-01:
@textaddict What I think would help most is if you could dump the GPIO state (using direct MMIO access to the GPIO control registers) for a working boot and a failed boot.  That would allow isolation fairly quickly of what changed.

textaddict, 2016-09-01:
Added a comment to 527 showing GPIO differences (since this isn't directly related to the pgood issue I wasn't sure they belonged here).  The GPIO data bits look (mostly) the same, it's the direction registers (offsets 0x04 and 0x24) that look like they are failing to be initialized in master.

amboar, 2016-09-02:
@williamspatrick @geissonator @textaddict can you fetch down and test `amboar/linux obmc-master-palmetto-fixes`? The patches are in a PoC state but if they come up okay I'll rework them into something we can apply to dev-4.7. There are three patches, addressing:

* Defining missing pins in the g4 pinmux driver for GPIO banks A-H (amboar/linux@85cb36fe172b7380a35325ffec887d0d7de97e61)
* A fix for the Aspeed pinmux engine to deal with corner cases in pin function requests (amboar/linux@40290d6fd1baa931eba3e860ce52479c0a5f1bf9)
* Add Palmetto GPIO hog descriptions to the devicetree (amboar/linux@861b7d4f94c834d91ce74e983f83dca8c343d225)

The last patch was generated from the reported GPIO register values in the https://github.com/openbmc/openbmc/issues/527#issuecomment-244239595 by @textaddict using this script: https://gist.github.com/amboar/8a01c61a9dd3194b02d2f08ffca70d08

geissonator, 2016-09-02:
Ahhh, thanks Andrew, I started to work on a part of 1 of these patches the other day and got in a bit over my head :)  I've got a build now and flashing it on my Palmetto.  If anyone else needs the image, just ping me.

legoater, 2016-09-02:
I gave the patchset a try on a palmetto and there are some progress at reboot (of the BMC). 
The kernel is still alive but in xmon. The backtrace reports eeh. I guess we still reset too much 
gpios. 
    
```
e:mon> 
[c0000003f9993a20] d000000000c6a6b8 bnx2x_prev_unload+0x132c/0x166c [bnx2x]
[c0000003f9993ae0] c000000000021b20 eeh_report_reset+0xb0/0x10c
[c0000003f9993b20] c000000000020d1c eeh_pe_dev_traverse+0x60/0xd8
[c0000003f9993bb0] c0000000000222e0 eeh_handle_normal_event+0x2f8/0x400
[c0000003f9993c30] c0000000000225ec eeh_handle_event+0x30/0x270
[c0000003f9993cd0] c00000000002293c eeh_event_handler+0x110/0x150
[c0000003f9993d80] c00000000007161c kthread+0xfc/0x108
[c0000003f9993e30] c0000000000094b4 ret_from_kernel_thread+0x5c/0xa8
```

legoater, 2016-09-02:
FYI,

here are the kernel log : https://gist.github.com/legoater/36fddd30c2d8f8484bfabbd8dab03854
and opal log : https://gist.github.com/legoater/e87989696c6c570fe9ec3b2295b880e9

exiting from xmon works fine but network is in a poor state. I had reboot finally

geissonator, 2016-09-02:
My testing was mixed.  The "good" Palmetto (which only occasionally fails) worked like a champ with this change.  By "bad" Palmetto showed the same behavior as before.  The initial boot after flash worked fine but then subsequent boots failed with the watchdog hitting and the BMC powering off/on again.

I used the pdbg tool to dump some SBE registers (to try and get some hint as to why we hang in the IPL).  The conclusion back from the SBE team was that it was hitting a unrecoverable error when trying to fetch it's software image over I2C from an seeprom.  I'll be fascinated to know how this gpio logic is causing this type of fail :)  

There's also the possibility that the bad Palmetto (palm205) is just bad all around, I haven't had a chance to go back and verify the stable branch works on it recently.  I do remember one of Andrew's previous patches working well on it though.

geissonator, 2016-09-20:
I used the image from https://openpower.xyz/job/openbmc-build-gerrit-trigger-multi/distro=ubuntu,target=palmetto/665/

On palm205, I saw the same behavior as before.  The system does not boot.
On palm309, after I reflashed PNOR, I was able to boot the system multiple times to petitboot

palm309 and palm205 both show this to the console during power on:

> root@palmetto:~# obmcutil poweron 
>  = None
> root@palmetto:~# ftgmac100 1e660000.ethernet eth0: NCSI interface down
> aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin AB1 already requested by 1e6e2000.syscon:pinctrl@1e6e2000; cannot claim for 1e780000.gpio:427
> aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-107 (1e780000.gpio:427) status -22
> ftgmac100 1e660000.ethernet eth0: NCSI interface up
> aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin W5 already requested by 1e6e2000.syscon:pinctrl@1e6e2000; cannot claim for 1e780000.gpio:428
> aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-108 (1e780000.gpio:428) status -22
>

textaddict, 2016-09-20:
I tested this image on my Tyan production level (version 2) Palmetto.  It does IPL on this system too BUT DOES NOT TURN ON THE FANS!!!!!

WARNING - Possible Safety and machine damage risk if your palmetto is running with the fan connector plugged in (not sure how common this is - the system shipped with it unplugged and I only attached it after a AMI BMC update that included fan control).  Be careful when testing this and shut your machine down quickly if you don't have separate cooling available.

madscientist159, 2016-09-20:
@textaddict Sounds like our Firestone.  You will need to add in the hardware monitoring information to the Palmetto.py configuration file, similar to these patches for Firestone:
https://gerrit.openbmc-project.xyz/#/c/554/
https://gerrit.openbmc-project.xyz/#/c/555/

I don't know how the fans are controlled on Palmetto but you will need to update the I2C address to match the location of your fan controller.

Automatic fan control won't work until https://gerrit.openbmc-project.xyz/#/c/120/ is fixed and merged, but at least this should prevent hardware damage!

textaddict, 2016-09-20:
The Palmetto fan controller was designed so that if the connector is unplugged it just runs full out.  So you can get by with this level of code (or similar) if the fans are unconnected - they just default to on max RPM.

legoater, 2016-09-20:
Is there a way to identify a Palmetto model ? Mine runs fine with Andrew's patches but we got one from Tyan and I would not want to fry it.

textaddict, 2016-09-20:
I'm not sure how to identify the different versions but I'm pretty sure Tyan only shipped one version for production (version 2).  Best way to be safe is to pull the fan cable from the planar (if it's plugged in, it's possible if nobody did that after a code update that the cable is still in a retaining clip on the side).  I only plugged mine in after we got some AMI BMC code that included active fan control.

geissonator, 2016-09-20:
palm205 also failed to boot with stable so seems like it's a goner.  On the bright side, I think that means this patch is good to go (other then maybe the fan control comments from above).

textaddict, 2016-09-21:
The palmetto system does IPL with fans running if you pull the fan control cable.  It's easy to identify on the main planar along the left side near the chassis with a label on it.

amboar, 2016-09-21:
@textaddict If the fans are controlled by PWM from the SoC and the control cable is plugged in then it's likely they're not running as we have no PWM driver, however the pins are muxed for PWM. I don't believe this is related to the GPIO problem, but it's another thing that's suboptimal.

@geissonator can you please comment on or score the patch in gerrit?

textaddict, 2016-09-22:
@amboar Ok but just fyi an earlier build of master was (somehow) turning them on.  v1.0-174 did manage to turn on the fans with the control cable plugged in (or not, if not they came on by default).  

That level didn't include these GPIO fixes so couldn't IPL until I hacked them with some Devmem commands.  

The thing that is **a little bit Dangerous** about this version is that if the machine is being IPLd in a remote room _**you could start the system up without fans running and not know until it was too late.**_  Although the OCC would probably try to power limit to avoid thermal runaway those indium pads tend to liquefy and run down into the sockets before that happens (at least sometimes).  Gets expensive.

amboar, 2016-09-22:
@textaddict I share your concerns. However I don't think this is the place we should be exploring the fan problems, this issue already has a lack of focus.

@williamspatrick @shenki I think we can close this issue now.

@textaddict I will open a separate issue for the fans.

shenki, 2016-09-27:
I flashed the following to palm5, a Palmetto in the lab in Canberra:

* palmetto.pnor
  * master-next 922a1c1441fa24c4d8323255a887b7b08eb1362b
  * https://openpower.xyz/job/openpower-op-build/181/

* flash-palmetto-20160923222626
  * `#265-a81f31c4bb55ed`
  * https://openpower.xyz/job/openbmc-build/265/

```
root@palmetto:~# obmcutil power
pgood = 1
state = 1
pgood_timeout = 10
root@palmetto:~# obmcutil power on
org.freedesktop.DBus.Error.UnknownMethod: No such method 'on'
ERROR: Invalid method: on
root@palmetto:~# obmcutil poweron 
 = None
root@palmetto:~# aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin AB1 already requested by 1e6e2000.syscon:pinctrl@1e6e2000; cannot claim for 1e780000.gpio:427
aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-107 (1e780000.gpio:427) status -22
aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin W5 already requested by 1e6e2000.syscon:pinctrl@1e6e2000; cannot claim for 1e780000.gpio:428
aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-108 (1e780000.gpio:428) status -22
```

On the host:
```
 20.14428|ISTEP  9. 2
 57.91705|IPMI: Initiate power cycle
 57.92607|sbe|System Rebooting To Perform SBE Update

 57.96355|================================================
 57.96356|Error reported by ipmi (0x2500)
 57.97291|  Set sensor reading command failed.
 57.98239|  ModuleId   0x03 IPMI::MOD_IPMISENSOR
 57.98240|  ReasonCode 0x2508 IPMI::RC_SET_SENSOR_FAILURE
 57.98240|  UserData1  BMC IPMI Completion code. : 0x00000000000000ce
 57.98241|  UserData2  bytes [0-3]sensor number bytes [4-7]HUID of target. : 0x0000000900000000
 57.98241|User Data Section 0, type UD
 57.98241|  Subsection type 0x06
 57.98241|  ComponentId errl (0x0100)
 57.98241|  CALLOUT
 57.98242|  PROCEDURE ERROR
 57.98242|  Procedure: 85
 57.98242|User Data Section 1, type UD
 57.98242|  Subsection type 0x15
 57.98243|  ComponentId hb-trace (0x3100)
 57.98243|User Data Section 2, type UD
 57.98243|  Subsection type 0x15
 57.98243|  ComponentId hb-trace (0x3100)
 57.98244|User Data Section 3, type UD
 57.98244|  Subsection type 0x03
 57.98244|  ComponentId errl (0x0100)
 57.98244|User Data Section 4, type UD
 57.98245|  Subsection type 0x01
 57.98245|  ComponentId errl (0x0100)
 57.98245|  STRING
 57.98245|  Hostboot Build ID: 
 57.98246|================================================
 60.31668|Stopping istep dispatcher
 65.41868|IPMI: shutdown complete

  1.47459|ERRL|Dumping errors reported prior to registration
  1.51149|================================================
  1.51149|Error reported by ipmi (0x2500)
  1.52994|  Set sensor reading command failed.
  1.53915|  ModuleId   0x03 IPMI::MOD_IPMISENSOR
  1.53915|  ReasonCode 0x2508 IPMI::RC_SET_SENSOR_FAILURE
  1.53915|  UserData1  BMC IPMI Completion code. : 0x00000000000000ce
  1.53916|  UserData2  bytes [0-3]sensor number bytes [4-7]HUID of target. : 0x0000000900000000
  1.53916|User Data Section 0, type UD
  1.53916|  Subsection type 0x06
  1.53917|  ComponentId errl (0x0100)
  1.53917|  CALLOUT
  1.53917|  PROCEDURE ERROR
  1.53917|  Procedure: 85
  1.53918|User Data Section 1, type UD
  1.53918|  Subsection type 0x03
  1.53918|  ComponentId errl (0x0100)
  1.53918|User Data Section 2, type UD
  1.53919|  Subsection type 0x01
  1.53919|  ComponentId errl (0x0100)
  1.53919|  STRING
  1.53919|  Hostboot Build ID: 
  1.53920|User Data Section 3, type UD
  1.53920|  Subsection type 0x15
  1.53920|  ComponentId hb-trace (0x3100)
  1.53920|User Data Section 4, type UD
  1.53921|  Subsection type 0x15
  1.53921|  ComponentId hb-trace (0x3100)
  1.53921|================================================
  3.52042|ISTEP  6. 3
  3.97157|ISTEP  6. 4
```

From there it continues to boot to Petitboot.

amboar, 2016-09-27:
palm6 host here in Adelaide also boots, reboots and survives a BMC reboot with the images @shenki points to above.

shenki, 2016-09-27:
Testing using images at https://ozlabs.au.ibm.com/~joel/openbmc-pgood-1/ (IBM internal URL)

| Machine | PGOOD | Host boot to petitboot | Host reboot | BMC reboot and host stays up|
| ----          | ----         | ----         | ----         | ----         |
| palm5     | Y           | Y            | Y           | Y            |
| palm6     | Y           | Y            | Y           | Y            |
| palm205 |              |               |               |               |
|                |              |               |               |               |

legoater, 2016-09-27:
mine is palm328. all is fine on my side now.

shenki, 2016-09-28:
@legoater thanks. I updated the table

[4 more replies]

closed 2017-08-25 by rfrandse