openbmc/openbmc #514: obmc-rest crashes on web browser access

hweaving, 2016-08-19:
With a master build pulled on 08/16/2016, obmc-rest running inside qemu can be accessed by using the example curl commands at [https://github.com/openbmc/docs/blob/master/rest-api.md].

However, pointing a web browser (iceweasel) to https://192.168.7.2 (the BMC's address) crashes obmc-rest.  In fact, obmc-rest crashes before even attempting to add a certificate exception in the web browser.

This crash is due to a Python exception.  From journalctl on the BMC:

Aug 19 04:18:33 qemuarm obmc-rest[307]: Exception in thread Thread-2:
Aug 19 04:18:33 qemuarm obmc-rest[307]: Traceback (most recent call last):
Aug 19 04:18:33 qemuarm obmc-rest[307]:   File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
Aug 19 04:18:33 qemuarm obmc-rest[307]:     self.run()
Aug 19 04:18:33 qemuarm obmc-rest[307]:   File "/usr/lib/python2.7/site-packages/rocket/worker.py", line 179, in run
Aug 19 04:18:33 qemuarm obmc-rest[307]:     handled = self._handleError(*exc)
Aug 19 04:18:33 qemuarm obmc-rest[307]:   File "/usr/lib/python2.7/site-packages/rocket/worker.py", line 98, in _handleError
Aug 19 04:18:33 qemuarm obmc-rest[307]:     if 'timed out' in val.args[0]:
Aug 19 04:18:33 qemuarm obmc-rest[307]: TypeError: argument of type 'int' is not iterable

hweaving, 2016-08-19:
Interestingly, the Python error path is the same as for https://github.com/openbmc/openbmc/issues/306.

I made sure to wait 3+ minutes after the BMC acquired an IP address before initiating the crash.

hweaving, 2016-08-19:
This bug seems to be due to an exception with a format the Rocket webserver does not expect.  I added traces to the start of _handleError in worker.py:

Aug 19 05:57:39 qemuarm obmc-rest[276]: TRACE typ: <class 'rocket.worker.SocketClosed'>
Aug 19 05:57:39 qemuarm obmc-rest[276]: TRACE val: Client closed socket.
Aug 19 05:57:39 qemuarm obmc-rest[276]: TRACE val.args: ('Client closed socket.',)
Aug 19 06:00:08 qemuarm systemd-timesyncd[151]: Network configuration changed, trying to establish connection.
Aug 19 06:01:05 qemuarm obmc-rest[276]: TRACE typ: <class 'rocket.worker.SocketClosed'>
Aug 19 06:01:05 qemuarm obmc-rest[276]: TRACE val: Client closed socket.
Aug 19 06:01:05 qemuarm obmc-rest[276]: TRACE val.args: ('Client closed socket.',)
Aug 19 06:01:07 qemuarm obmc-rest[276]: TRACE typ: <class 'ssl.SSLError'>
Aug 19 06:01:07 qemuarm obmc-rest[276]: TRACE val: [SSL: TLSV1_ALERT_UNKNOWN_CA] unknown error (_ssl.c:1751)
Aug 19 06:01:07 qemuarm obmc-rest[276]: TRACE val.args: (1, u'[SSL: TLSV1_ALERT_UNKNOWN_CA] unknown error (_ssl.c:1751)')

Note that val.args[0] is an integer rather than a string in this case, causing the crash.

hweaving, 2016-08-19:
After some digging, I found the following:

1.  Rocket does not seem to have been updated since 2011.

2.  In Rocket's worker.py file, it expects an SSLError's "args" member to be a tuple or string [https://github.com/errbotio/rocket/blob/master/rocket/worker.py#L98](https://github.com/errbotio/rocket/blob/master/rocket/worker.py#L98) (not our precise version of Rocket, but the line of code should be the same)

3.  Back in 2010, ([https://github.com/python/cpython/blob/3ec894f0248d520a477f4e989c22e710caf892eb/Modules/_ssl.c#L254](https://github.com/python/cpython/blob/3ec894f0248d520a477f4e989c22e710caf892eb/Modules/_ssl.c#L254)), Python SSL code passed a tuple to create an SSLError object.  This presumably resulted in args[0] being an iterable tuple.  See [https://docs.python.org/2/c-api/arg.html](https://docs.python.org/2/c-api/arg.html) for reference.

4.  Since 2010, Python has changed ([https://github.com/python/cpython/blob/2.7/Modules/_ssl.c#L349](https://github.com/python/cpython/blob/2.7/Modules/_ssl.c#L349)) to use a different format string.  This presumably results in arg[0] being an integer error code.  Note that [https://docs.python.org/3.0/library/ssl.html#ssl.SSLSocket.do_handshake](https://docs.python.org/3.0/library/ssl.html#ssl.SSLSocket.do_handshake) describes args[0] as an error code like SSL_ERROR_WANT_READ, for example.  This was not documented before Python 3.0 but still seems to be valid for Python 2.7 in some cases (see item 6 below).

5.  As a result of the above, Rocket encounters an integer error code in args[0] and executes "if 'timed out' in val.args[0]:", crashing obmc-rest because integers are not iterable.

6.  However, the above may be a red herring despite the "(is)" to "iN" change.  Merely checking val.args[1] is not valid because some cases DO generate an SSLError where args is a tuple containing a single string (repeatedly reloading the webpage results in "The read operation timed out", for example).  Rocket would then crash on those cases.

7.  My conclusion:  This is a Rocket webserver bug which may crash the REST interface on certain occurrences of SSLError.  Possible solutions are patching this to check for tuple lengths of 1 or 2 (are other lengths possible?) or replacing Rocket ASAP if it is no longer maintained.

causten, 2016-08-19:
Nice research, 
for our 1.0 Stable branch I recommend "patching this to check for tuple lengths of 1 or 2"

for Master is there a supported alternative that fits the embedded space?  Anything already recommended with Yocto?

Chris Austen
POWER Systems Enablement Manager 
(512) 286-5184 (T/L: 363-5184)

-----

   Rocket does not seem to have been updated since 2011. In Rocket's worker.py file, it expects an SSLError's "args" member to be a tuple or string https://github.com/errbotio/rocket/blob/master/rocket/worker.py#L98 (not our precise version of Rocket, but the line of code should be the same) Back in 2010, (https://github.com/python/cpython/blob/3ec894f0248d520a477f4e989c22e710caf892eb/Modules/_ssl.c#L254), Python SSL code passed a tuple to create an SSLError object.  This presumably resulted in args[0] being an iterable tuple.  See https://docs.python.org/2/c-api/arg.html for reference. Since 2010, Python has changed (https://github.com/python/cpython/blob/2.7/Modules/_ssl.c#L349) to use a different format string.  This presumably results in arg[0] being an integer error code.  Note that https://docs.python.org/3.0/library/ssl.html#ssl.SSLSocket.do_handshake describes args[0] as an error code like SSL_ERROR_WANT_READ, for example.  This was not documented before Python 3.0 but still seems to be valid for Python 2.7 in some cases (see item 6 below). As a result of the above, Rocket encounters an integer error code in args[0] and executes "if 'timed out' in val.args[0]:", crashing obmc-rest because integers are not iterable. However, the above may be a red herring despite the "(is)" to "iN" change.  Merely checking val.args[1] is not valid because some cases DO generate an SSLError where args is a tuple containing a single string (repeatedly reloading the webpage results in "The read operation timed out", for example).  Rocket would then crash on those cases. My conclusion:  This is a Rocket webserver bug which may crash the REST interface on certain occurrences of SSLError.  Possible solutions are patching this to check for tuple lengths of 1 or 2 (are other lengths possible?) or replacing Rocket ASAP if it is no longer maintained.   —

bradbishop, 2016-08-19:
@hweaving thanks for digging in.

I tried python-gevent which worked ok but haven't done any extensive testing.  I like that its single threaded.  It also only took about 5 minutes to deploy.

I'm open to alternative suggestions assuming they don't use too much memory, and have a small flash footprint.  Good performance would be nice too.

hweaving, 2016-08-22:
I'm looking into nginx and gunicorn at present.

hweaving, 2016-08-29:
I confirmed nginx + gunicorn could work for https proxying of the obmc-rest Bottle app.  Logging in, doing GET, and doing POST worked.

Unfortunately, gunicorn seems to use a huge amount of flash space, plus a huge amount of RAM even with 2 workers, compared to Rocket.

I am not investigating further at the moment, but I recommend any interested parties investigate nginx + uwsgi (requires meta-openstack) and/or nginx + gevent instead.  Running nginx itself seems to be lightweight.

bradbishop, 2017-08-25:
I haven't seen this since we switched to gevent.

closed 2017-08-25 by bradbishop