openbmc/openbmc #516: SDK fails to build on Ubuntu 16.04 ppc64le

causten, 2016-08-19:
ERROR: QA Issue: nativesdk-libgcc: Files/directories were installed but not shipped in any package:
  /opt/openbmc-phosphor/2.0.1/sysroots/ppc64le-pokysdk-linux/lib
  /opt/openbmc-phosphor/2.0.1/sysroots/ppc64le-pokysdk-linux/usr/lib64
  /opt/openbmc-phosphor/2.0.1/sysroots/ppc64le-pokysdk-linux/usr/lib64/libgcc_s.so
  /opt/openbmc-phosphor/2.0.1/sysroots/ppc64le-pokysdk-linux/usr/lib64/libgcc_s.so.1
  /opt/openbmc-phosphor/2.0.1/sysroots/ppc64le-pokysdk-linux/usr/lib64/.debug
  /opt/openbmc-phosphor/2.0.1/sysroots/ppc64le-pokysdk-linux/usr/lib64/.debug/libgcc_s.so.1
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
nativesdk-libgcc: 6 installed and not shipped files. [installed-vs-shipped]
ERROR: Function failed: do_configure (log file is located at /home/causten/workspaces/obmc_stable/openbmc/build/tmp/work/ppc64le-nativesdk-pokysdk-linux/nativesdk-gcc-runtime/4.9.3-r0/temp/log.do_configure.16311)
ERROR: Logfile of failure stored in: /home/causten/workspaces/obmc_stable/openbmc/build/tmp/work/ppc64le-nativesdk-pokysdk-linux/nativesdk-gcc-runtime/4.9.3-r0/temp/log.do_configure.16311
Log data follows:
| DEBUG: Executing python function sysroot_cleansstate
| DEBUG: Python function sysroot_cleansstate finished
| DEBUG: SITE files ['endian-little', 'bit-64', 'powerpc-common', 'common-linux', 'common-glibc', 'ppc64le-linux', 'common']
| DEBUG: Executing shell function autotools_preconfigure
| DEBUG: Shell function autotools_preconfigure finished
| DEBUG: Executing python function autotools_copy_aclocals
| DEBUG: SITE files ['endian-little', 'bit-64', 'powerpc-common', 'common-linux', 'common-glibc', 'ppc64le-linux', 'common']
| DEBUG: Python function autotools_copy_aclocals finished
| DEBUG: Executing python function gcc_multilib_setup
| DEBUG: Python function gcc_multilib_setup finished
| DEBUG: Executing shell function do_configure
| 0 blocks
| Configuring libgcc
| configure: WARNING: unrecognized options: --disable-silent-rules, --disable-dependency-tracking, --with-libtool-sysroot, --with-gnu-ld, --enable-languages, --enable-threads, --disable-multilib, --enable-c99, --enable-long-long, --enable-symvers, --enable-libstdcxx-pch, --without-local-prefix, --enable-target-optspace, --enable-lto, --enable-libssp, --disable-bootstrap, --disable-libmudflap, --with-system-zlib, --with-linker-hash-style, --enable-linker-build-id, --with-ppl, --with-cloog, --enable-checking, --enable-cheaders, --with-gxx-include-dir, --with-sysroot, --with-build-sysroot, --disable-nls, --enable-__cxa_atexit
| checking build system type... powerpc64le-unknown-linux-gnu
| checking host system type... powerpc64le-pokysdk-linux-gnu
| checking for --enable-version-specific-runtime-libs... no
| checking for a BSD-compatible install... /usr/bin/install -c
| checking for gawk... gawk
| checking for ppc64le-pokysdk-linux-ar... ppc64le-pokysdk-linux-ar
| checking for ppc64le-pokysdk-linux-lipo... no
| checking for lipo... no
| checking for ppc64le-pokysdk-linux-nm... ppc64le-pokysdk-linux-nm
| checking for ppc64le-pokysdk-linux-ranlib... ppc64le-pokysdk-linux-ranlib
| checking for ppc64le-pokysdk-linux-strip... ppc64le-pokysdk-linux-strip
| checking whether ln -s works... yes
| checking for ppc64le-pokysdk-linux-gcc... ppc64le-pokysdk-linux-gcc  --sysroot=/home/causten/workspaces/obmc_stable/openbmc/build/tmp/sysroots/ppc64le-nativesdk-pokysdk-linux
| checking for suffix of object files... o
| checking whether we are using the GNU C compiler... yes
| checking whether ppc64le-pokysdk-linux-gcc  --sysroot=/home/causten/workspaces/obmc_stable/openbmc/build/tmp/sysroots/ppc64le-nativesdk-pokysdk-linux accepts -g... yes
| checking for ppc64le-pokysdk-linux-gcc  --sysroot=/home/causten/workspaces/obmc_stable/openbmc/build/tmp/sysroots/ppc64le-nativesdk-pokysdk-linux option to accept ISO C89... none needed
| checking how to run the C preprocessor... ppc64le-pokysdk-linux-gcc -E --sysroot=/home/causten/workspaces/obmc_stable/openbmc/build/tmp/sysroots/ppc64le-nativesdk-pokysdk-linux
| checking size of double... 8
| checking size of long double... 8
| checking whether decimal floating point is supported... yes
| checking whether fixed-point is supported... no
| checking whether to use setjmp/longjmp exceptions... no
| checking if the linker (ppc64le-pokysdk-linux-ld --sysroot=/home/causten/workspaces/obmc_stable/openbmc/build/tmp/sysroots/ppc64le-nativesdk-pokysdk-linux ) is GNU ld... yes
| checking for thread model used by GCC... posix
| checking whether assembler supports CFI directives... yes
| checking for __attribute__((visibility("hidden")))... yes
| checking whether the target assembler supports thread-local storage... yes
| checking whether the thread-local storage support is from emutls... no
| checking for init priority support... yes
| configure: creating ./config.status
| config.status: creating Makefile
| config.status: creating auto-target.h
| config.status: linking /home/causten/workspaces/obmc_stable/openbmc/build/tmp/work-shared/gcc-4.9.3-r0/gcc-4.9.3/libgcc/enable-execute-stack-empty.c to enable-execute-stack.c
| config.status: linking /home/causten/workspaces/obmc_stable/openbmc/build/tmp/work-shared/gcc-4.9.3-r0/gcc-4.9.3/libgcc/unwind-generic.h to unwind.h
| config.status: linking /home/causten/workspaces/obmc_stable/openbmc/build/tmp/work-shared/gcc-4.9.3-r0/gcc-4.9.3/libgcc/config/rs6000/linux-unwind.h to md-unwind-support.h
| config.status: linking /home/causten/workspaces/obmc_stable/openbmc/build/tmp/work-shared/gcc-4.9.3-r0/gcc-4.9.3/libgcc/config/rs6000/sfp-machine.h to sfp-machine.h
| config.status: linking /home/causten/workspaces/obmc_stable/openbmc/build/tmp/work-shared/gcc-4.9.3-r0/gcc-4.9.3/libgcc/gthr-posix.h to gthr-default.h
| config.status: executing default commands
| configure: WARNING: unrecognized options: --disable-silent-rules, --disable-dependency-tracking, --with-libtool-sysroot, --with-gnu-ld, --enable-languages, --enable-threads, --disable-multilib, --enable-c99, --enable-long-long, --enable-symvers, --enable-libstdcxx-pch, --without-local-prefix, --enable-target-optspace, --enable-lto, --enable-libssp, --disable-bootstrap, --disable-libmudflap, --with-system-zlib, --with-linker-hash-style, --enable-linker-build-id, --with-ppl, --with-cloog, --enable-checking, --enable-cheaders, --with-gxx-include-dir, --with-sysroot, --with-build-sysroot, --disable-nls, --enable-__cxa_atexit
| Configuring libssp
| configure: WARNING: unrecognized options: --disable-silent-rules, --with-libtool-sysroot, --enable-languages, --enable-threads, --enable-c99, --enable-long-long, --enable-libstdcxx-pch, --without-local-prefix, --enable-target-optspace, --enable-lto, --enable-libssp, --disable-bootstrap, --disable-libmudflap, --with-system-zlib, --with-linker-hash-style, --enable-linker-build-id, --with-ppl, --with-cloog, --enable-checking, --enable-cheaders, --with-gxx-include-dir, --with-sysroot, --with-build-sysroot, --disable-nls, --enable-__cxa_atexit
| checking build system type... powerpc64le-unknown-linux-gnu
| checking host system type... powerpc64le-pokysdk-linux-gnu
[... 65 lines cut ...]
| checking whether the shell understands some XSI constructs... yes
| checking whether the shell understands "+="... yes
| checking for ppc64le-pokysdk-linux-ld --sysroot=/home/causten/workspaces/obmc_stable/openbmc/build/tmp/sysroots/ppc64le-nativesdk-pokysdk-linux  option to reload object files... -r
| checking for ppc64le-pokysdk-linux-objdump... ppc64le-pokysdk-linux-objdump
| checking how to recognize dependent libraries... pass_all
| checking for ppc64le-pokysdk-linux-ar... ppc64le-pokysdk-linux-ar
| checking for ppc64le-pokysdk-linux-strip... (cached) ppc64le-pokysdk-linux-strip
| checking for ppc64le-pokysdk-linux-ranlib... ppc64le-pokysdk-linux-ranlib
| checking command to parse ppc64le-pokysdk-linux-nm output from ppc64le-pokysdk-linux-gcc  --sysroot=/home/causten/workspaces/obmc_stable/openbmc/build/tmp/sysroots/ppc64le-nativesdk-pokysdk-linux object... failed
| checking for dlfcn.h... yes
| checking for objdir... .libs
| checking if ppc64le-pokysdk-linux-gcc  --sysroot=/home/causten/workspaces/obmc_stable/openbmc/build/tmp/sysroots/ppc64le-nativesdk-pokysdk-linux supports -fno-rtti -fno-exceptions... no
[... cut ...] )
ERROR: Task 794 (virtual:nativesdk:/home/causten/workspaces/obmc_stable/openbmc/meta/recipes-devtools/gcc/gcc-runtime_4.9.bb, do_configure) failed with exit code '1'

I can build master and stable on a Ubuntu 16.04 ppc64le system.  When I try to create teh SDK it fails with the above error message.  I used the "bitbake -c populate_sdk  obmc-phosphor-image" to start the build.  I was based off of the v1.0.4 stable tag (63288c0590fc9021caaed0bbe2d12ebf86d8f87e).  Results also happen with master.

geissonator, 2016-08-19:
This popped up in a search.  Not sure why the new Ubuntu would cause it though - https://lists.yoctoproject.org/pipermail/yocto/2016-January/028127.html

geissonator, 2016-09-29:
Also, it's not just ubuntu:16.04, I get on ubuntu:14.04 as well.

amboar, 2016-09-30:
@causten @geissonator we should fix the issue summary if the problem isn't specific to Ubuntu 16.04

gtmills, 2018-05-23:
@geissonator @causten still valid? Or can this be closed?

anoo1, 2018-09-05:
I just tried to build an SDK in a POWER system and it builds an x86 one. @mine260309 has separately reported this a while back. I guess we can close this since it's building, it's just not building the right one, or I may not know how to set the env to build on ppc64le.

```
tmp/deploy/sdk/oecore-x86_64-armv6-toolchain-nodistro.0.sh

Error: Incompatible SDK installer! Your host is ppc64le and this SDK was built for x86_64 hosts.
```

closed 2018-09-05 by anoo1

bjwyman, 2021-04-22:
Somewhere along the line I saw this stick out:
https://github.com/openembedded/openembedded-core/commit/c915619b78e0c3e0d73d9de0b85c658c563ed5ba
I thought perhaps that fixed the issue with the "tried to build an SDK in a POWER system and it builds an x86 one." so I gave `-c populate_sdk` a whirl on a POWER Ubuntu 20.04.1 LTS system. It failed with some similar messages, but I think the error list might be shorter now.

bjwyman, 2021-05-05:
```
...
ERROR: QA Issue: nativesdk-libgcc: Files/directories were installed but not shipped in any package:
  /usr/local/oecore-ppc64le/sysroots/ppc64le-oesdk-linux/lib
  /usr/local/oecore-ppc64le/sysroots/ppc64le-oesdk-linux/usr/lib64/libgcc_s.so.1
  /usr/local/oecore-ppc64le/sysroots/ppc64le-oesdk-linux/usr/lib64/libgcc_s.so
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
nativesdk-libgcc: 3 installed and not shipped files. [installed-vs-shipped]
...
ERROR: QA Issue: nativesdk-glibc-locale: Files/directories were installed but not shipped in any package:
  /usr/local/oecore-ppc64le/sysroots/ppc64le-oesdk-linux/usr/lib/locale
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
nativesdk-glibc-locale: 1 installed and not shipped files. [installed-vs-shipped]
...
```
```
$ cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.1 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.1 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```