openbmc/openbmc #517: SLP support

williamspatrick, 2016-08-22:


williamspatrick, 2016-08-22:
IBM's FSP will return an SLP response as below.  The manufacturing tooling uses SLP to identify if the device it is communicating with is an FSP or BMC.  We need to add an SLP daemon and insert enough of an SLP response that can be used to differentiate between a system that is running OpenBMC and one that is not.

```
Command:
slptool -u <FSP IP/Host> findattrs service:management-hardware.IBM:cec-service-processor

Response:
(type=cec-service-processor),(serial-number=1069A8T),(machinetype-model=8286-42A),(fru-serial-number=YL30UF34800B),(name=tul127fp),(frame-number=0),(cage-number=0),(ip-address=x.x.x.x,169.254.3.147),(web-url=https://x.x.x.x:473),(slot=1),(bpc-machinetype-model=0),(bpc-serial-number=0),(Image=fips860/b0726a_1632.860),(cimom-namespace=root/ibmsd),(secure-cimom-port=5989),(firmware-image-info=TV860_027),(mtms=8286-42A1069A8T),(MicroActiveCodeImage=TV860_027),(MicroInactiveCodeImage=TV860_027),(IPv6-ipaddress=xxxx:0000:0000:0000:xxxx:xxxx:xxxx:xxxx,),(IPv4-enabled=true),(IPv6-enabled=true),(level=4)
```

causten, 2016-08-22:
So an openbmc system will modify the 'type' parameter? I.e. openbmc-service-processor
   
  Sent from my iPhone using IBM Verse

williamspatrick, 2016-08-23:
I haven't decided on the entire SLP response yet. Probably whoever owns this issue will come up with a proposal and it will be sent to various stakeholders for confirmation.

williamspatrick, 2016-09-07:
A portion of this will be implemented with epic #547.

geissonator, 2017-03-29:
Verified with @ratagupt that all required function for mfg delivered.

closed 2017-03-29 by geissonator