openbmc/openbmc #518: BMC is not ready in obmc 1.0.5

johnhcwang, 2016-08-23:
Unexpected translate to HOST_POWERED_ON state and BMC never translate to ready state when running the obmc 1.0.5 on Barreleye.
````
root@barreleye:~# journalctl --no-pager|grep "System State"
Aug 23 01:37:32 barreleye system_manager.py[975]: Running System State: BASE_APPS
Aug 23 01:37:41 barreleye system_manager.py[975]: SystemManager Goto System State: BMC_STARTING
Aug 23 01:37:41 barreleye system_manager.py[975]: Running System State: BMC_STARTING
Aug 23 01:37:51 barreleye system_manager.py[975]: Running System State: HOST_POWERED_ON
Aug 23 01:38:33 barreleye system_manager.py[975]: Running System State: INVENTORY_UPLOADED
root@barreleye:~# 
````
Back to the commit 5051c6836a0880b001aa3fcf5539987b7aedd536, I did not see this situation.
````
root@barreleye:~# cat /etc/os-release 
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=v1.0.4-2-g5051c68-dirty
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=v1.0.4
````
````
root@barreleye:~# journalctl --no-pager|grep "System State"
Aug 23 01:37:48 barreleye system_manager.py[973]: Running System State: BASE_APPS
Aug 23 01:37:57 barreleye system_manager.py[973]: SystemManager Goto System State: BMC_STARTING
Aug 23 01:37:57 barreleye system_manager.py[973]: Running System State: BMC_STARTING
Aug 23 01:38:33 barreleye system_manager.py[973]: SystemManager Goto System State: BMC_STARTING2
Aug 23 01:38:33 barreleye system_manager.py[973]: Running System State: BMC_STARTING2
Aug 23 01:38:50 barreleye system_manager.py[973]: SystemManager Goto System State: BMC_READY
Aug 23 01:38:50 barreleye system_manager.py[973]: Running System State: BMC_READY
Aug 23 01:38:57 barreleye system_manager.py[973]: Last System State: ;  Power Policy: RESTORE_LAST_STATE
root@barreleye:~# 
````

amboar, 2016-08-23:
@johnhcwang I'm investigating this in #513

amboar, 2016-08-26:
@johnhcwang This is resolved by openbmc/skeleton@0f2cf104e25b6d613f15ca91a608879b55e188e5

closed 2016-08-26 by amboar