openbmc/openbmc #519: [Barelleye/Palmeto Build #191 ]  Warmrest not rebooting the BMC

gkeishin, 2016-08-23:
Build Name: #191-3636259108c2ec 	 Build Number: 191

---------------------------------------------------------
-----------------
Build Info:
-----------------
Build #191-3636259108c2ec (Aug 22, 2016 5:50:52 PM) 

-----------------
Build Change info:
-----------------

Changes
Git (git https://github.com/openbmc/openbmc.git)
yocto-2.1: meta-yocto to meta-poky transition (commit: e18c612) (detail)
yocto-poky: Move to import-layers subdir (commit: 60f9d69) (detail)
Squashed 'import-layers/meta-virtualization/' content from commit (commit: d849ec7) (detail)
Squashed 'import-layers/meta-openembedded/' content from commit 247b126 (commit: b48b7b4) (detail)
Squashed 'import-layers/meta-raspberrypi/' content from commit 2745399 (commit: 8b8bc41) (detail)
skeleton-rev.bbclass: bump version (commit: 549aa3d) (detail)
pyphosphor.bb: bump version (commit: 3636259) (detail)
Started by GitHub push by williamspatrick
Revision: 3636259108c2ecb4e6cb1d363c4d27d8737b4ca6
refs/remotes/origin/master

The REST method
..URI:https://9.3.164.153/org/openbmc/control/bmc0/action/warmReset, method:Post, args:{u'data': {u'data': []}, u'headers': {u'Content-Type': u'application/json'}}
Response code:200, Content:{
  "data": null, 
  "message": "200 OK", 
  "status": "ok"
}

Doesn't seems to apply reboot onto the BMC. I manually ran the warm reset test but in actual it didnt reboot the BMC.

@williamspatrick  Is this a known issue ?

williamspatrick, 2016-08-23:
Until we get to the bottom of #513, the master code doesn't even get to standby correctly. I have no expectation that a reboot will work.

causten, 2016-08-23:
This issue also fails with v1.0.5 stable.  Journal message is...
Aug 22 21:33:08 palmetto system_manager.py[666]: warmReset() error: Failed to read enough data from child pid pipe (No child processes)

anoo1, 2016-08-23:
Not seeing this issue on Barreleye v1.0.5.
The logs are only from Palmettos, will try on that platform next.

gkeishin, 2016-08-23:
@anoo1 Here is a palmetto log..  attached
 630 Aug 23 08:22:32 palmetto system_manager.py[708]: warmReset() error: Failed to read enough data from child pid pipe (No child processes)

[journalctl.palmetto.9.3.164.153.txt](https://github.com/openbmc/openbmc/files/433081/journalctl.palmetto.9.3.164.153.txt)

gkeishin, 2016-08-23:
@anoo1 Here is a palmetto log..  attached
 630 Aug 23 08:22:32 palmetto system_manager.py[708]: warmReset() error: Failed to read enough data from child pid pipe (No child processes)

[journalctl.palmetto.txt](https://github.com/openbmc/openbmc/files/433091/journalctl.palmetto.txt)

geissonator, 2016-08-25:
Here's a trace from my Palmetto with all the recent gpio related fixes on the warm reset:

eb 29 19:45:26 palmetto system_manager.py[696]: Download complete. Updating...
Feb 29 19:45:29 palmetto system_manager.py[696]: Preserving network settings
Feb 29 19:45:29 palmetto system_manager.py[696]: update: --no-flash --no-save-files --no-restore-files --no-clean-saved-files
Feb 29 19:45:30 palmetto system_manager.py[696]: /dev/mtdblock5 /run/initramfs/ro squashfs ro,relatime 0 0
Feb 29 19:45:30 palmetto system_manager.py[696]: ERROR: Device mtd5 is mounted, image-bmc is busy.
Feb 29 19:45:31 palmetto system_manager.py[696]: warmReset() error: Failed to read enough data from child pid pipe (No child processes)
Feb 29 19:45:49 palmetto system_manager.py[696]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/

anoo1, 2016-08-25:
This is broken on master (both barreleye and palmetto fail the same way) , but works on stable (on both palmetto and barreleye).

geissonator, 2016-09-26:
Did a little more digging on this one since I hit it today with a test case.

Here's my log
```
Sep 20 13:22:08 barreleye phosphor-gevent[910]: ::c0f3:adb6:8e91:f402%1339722056 - - [2016-09-20 13:22:08] "POST /login HTTP/1.1" 200 320 0.040699
Sep 20 13:22:09 barreleye control_bmc.exe[679]: warmReset() error: Failed to read enough data from child pid pipe (No child processes)
Sep 20 13:22:09 barreleye phosphor-gevent[910]: ::c0f3:adb6:8e91:f402%1339722056 - - [2016-09-20 13:22:09] "POST /org/openbmc/control/bmc0/action/warmReset HTTP/1.1" 200 169 0.362596
```

Here's the code:

```
static gboolean
on_warm_reset(ControlBmc *bmc,
		GDBusMethodInvocation *invocation,
		gpointer user_data)
{
	GError *err = NULL;
	/* Wait a while before reboot, so the caller can be responded.
	 * Note that g_spawn_command_line_async() cannot parse ';' as
	 * a command separator. Need to use 'sh -c' to let shell parse it.
	 */
	gchar *reboot_command = "/bin/sh -c 'sleep 3;reboot'";

	g_spawn_command_line_async(reboot_command, &err);
	if(err != NULL) {
		fprintf(stderr, "warmReset() error: %s\n", err->message);
		g_error_free(err);
	}

	control_bmc_complete_warm_reset(bmc, invocation);
	return TRUE;
}
```

I manually tried the command and it worked as expected so something to do with the g_spawn_asynch maybe.  Nothing obvious jumping out though.  It's running into some sort of issue with starting the asynch child process.

mdmillerii, 2016-09-28:
I've been interpreting the error message as a short read of data across DBUS from the REST server and not related to actually invoking the async service.  But I see now it say "from child pid pipe" , so I looked at the `glib` source.

The error says the parent was unable to read 1 `int` worth of bytes from the pipe that the child process should have written after spawning the grandchild and gettings its pid.  As `errno` is `ECHILD` its likely from the `waitpid` for the intermediate child, and `read_ints` returned true indicating no error (just read to EOF and divide by `sizeof(gint)`).  The `fork` failed and `fork` parent branch of the intermediate child both write to the pipe, so I don't see an obvious error.

If the child actually fails it should write its own error status to the error pipe, which is checked before the pid pipe.  Since the reboot does not occur it would seem some path along the double fork failed or the setup of the pipe.

mdmillerii, 2016-09-28:
yocto-2.0.1 updated from glib-2.0_2.44.1 to 2.0_2.46.1, but glib-unix.c and gspawn.c are identical upstream.

anoo1, 2016-09-29:
@bradbishop recommends that we replace the g_spawn_command_line_async() call that spawns the request in the background with a call to a systemd api.

mdmillerii, 2016-09-29:
@bradbishop that won't address the desire to ack the ipmi before doing the reset ... but it would likely bypass whatever glib issues are being encountered.

williamspatrick, 2016-09-30:
Make sure you call Reboot on logind object and not the systemd object.  The documentation states that systemd interface performs a reboot immediately and does not shutdown any processes, while the logind one goes through a graceful shutdown.  (Appears to be reboot vs reboot -f).

https://www.freedesktop.org/wiki/Software/systemd/logind/
https://www.freedesktop.org/wiki/Software/systemd/dbus/

bradbishop, 2016-10-02:
I had mentioned that to Andrew but he found that logind isn't running.  I think it may have been removing during one of the footprint pruning exercises...

geissonator, 2016-10-03:
- [ ] Fix up formatting in bmcctl/control_bmc_obj.c
- [ ] Call reboot interface which will shutdown process's prior to reboot

williamspatrick, 2016-10-03:
You might also check with Eddie due to issue #619.  It is possible that due to PAM support we added we also need to add logind back in.

williamspatrick, 2016-10-03:
@eddiejames -- see previous comment.

sivassrr, 2016-11-08:
@gkeishin -Are we still seeing it in the latest driver? could you try it on

gkeishin, 2016-11-10:
@sivassrr  This issues still persist..

closed 2016-11-18 by williamspatrick