openbmc/openbmc #520: [Barreleye Build#192 ]Test WarmReset via REST /tmp/before_warmreset  persist after reboot

gkeishin, 2016-08-24:
Build Name: #192-212e8d678ffb8d 	 Build Number: 192
-----------------
Build Info:
-----------------
Build #192-212e8d678ffb8d (Aug 23, 2016 9:26:42 PM) 

-----------------
Build Change info:
-----------------

Changes
Git (git https://github.com/openbmc/openbmc.git)
Remove /sbin/ldconfig (commit: 212e8d6) (detail)
Started by GitHub push by williamspatrick
Revision: 212e8d678ffb8db8d6718af2c8f61796d86904d0
refs/remotes/origin/master

---------------------------------------------------------

Test case : openbmc-test-automation/tests/test_warmreset.robot

The test creates a file under /tmp 

${stdout}, ${stderr}, ${rc} = SSHLibrary . Execute Command touch ${warm_test_file}, return_stderr=True, return_rc=True
Documentation: 	

Executes `command` on the remote machine and returns its outputs.
Start / End / Elapsed: 	20160824 15:17:02.247 / 20160824 15:17:02.318 / 00:00:00.071
15:17:02.248 	INFO 	Executing command 'touch /tmp/before_warmreset'. 	
15:17:02.317 	INFO 	Command exited with return code 0. 	
15:17:02.318 	INFO 	${stdout} = 	
15:17:02.318 	INFO 	${stderr} = 	
15:17:02.318 	INFO 	${rc} = 0

AFTER reboot

${stdout}, ${stderr}, ${rc} = SSHLibrary . Execute Command ls ${warm_test_file}, return_stderr=True, return_rc=True
Documentation: 	

Executes `command` on the remote machine and returns its outputs.
Start / End / Elapsed: 	20160824 15:22:15.660 / 20160824 15:22:15.730 / 00:00:00.070
15:22:15.660 	INFO 	Executing command 'ls /tmp/before_warmreset'. 	
15:22:15.729 	INFO 	Command exited with return code 0. 	
15:22:15.730 	INFO 	${stdout} = /tmp/before_warmreset 	
15:22:15.730 	INFO 	${stderr} = 	
15:22:15.730 	INFO 	${rc} = 0

File /tmp/before_warmreset does exist even after reboot of BMC, error:, stdput: /tmp/before_warmreset: 0 != 1

This indicates this file persist event after warm reset.

mdmillerii, 2016-08-24:
Although the current implementation creates a new `/tmp` for each boot, the [FHS](http://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch03s18.html) does not require it.
The test should be in a subdirectory of `/run`.

That said the error output should state the expected behavior and the implications of the file existing.
In this case its likely to the other issue causing warm reset to not be serviced.

gkeishin, 2016-08-25:
@mdmillerii Ok that make sense Will keep an eye on this in next driver if this persist...

gkeishin, 2016-08-31:
In last couple of driver seeing timeout connection so it has reached to point where it comes back.. probably due to network and reset issues.. but will still keep eye open for this one once it passes those issues..

gkeishin, 2016-10-04:
The Current few driver are not yet good to confirm this. Will check on it...

gkeishin, 2016-10-07:
Not seeing in latter drivers.. closing this

closed 2016-10-07 by gkeishin