openbmc/openbmc #521: [Barreleye Build#192 ]  minimal fan inventory

gkeishin, 2016-08-24:
Build Name: #192-212e8d678ffb8d 	 Build Number: 192
-----------------
Build Info:
-----------------
Build #192-212e8d678ffb8d (Aug 23, 2016 9:26:42 PM) 

-----------------
Build Change info:
-----------------

Changes
Git (git https://github.com/openbmc/openbmc.git)
Remove /sbin/ldconfig (commit: 212e8d6) (detail)
Started by GitHub push by williamspatrick
Revision: 212e8d678ffb8db8d6718af2c8f61796d86904d0
refs/remotes/origin/master

---------------------------------------------------------

Test Case , check if this has to have a minimal 2 or more enable / present in the inventory, but turns out NONE are present..

minimal fan inventory
        ${count} =      Get Total Present       FAN
        Should Be True  ${count}>${2}

Documentation: 	
Fails if the given condition is not true.
Start / End / Elapsed: 	20160824 15:03:11.229 / 20160824 15:03:11.231 / 00:00:00.002
15:03:11.231 	FAIL 	'0>2' should be true.

@williamspatrick  @causten Is this a known setup for now or a problem that needs to be looked into. I have seen this for couple of drivers, but decided to open this as it is now more persistent error in the test runs.

sivassrr, 2016-08-31:
@gkeishin Need to change our test code whether we are in host booted state then we need to run the inventory. (There is some difference between FSP And BMC behavior especially in inventory).

gkeishin, 2016-08-31:
@sivassrr  ok that make sense..

gkeishin, 2016-08-31:
nope @sivassrr I added code to wait for host to boot.. still seeing issues

minimal cpu inventory                                                 | PASS |
------------------------------------------------------------------------------
minimal dimm inventory                                                | PASS |
------------------------------------------------------------------------------
minimal core inventory                                                | PASS |
------------------------------------------------------------------------------
minimal memory buffer inventory                                       | PASS |
------------------------------------------------------------------------------
minimal fan inventory                                                 .F
FFDC Collection Started 	: 2016-08-31 02:08:52,849403
FFDC Collection Completed 	: 2016-08-31 02:09:18,252501
minimal fan inventory                                                 | FAIL |
'0>2' should be true.
------------------------------------------------------------------------------
minimal main planar inventory                                         | PASS |
------------------------------------------------------------------------------
minimal system inventory                                              | PASS |
------------------------------------------------------------------------------
Verify CPU VPD Properties                                             F
FFDC Collection Started 	: 2016-08-31 02:09:18,462062
FFDC Collection Completed 	: 2016-08-31 02:09:46,560333
Verify CPU VPD Properties                                             | FAIL |
Lengths are different: 18 != 5
------------------------------------------------------------------------------
Verify DIMM VPD Properties                                            F
FFDC Collection Started 	: 2016-08-31 02:09:46,688601
FFDC Collection Completed 	: 2016-08-31 02:09:49,812465
Verify DIMM VPD Properties                                            | FAIL |
Lengths are different: 20 != 5
------------------------------------------------------------------------------
Verify Memory Buffer VPD Properties                                   F
FFDC Collection Started 	: 2016-08-31 02:09:55,234662
FFDC Collection Completed 	: 2016-08-31 02:10:41,204072
Verify Memory Buffer VPD Properties                                   | FAIL |
Lengths are different: 18 != 5
------------------------------------------------------------------------------
Verify Fan VPD Properties                                             | PASS |
------------------------------------------------------------------------------
Verify System VPD Properties                                          F
FFDC Collection Started 	: 2016-08-31 02:10:41,350433
FFDC Collection Completed 	: 2016-08-31 02:11:03,358407
Verify System VPD Properties                                          | FAIL |
Lengths are different: 19 != 5
------------------------------------------------------------------------------
Test Inventory :: This testsuite is for testing inventory             | FAIL |
12 critical tests, 7 passed, 5 failed
12 tests total, 7 passed, 5 failed
==============================================================================

FFDC for all the test failure attached here:
[FFDC_inventory.tar.gz](https://github.com/openbmc/openbmc/files/446819/FFDC_inventory.tar.gz)

williamspatrick, 2016-09-06:
I've said many times to stop referring to something like 'build #192'. Please use the os-release information and/or git commit number.  It appears that this defect was opened with code level:
```
212e8d6 Remove /sbin/ldconfig
```

This appears to be a duplicate of #93, which is assigned to someone already.

closed 2016-09-06 by williamspatrick