openbmc/openbmc #522: Code update of BMC flash hangs during erase

geissonator, 2016-08-24:
This has been seen internally a few times now.  It seems to hang at different points during the erase.

I was flashing a barreley with master and got the hang here.  Note that this had the master code which has the bug where warm reboot doesn't work so I manually called "reboot" after setting up the update using a scp of the .all.tar file to /tmp/ and the REST api's (using causten codeupdate.py tool).

Here's the tool from Chris we're using - https://github.com/causten/tools/blob/master/obmc/codeupdate.py

Patrick is thinking the tool itself may need to first reboot us into running from RAM, and then copy the image into /tmp/, and then do the update.

@mdmillerii thoughts?

```
root@barreleye:~# reboot
[  OK  ] Removed slice system-getty.slice.
[  OK  ] Stopped target System Time Synchronized.
[  OK  ] Stopped target Network.
         Stopping SSH Per-Connection Server...
[  OK  ] Stopped target Timers.
[  OK  ] Stopped Daily Cleanup of Temporary Directories.
[  OK  ] Stopped target Multi-User System.
         Stopping Phosphor OpenBMC DBus REST daemon...
         Stopping Settings DBUS Object...
         Stopping DNS update handler...
         Stopping Phosphor OpenBMC IPMI daemon...
         Stopping Login Service...
         Stopping Phosphor OpenBMC BT to DBUS...
[  OK  ] Stopped target Login Prompts.
         Stopping Serial Getty on ttyS4...
[  OK  ] Stopped target OpenBMC System.
         Stopping Kernel Logging Service...
         Stopping Phosphor OpenBMC user management daemon...
         Stopping OpenBMC system manager...
         Stopping System Logging Service...
         Stopping Network Service...
[  OK  ] Stopped target Remote File Systems.
         Stopping Phosphor OpenBMC event management daemon...
         Stopping Phosphor Inverse ARP daemon...
[  OK  ] Stopped Phosphor OpenBMC user management daemon.
[  OK  ] Stopped Phosphor OpenBMC event management daemon.
[  OK  ] Stopped Settings DBUS Object.
[  OK  ] Stopped Phosphor Inverse ARP daemon.
[  OK  ] Stopped Phosphor OpenBMC BT to DBUS.
[  OK  ] Stopped DNS update handler.
[  OK  ] Stopped Kernel Logging Service.
[  OK  ] Stopped System Logging Service.
[  OK  ] Stopped Serial Getty on ttyS4.
[  OK  ] Stopped Network Service.
[  OK  ] Stopped Login Service.
[  OK  ] Stopped Phosphor OpenBMC DBus REST daemon.
[  OK  ] Stopped OpenBMC system manager.
[  OK  ] Stopped SSH Per-Connection Server.
[  OK  ] Stopped Phosphor OpenBMC IPMI daemon.
[  OK  ] Stopped Clear one time boot overrides.
[  OK  ] Removed slice system-dropbear.slice.
         Stopping Phosphor OpenBMC DBus service management daemon...
         Stopping D-Bus System Message Bus...
[  OK  ] Removed slice system-serial\x2dgetty.slice.
         Stopping Network DBUS object...
[  OK  ] Stopped Phosphor OpenBMC DBus service management daemon.
[  OK  ] Stopped D-Bus System Message Bus.
[  OK  ] Stopped Network DBUS object.
[  OK  ] Stopped target Basic System.
[  OK  ] Stopped target Sockets.
[  OK  ] Closed dropbear.socket.
[  OK  ] Closed D-Bus System Message Bus Socket.
[  OK  ] Closed Syslog Socket.
[  OK  ] Stopped target Paths.
[  OK  ] Stopped Forward Password Requests to Wall Directory Watch.
[  OK  ] Stopped Dispatch Password Requests to Console Directory Watch.
[  OK  ] Stopped target Slices.
[  OK  ] Removed slice User and Session Slice.
[  OK  ] Stopped target System Initialization.
         Stopping Load/Save Random Seed...
         Stopping Network Time Synchronization...
[  OK  ] Stopped target Swap.
[  OK  ] Stopped Apply Kernel Variables.
[  OK  ] Stopped Network Time Synchronization.
[  OK  ] Stopped Load/Save Random Seed.
[  OK  ] Stopped Create Volatile Files and Directories.
[  OK  ] Stopped target Local File Systems.
         Unmounting /run/initramfs/rw...
         Unmounting Temporary Directory...
         Unmounting /run/initramfs/ro...
         Unmounting /var/volatile...
[  OK  ] Unmounted /run/initramfs/rw.
[  OK  ] Unmounted Temporary Directory.
[  OK  ] Unmounted /run/initramfs/ro.
[  OK  ] Unmounted /var/volatile.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped target Local File Systems (Pre).
[  OK  ] Stopped Create Static Device Nodes in /dev.
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Reached target Shutdown.
watchdog: watchdog0: watchdog did not stop!
systemd-shutdown[1]: Sending SIGTERM to remaining processes...
systemd-journald[399]: Received SIGTERM from PID 1 (systemd-shutdow).
systemd-shutdown[1]: Sending SIGKILL to remaining processes...
systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
systemd-shutdown[1]: Unmounting file systems.
systemd-shutdown[1]: Remounting '/' read-only with options 'lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work'.
systemd-shutdown[1]: Remounting '/' read-only with options 'lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work'.
systemd-shutdown[1]: All filesystems unmounted.
systemd-shutdown[1]: Deactivating swaps.
systemd-shutdown[1]: All swaps deactivated.
systemd-shutdown[1]: Detaching loop devices.
systemd-shutdown[1]: All loop devices detached.
systemd-shutdown[1]: Detaching DM devices.
systemd-shutdown[1]: All DM devices detached.
systemd-shutdown[1]: Successfully changed into root pivot.
systemd-shutdown[1]: Returning to initrd...
watchdog: watchdog0: watchdog did not stop!
shutdown: reboot --log-level 6 --log-target kmsg --log-color
+ sort -r
+ awk /oldroot|mnt/ { print $2 }
+ umount /oldroot/sys/kernel/debug
+ umount /oldroot/sys/kernel/config
+ umount /oldroot/sys/fs/cgroup/systemd
+ umount /oldroot/sys/fs/cgroup
+ umount /oldroot/sys
+ umount /oldroot/proc
+ umount /oldroot/dev/shm
+ umount /oldroot/dev/pts
+ umount /oldroot/dev
+ umount /oldroot
+ umount /mnt
+ set +x
update: --clean-saved-files
jffs2: notice: (1371) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
Updating bmc...
Erasing block: 6646/8192 (81%) 
```

mdmillerii, 2016-08-24:
So it made it to 81% and then you claim that was end of output ...

I see this line:

    watchdog: watchdog0: watchdog did not stop!

That says systemd was unable to stop the watchdog, which would start a timer.

Please review the kernel configuration to make sure the `WATCHDOG_NO_WAY_OUT` is not set.  I probably have the name wrong.

If the watchdog fires at this point the in the process u-boot has been erased and there is no backup flash (and it has not been enabled for the platforms that have it).

geissonator, 2016-08-24:
So this was definitely the slowest flash update I've seen.  Sounds like we hit the bootloader watchdog then?  Is there a way to disable that during code updates?

mdmillerii, 2016-08-24:
hmm, so that watchdog message came from [the watchdog core]( https://github.com/openbmc/linux/blob/dev-4.6/drivers/watchdog/watchdog_dev.c#L735) not from the core.  Will need someone to investigate why the driver failed to turn off the watchdog.

I verified my palmetto build allows it to be shut off:
`tmp/work/palmetto-openbmc-linux-gnueabi/linux-obmc/4.6+gitAUTOINC+2d15238f37-r0/linux-palmetto-standard-build/.config`:

    CONFIG_WATCHDOG=y
    CONFIG_WATCHDOG_CORE=y
    # CONFIG_WATCHDOG_NOWAYOUT is not set
    # CONFIG_WATCHDOG_SYSFS is not set
    ...
    CONFIG_ASPEED_WATCHDOG=y

amboar, 2016-08-25:
Probably not surprising, but I'm seeing this on palmetto also. Starting to investigate.

amboar, 2016-08-25:
@geissonator was complaining that the flash process was slow, and that we possibly tripped the watchdog. I believe this also to be the case, and the solution is to increase systemd's `ShutdownWatchdogSec` value from 10min to something larger.

Here's Lennart's article on systemd and watchdogs:

http://0pointer.de/blog/projects/watchdog.html

We don't appear to set `RuntimeWatchdogSec`, so that feature should default to disabled:

> To make use of the hardware watchdog it is sufficient to set the RuntimeWatchdogSec= option in /etc/systemd/system.conf. It defaults to 0 (i.e. no hardware watchdog use).

However, there's a second watchdog configuration option:

> ShutdownWatchdogSec= is another option that can be configured in /etc/systemd/system.conf. It controls the watchdog interval to use during reboots. It defaults to 10min, and adds extra reliability to the system reboot logic: if a clean reboot is not possible and shutdown hangs, we rely on the watchdog hardware to reset the system abruptly, as extra safety net.

Booting a lightly patched 4.7 kernel (for debugging) with the OpenBMC userspace under a lightly patched qemu (also for debugging) shows the following behaviour towards the end of the reboot sequence:

```
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
aspeed_wdt_write: offset=0x4, data=0x23c34600, size=4
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
watchdog: drivers/watchdog/watchdog_dev.c : watchdog_release : 749
watchdog: watchdog0: watchdog did not stop!
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
watchdog: drivers/watchdog/watchdog_dev.c : watchdog_release : 757
watchdog: drivers/watchdog/watchdog_dev.c : watchdog_release : 765
watchdog: drivers/watchdog/watchdog_dev.c : watchdog_release : 774
watchdog: drivers/watchdog/watchdog_dev.c : watchdog_release : 778
systemd-shutdown[1]: Sending SIGTERM to remaining processes...
systemd-journald[321]: Received SIGTERM from PID 1 (systemd-shutdow).
systemd-shutdown[1]: Sending SIGKILL to remaining processes...
systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
systemd-shutdown[1]: Unmounting file systems.
systemd-shutdown[1]: Remounting '/' read-only with options 'size=245296k,nr_inodes=61324'.
systemd-shutdown[1]: Remounting '/' read-only with options 'size=245296k,nr_inodes=61324'.
systemd-shutdown[1]: All filesystems unmounted.
systemd-shutdown[1]: Deactivating swaps.
systemd-shutdown[1]: All swaps deactivated.
systemd-shutdown[1]: Detaching loop devices.
systemd-shutdown[1]: All loop devices detached.
systemd-shutdown[1]: Detaching DM devices.
systemd-shutdown[1]: All DM devices detached.
systemd-shutdown[1]: Rebooting.
reboot: Restarting system
aspeed_wdt_write: offset=0xc, data=0x0, size=4
aspeed_wdt_write: offset=0x4, data=0x1f400, size=4
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
aspeed_wdt_write: offset=0xc, data=0x13, size=4
```
Several things are happening here:

1. Constant feeding of the watchdog with: `aspeed_wdt_write: offset=0x8, data=0x4755, size=4`
2. A change in the watchdog reset value: `aspeed_wdt_write: offset=0x4, data=0x23c34600, size=4`
3. Execution path tracing through watchdog_dev.c. Of interest is `watchdog: drivers/watchdog/watchdog_dev.c : watchdog_release : 749`
4. We hit the `watchdog: watchdog0: watchdog did not stop!` referenced above
5. And finish off with a few further writes to the watchdog, including disabling it: aspeed_wdt_write: offset=0xc, data=0x0, size=4, reprogramming and re-enabling: `aspeed_wdt_write: offset=0xc, data=0x13, size=4`

Points 1 and 5 are largely uninteresting at this point. So, addressing the 3rd and 4th points, @mdmillerii linked to the `pr_crit()` call that prints our "watchdog did not stop!" message. The context is:

```c
static int watchdog_release(struct inode *inode, struct file *file)
{
	struct watchdog_core_data *wd_data = file->private_data;
	struct watchdog_device *wdd;
	int err = -EBUSY;
	bool running;

    ...

	/*
	 * We only stop the watchdog if we received the magic character
	 * or if WDIOF_MAGICCLOSE is not set. If nowayout was set then
	 * watchdog_stop will fail.
	 */
	if (!test_bit(WDOG_ACTIVE, &wdd->status))
		err = 0;
	else if (test_and_clear_bit(_WDOG_ALLOW_RELEASE, &wd_data->status) ||
		 !(wdd->info->options & WDIOF_MAGICCLOSE))
		err = watchdog_stop(wdd);

	/* If the watchdog was not stopped, send a keepalive ping */
	if (err < 0) {
		pr_crit("watchdog%d: watchdog did not stop!\n", wdd->id);
		watchdog_ping(wdd);
	}
```
Hitting my `watchdog: drivers/watchdog/watchdog_dev.c : watchdog_release : 749` log message showed that neither of the conditions above where `err` is assigned were met, before the `pr_crit()` call. Interestingly this means that systemd considers the watchdog active, but doesn't shut it down cleanly (either by the DISABLECARD ioctl or writing 'V' to `/dev/watchdog`). This is contrary to the configuration file which doesn't set `RuntimeWatchdogSec` and probably needs further investigation.

The remaining point is the reprogramming of the reset value to 0x23c34600. The watchdog is driven at either the PCLK rate or 1MHz in the AST2400 case or at 1MHz in the AST2500 case. To cater for both the AST2400 and AST2500 the watchdog driver chooses the 1MHz source, and with this context the 0x23c34600 value gives 10 minutes. If Barreleye is slow to write its flash, we should consider increasing systemd's `ShutdownWatchdogSec` value.

williamspatrick, 2016-08-25:
@amboar - Any ideas on why the disable is happening?  As @mdmillerii mentioned, we don't the error message with the same userspace but older kernel.

Having systemd do the default behavior of 10 minute watchdog on shutdown seems reasonable.  You mentioned the periodic poke of the watchdog at running (item 1 above).  What is the time value being set there?  If we don't have a watchdog at runtime, we should, so we can make a minor change to the systemd config as needed to insert one.

@mdmillerii - Would it be reasonable to update the initrd scripts to push the watchdog to 1-2hrs specifically while we do the code update?  I don't want to generally increase the shutdown watchdog to that length.  I assume it is a relatively straight-forward `echo X > /dev/watchdog`.

amboar, 2016-08-25:
>  As @mdmillerii mentioned, we don't the error message with the same userspace but older kernel.

It looks like the warning is hidden behind another test in 4.4:

`git diff obmc/dev-4.{4,7} -- drivers/watchdog/watchdog_dev.c`:
```udiff
       /* If the watchdog was not stopped, send a keepalive ping */
        if (err < 0) {
-               mutex_lock(&wdd->lock);
-               if (!test_bit(WDOG_UNREGISTERED, &wdd->status))
-                       dev_crit(wdd->dev, "watchdog did not stop!\n");
-               mutex_unlock(&wdd->lock);
+               pr_crit("watchdog%d: watchdog did not stop!\n", wdd->id);
                watchdog_ping(wdd);
        }
```
I'll poke around at that difference a bit further tomorrow.

williamspatrick, 2016-08-25:
@amboar - Systemd uses the ioctl to disable the watchdog and then sets the V character to disable it.   Effectively trying two approaches to shut it down: https://github.com/systemd/systemd/blob/09541e49ebd17b41482e447dd8194942f39788c0/src/shared/watchdog.c#L144

I suspect that the removal of the current-config check causes this spurious dmesg.  Agreed?

mdmillerii, 2016-08-25:
@williamspatrick  That may explain why the message is printed after that diff, but we still need to figure out why the watchdog is still armed.  I have had success with flashing that took 30 minutes at that point in time in prior kernels (possibly in the 4.3 timeframe), so it was being disarmed before.

amboar, 2016-08-26:
> @amboar - Systemd uses the ioctl to disable the watchdog and then sets the V character to disable it. Effectively trying two approaches to shut it down: https://github.com/systemd/systemd/blob/09541e49ebd17b41482e447dd8194942f39788c0/src/shared/watchdog.c#L144

Right, so what I meant by:

> Interestingly this means that systemd considers the watchdog active, but doesn't shut it down cleanly (either by the DISABLECARD ioctl or writing 'V' to /dev/watchdog). 

Was that I was aware of the systemd code you reference and had instrumented the kernel side. However I didn't see any log output suggesting that systemd was either writing 'V' to `/dev/watchdog` or invoking the `DISABLECARD` ioctl.

> 
> I suspect that the removal of the current-config check causes this spurious dmesg. Agreed?

That was my thought.

amboar, 2016-08-26:
Here's a trace from a dev-4.4 kernel with the same userspace as above (from openbmc/master):

```
aspeed_wdt_write: offset=0xc, data=0x0, size=4
aspeed_wdt_write: offset=0x4, data=0xe4e1c00, size=4
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
aspeed_wdt_write: offset=0xc, data=0x3, size=4
aspeed_wdt_write: offset=0xc, data=0x0, size=4
[   35.540000] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[   35.600000] systemd-journald[274]: Received SIGTERM from PID 1 (systemd-shutdow).
[   35.600000] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
aspeed_wdt_write: offset=0xc, data=0x0, size=4
aspeed_wdt_write: offset=0x4, data=0xe4e1c00, size=4
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
aspeed_wdt_write: offset=0xc, data=0x3, size=4
[   35.670000] systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
[   35.670000] systemd-shutdown[1]: Unmounting file systems.
[   35.670000] systemd-shutdown[1]: Remounting '/' read-only with options 'size=244812k,nr_inodes=61203'.
[   35.670000] systemd-shutdown[1]: Remounting '/' read-only with options 'size=244812k,nr_inodes=61203'.
[   35.740000] systemd-shutdown[1]: All filesystems unmounted.
[   35.740000] systemd-shutdown[1]: Deactivating swaps.
[   35.740000] systemd-shutdown[1]: All swaps deactivated.
[   35.740000] systemd-shutdown[1]: Detaching loop devices.
[   35.740000] systemd-shutdown[1]: All loop devices detached.
[   35.740000] systemd-shutdown[1]: Detaching DM devices.
[   35.740000] systemd-shutdown[1]: All DM devices detached.
[   35.740000] systemd-shutdown[1]: Rebooting.
[   35.740000] reboot: Restarting system
aspeed_wdt_write: offset=0xc, data=0x0, size=4
aspeed_wdt_write: offset=0x4, data=0x1c36c8, size=4
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
aspeed_wdt_write: offset=0xc, data=0x3, size=4
```
In this case the driver uses the rate from the APB clock which the devicetree specifies as 48MHz. At 48MHz 0xe4e1c00 is 5 seconds, and 0x1c36c8 is ~39ms. I am still looking into why the behaviour is so different.

amboar, 2016-08-26:
Okay, previously I said:

> This is contrary to the configuration file which doesn't set RuntimeWatchdogSec and probably needs further investigation.

That's actually inaccurate. I just remembered I had been experimenting with a tweaked initrd where I had set `RuntimeWatchdogSec=10s`. So disregard the quote (or consider it resolved).

amboar, 2016-08-26:
Having investigated further, and the essence of it is `watchdog: watchdog0: watchdog did not stop!` is expected behaviour when `ShutdownWatchdogSec` > 0. This seems completely sensible.

Here both `RuntimeWatchdogSec=0` and `ShutdownWatchdogSec=0` we see no output from the watchdog tracing in qemu, aside from the driver's restart callback (the final lines):
```
$ qemu-system-arm -M palmetto-bmc -m 512 -kernel ~/env/openbmc/openbmc/issues/522/dev-4.7/zImage -dtb ~/env/openbmc/openbmc/issues/522/dev-4.7/aspeed-bmc-opp-palmetto.dtb -initrd ~/env/openbmc/openbmc/issues/522/initrd.rws0.sws0.cpio.gz -nographic
...
systemd-shutdown[1]: Sending SIGTERM to remaining processes...
systemd-journald[333]: Received SIGTERM from PID 1 (systemd-shutdow).
systemd-shutdown[1]: Sending SIGKILL to remaining processes...
systemd-shutdown[1]: Unmounting file systems.
systemd-shutdown[1]: Remounting '/' read-only with options 'size=245296k,nr_inodes=61324'.
systemd-shutdown[1]: Remounting '/' read-only with options 'size=245296k,nr_inodes=61324'.
systemd-shutdown[1]: All filesystems unmounted.
systemd-shutdown[1]: Deactivating swaps.
systemd-shutdown[1]: All swaps deactivated.
systemd-shutdown[1]: Detaching loop devices.
systemd-shutdown[1]: All loop devices detached.
systemd-shutdown[1]: Detaching DM devices.
systemd-shutdown[1]: All DM devices detached.
systemd-shutdown[1]: Rebooting.
reboot: Restarting system
aspeed_wdt_write: offset=0xc, data=0x0, size=4
aspeed_wdt_write: offset=0x4, data=0x1f400, size=4
aspeed_wdt_write: offset=0x8, data=0x4755, size=4
aspeed_wdt_write: offset=0xc, data=0x13, size=4
Reboot failed -- System halted
```
Some snippets from `src/core/main.c`:

```
{ "Manager", "ShutdownWatchdogSec",       config_parse_sec,              0, &arg_shutdown_watchdog                 },
```

```c
                case MANAGER_REBOOT:
                case MANAGER_POWEROFF:
                case MANAGER_HALT:
                case MANAGER_KEXEC: {
                        static const char * const table[_MANAGER_EXIT_CODE_MAX] = {
                                [MANAGER_EXIT] = "exit",
                                [MANAGER_REBOOT] = "reboot",
                                [MANAGER_POWEROFF] = "poweroff",
                                [MANAGER_HALT] = "halt",
                                [MANAGER_KEXEC] = "kexec"
                        };

                        assert_se(shutdown_verb = table[m->exit_code]);
                        arm_reboot_watchdog = m->exit_code == MANAGER_REBOOT;

                        log_notice("Shutting down.");
                        goto finish;
                }
```

```c
                if (arm_reboot_watchdog && arg_shutdown_watchdog > 0) {
                        char *e;

                        /* If we reboot let's set the shutdown
                         * watchdog and tell the shutdown binary to
                         * repeatedly ping it */
                        r = watchdog_set_timeout(&arg_shutdown_watchdog);
                        watchdog_close(r < 0);

                        /* Tell the binary how often to ping, ignore failure */
                        if (asprintf(&e, "WATCHDOG_USEC="USEC_FMT, arg_shutdown_watchdog) > 0)
                                (void) strv_push(&env_block, e);
                } else
                        watchdog_close(true);
```
So we are rebooting, therefore `arm_reboot_watchdog` is set, and lets assume `ShutdownWatchdogSec` is set, therefore `arg_shutdown_watchdog > 0` is true. `watchdog_set_timeout()` succeeds as we've seen from the reload value written in the qemu tracing, and we have `r == 0`. Thus `watchdog_close()` is called with `false` as the argument, and as we see below this means we do *not* disarm the watchdog:

```c

void watchdog_close(bool disarm) {
        int r;

        if (watchdog_fd < 0)
                return;

        if (disarm) {
                int flags;

                /* Explicitly disarm it */
                flags = WDIOS_DISABLECARD;
                r = ioctl(watchdog_fd, WDIOC_SETOPTIONS, &flags);
                if (r < 0)
                        log_warning_errno(errno, "Failed to disable hardware watchdog: %m");

                /* To be sure, use magic close logic, too */
                for (;;) {
                        static const char v = 'V';

                        if (write(watchdog_fd, &v, 1) > 0)
                                break;

                        if (errno != EINTR) {
                                log_error_errno(errno, "Failed to disarm watchdog timer: %m");
                                break;
                        }
                }
        }

        watchdog_fd = safe_close(watchdog_fd);
}
```

mdmillerii, 2016-08-31:
I'll point out that the code in `src/core/main

amboar, 2016-10-06:
@williamspatrick @mdmillerii what is the plan for this issue? Do we extend the watchdog timeout? Something else?

geissonator, 2016-11-08:
This issue popped back up again on BarD04 today.  I assume the new pflash updates we put into the code recently are the issue (we bricked 2 systems yesterday).  The erase portion of the update was running super slow and we hit this watchdog again :(  We'd mostly worked around this issue on our system by putting the faster flash chips in.  We really need a long term solution to this problem.

shenki, 2016-11-08:
@geissonator can you please provide details with your bug report.

What was the kernel version?

What was the system version (/etc/os-release)?

What were you flashing?

What was the command(s) you used?

Have you reproduced it more than once?

Thanks

geissonator, 2016-11-08:
I didn't check the code level before the update :(
I was going to ubuntu,barreleye #337-b168869f062b3d
Code update flow is prepareForUpdate, preserve network settings, copy in all.tar, reboot into update.
System was BarD04

```
shutdown: reboot --log-level 6 --log-target kmsg --log-color
+ sort -r
+ awk /oldroot|mnt/ { print $2 }
+ umount /oldroot/sys/kernel/debug
+ umount /oldroot/sys/fs/cgroup/systemd
+ umount /oldroot/sys/fs/cgroup
+ umount /oldroot/sys
+ umount /oldroot/proc
+ umount /oldroot/dev/shm
+ umount /oldroot/dev/pts
+ umount /oldroot/dev
+ umount /oldroot
+ umount /mnt
+ set +x
update: --clean-saved-files
jffs2: notice: (1063) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
Updating bmc...
Erasing block: 6345/8192 (77%) 
```

shenki, 2016-11-09:
I believe this bug as been resolved as of  b168869f062b3d6ef49bfa51022c99959cc03160. Please reopen if you find it persists.

amboar, 2016-11-21:
@shenki please reopen this issue. I've been bitten by it twice recently.

amboar, 2016-11-22:
Half-serious commit that works around the problem for the moment:

https://gerrit.openbmc-project.xyz/#/c/1238/

geissonator, 2017-09-18:
Haven't seen this one in a long time.  We've had a lot of changes in this area so closing it out.

closed 2017-09-18 by geissonator