openbmc/openbmc #523: Console to host does not work with master

geissonator, 2016-08-24:
There's been some discussion on this on the mailing list and I chatted with @williamspatrick a bit as well today.  Basically the tty has moved for the host console in the latest code.  It went from ttyS5 to ttyS0.  Also, it seems like we have a issue with obmc-console-server config/startup.

The workaround was to simply run "obmc-console-server ttyS0" in one ssh session and then obmc-console-client showed the host output during boot.

This issue is to track the short term fix for this problem.

Aug 24 13:54:31 palmetto systemd[1]: Started OpenBMC console daemon.
Aug 24 13:54:31 palmetto systemd[1]: obmc-console.service: Main process exited, code=exited, status=1/FAILURE
Aug 24 13:54:31 palmetto obmc-console-server[692]: obmc-console-server: Required argument <DEVICE> missing
Aug 24 13:54:31 palmetto obmc-console-server[692]: usage: /usr/sbin/obmc-console-server [options] <DEVICE>
Aug 24 13:54:31 palmetto obmc-console-server[692]: Options:
Aug 24 13:54:31 palmetto obmc-console-server[692]:   --config <FILE>  Use FILE for configuration
Aug 24 13:54:31 palmetto systemd[1]: obmc-console.service: Unit entered failed state.
Aug 24 13:54:31 palmetto systemd[1]: obmc-console.service: Failed with result 'exit-code'.
Aug 24 13:54:31 palmetto systemd[1]: obmc-console.service: Service hold-off time over, scheduling restart.
Aug 24 13:54:31 palmetto systemd[1]: Stopped OpenBMC console daemon.
Aug 24 13:54:31 palmetto systemd[1]: obmc-console.service: Start request repeated too quickly.
Aug 24 13:54:31 palmetto systemd[1]: Failed to start OpenBMC console daemon.
root@palmetto:~#

Need to update /etc/obmc-console.conf

shenki, 2016-08-29:
This is a duplicate of #528

williamspatrick, 2016-08-29:
This is going to track the first change to move to S0.  #528 is tracking the long-term solution.

closed 2016-08-29 by williamspatrick