openbmc/openbmc #524: [System - Barreleye/Palmetto ] REST Post method are failing due to ConnectionError

rahulmah, 2016-08-24:
Most REST Post method are failing with due to ConnectionError: HTTPSConnectionPool. This seem to be happening because its response is getting timeout or did not receive a reply.

This issue is seen with latest builds of both Palmetto (20160822184209) and Barreleye(20160823212851).

**Example of failure test case in Palmetto:**

*******************************************************
	Palmetto BMC Info
*******************************************************
BMC Version: 20160822184209
BUILD_ID="v1.0"
*******************************************************

**Create and delete user without groupname :: ***GOOD PATH*****           URI:https://9.3.164.153/org/openbmc/UserManager/Users/action/UserAdd, method:Post, args:{u'data': {u'data': ['1oxANYdE', 'PPRgjKtn', '', 'fmgLMurs']}, u'headers': {u'Content-Type': u'application/json'}}
| FAIL |
ConnectionError: HTTPSConnectionPool(host='9.3.164.153', port=443): Max retries exceeded with url: /org/openbmc/UserManager/Users/action/UserAdd (Caused by ReadTimeoutError("HTTPSConnectionPool(host='9.3.164.153', port=443): Read timed out. (read timeout=5.0)",))

curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d "{\"data\": [\1oxANYdE\", \"PPRgjKtn\", \"\", \"fmgLMurs\"]}" https://9.3.164.153/org/openbmc/UserManager/Users/action/UserAdd
{
  "data": {
    "description": "Internal Server Error",
    "exception": "DBusException('Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.',)",
    "traceback": [
      "Traceback (most recent call last):",
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle",
      "    return route.call(**args)",
      "  File \"/usr/sbin/obmc-rest\", line 524, in __call__",
      "    return self.callback(*a, **kw)",
      "  File \"/usr/sbin/obmc-rest\", line 680, in wrap",
      "    return self.to_jsonp(callback(*a, **kw))",
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1734, in wrapper",
      "    rv = callback(*a, **ka)",
      "  File \"/usr/sbin/obmc-rest\", line 615, in wrap",
      "    resp = {'data': callback(*a, **kw)}",
      "  File \"/usr/sbin/obmc-rest\", line 578, in wrap",
      "    return callback(*a, **kw)",
      "  File \"/usr/sbin/obmc-rest\", line 603, in wrap",
      "    return callback(*a, **kw)",
      "  File \"/usr/sbin/obmc-rest\", line 93, in __call__",
      "    return getattr(self, 'do_' + request.method.lower())(**kw)",
      "  File \"/usr/sbin/obmc-rest\", line 214, in do_post",
      "    return request.route_data['method'](*request.parameter_list)",
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__",
      "    **keywords)",
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking",
      "    message, timeout)",
      "DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken."
    ]
  },
  "message": "500 Internal Server Error",
  "status": "error"
}rahulmah@rackspace:~$

While on the other hand Post method to check existing userlist works fine.
rahulmah@rackspace:~$ curl c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d "{\"data\": []}" https://9.3.164.153/org/openbmc/UserManager/Users/action/UserList
{
  "data": [
    "fhMoWDpa",
    "IYcqs4qo",
    ..
    ..
    "PPRgjKtn"
  ],
  "message": "200 OK",
  "status": "ok"

gkeishin, 2016-08-25:
@rahulmah I see it working in today test run on

BMC Version: 20160824194547
BUILD_ID="v1.0"

which is a latter version .. so i think you can close this one..

Create and delete user group :: ***GOOD PATH***                       URI:https://9.3.164.153/org/openbmc/UserManager/Groups/action/GroupAddUsr, method:Post, args:{u'data': {u'data': ['jHswPAnw']}, u'headers': {u'Content-Type': u'application/json'}}
Response code:200, Content:{
  "data": 0,
  "message": "200 OK",
  "status": "ok"
}
URI:https://9.3.164.153/org/openbmc/UserManager/Groups/action/GroupListUsr, method:Post, args:{u'data': {u'data': []}, u'headers': {u'Content-Type': u'application/json'}}
Response code:200, Content:{
  "data": [
    "fhMoWDpa",
    "IYcqs4qo",
    "ZPK27mag",
    "aPkRbJCb",
    "b4dqE6U3",
    "g0wSkdhY",
    "5y7FZgAu",
    "4WhINzIZ",
    "x9bwbGDe",
    "B6khBnFw",
    "iEX0CwMg",
    "ppZIKxXC",
    "Fr28IeR3",
    "mQJ9Bn5B",
    "dEDA2tBi",
    "O3V9iosG",
    "ukVSZuyF",
    "toDaLQM4",
    "mGPAQHpY",
    "aZ5i5GAp",
    "vV5eOK0v",
    "I7nYw3iH",
    "VgJYXvlz",
    "E9SClSh3",
    "tJRx8Be0",
    "hSkoGSmr",
    "raroSI7b",
    "JnKa9cNE",
    "I3aKqHFe",
    "1d3TSa3w",
    "jaFUfqrs",
    "RZpv0HnP",
    "uhIL7g3U",
    "xwJVcGPd",
    "1FTQyFCa",
    "RNrTn19D",
    "5tHpVKTM",
    "PPRgjKtn",
    "jHswPAnw"
  ],
  "message": "200 OK",
  "status": "ok"
}
URI:https://9.3.164.153/org/openbmc/UserManager/Group/action/GroupDel, method:Post, args:{u'data': {u'data': ['jHswPAnw']}, u'headers': {u'Content-Type': u'application/json'}}
Response code:200, Content:{
  "data": 0,
  "message": "200 OK",
  "status": "ok"
}
| PASS |

rahulmah, 2016-08-25:
Opened issue in openbmc-test-automation - https://github.com/openbmc/openbmc-test-automation/issues/83

gkeishin, 2016-08-25:
@bradbishop If there is anything on this we need to know for REST .. From test perspective, we are pumping those REST request back to back.. so if one fails, rest seems to fail..

rahulmah, 2016-08-25:
@bradbishop : This is what might be causing rest commands to fail.
In both Barreleye and Palmetto system, after applying set current IP address, we try to check if the set ip is pinging .. There it passes. But then I am seeing that rest command are failing due to ConnectionError. So it seem like configuring IP is leading to ConnectionError: HTTPSConnectionPool (due to rest login problem).

bradbishop, 2016-08-25:
this is a known issue: https://github.com/openbmc/phosphor-rest-server/issues/24

can you verify you have the fix?

bradbishop, 2016-08-25:
could also be this one:

https://github.com/openbmc/openbmc/issues/514

rahulmah, 2016-08-25:
@bradbishop The issue (RestAPI curl couldn't connect) which you have pointed is same but it has already been closed. I have tested with latest OpenBMC v1.0 build - 20160824194547 and this issue is still seen. Other defect which you pointed does not seem to related to network change.

hweaving, 2016-08-25:
rahulmah, can you ssh to the BMC (or VNC if you are using qemu) and check journalctl for Python errors?  Also, can you run "ps | grep rest" to check whether the REST service is running on the BMC?

rahulmah, 2016-08-26:
@hweaving : Sure I will checking that

hweaving, 2016-08-26:
@bradbishop @rahulmah That looks different than #514.  In your case, obmc-rest is using 100% CPU rather than crashing.

williamspatrick, 2016-08-29:
Discussed this on a call with @rahulmah and @sivassrr.

1. We believe the test framework is attempting to access the rest server too quickly after changing the IP address.  Due to the server restarting as part of the ip address change, the test framework needs to wait 10-30 seconds.

2. There is a pending change to move from Rocket to gevent.  #147.  This is scheduled for this month or early next month.

closed 2016-08-29 by williamspatrick